Koenkk/zigbee2mqtt #15349: [New device support]: Датчик дыма Tuya Zigbee TS0601

seesrza, 2022-12-04:
### Link

https://aliexpress.ru/item/1005003970944017.html

### Database entry

{"id":3,"type":"EndDevice","ieeeAddr":"0xa4c13879b8a471c2","nwkAddr":22977,"manufId":4417,"manufName":"_TZE200_m9skfctm","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�\"\u001e+\u0011","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_m9skfctm","powerSource":3,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1670154805583,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to add support, but I'm extremely new to doing anything under the hood on zigbee2mqtt. I followed the adding support guide, but got stumped as to what to do next.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_m9skfctm',
        },
    ],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'Photoelectric Smoke Detector',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
        ],
    },
};

module.exports = definition;
```

Koenkk, 2022-12-04:
Is anything logged int he debug log when triggering the sensor with smoke?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

seesrza, 2022-12-05:
> Записывается ли что-нибудь в журнал отладки при срабатывании датчика с дымом?
> 
> См. https://www.zigbee2mqtt.io/guide/usage/debug.html о том, как включить ведение журнала отладки.

this report?

Debug 2022-12-05 10:44:38Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":101}],"seq":768}' from endpoint 1 with groupID 0
Debug 2022-12-05 10:44:38Datapoint 101 not defined for '_TZE200_m9skfctm' with data {"dp":101,"datatype":1,"data":{"type":"Buffer","data":[0]}}
Debug 2022-12-05 10:44:38Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":1024}' from endpoint 1 with groupID 0
Debug 2022-12-05 10:44:38Datapoint 1 not defined for '_TZE200_m9skfctm' with data {"dp":1,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Debug 2022-12-05 10:44:38Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,150],"type":"Buffer"},"datatype":2,"dp":2}],"seq":1280}' from endpoint 1 with groupID 0
Debug 2022-12-05 10:44:38Datapoint 2 not defined for '_TZE200_m9skfctm' with data {"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,150]}}
Debug 2022-12-05 10:44:39Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15}],"seq":1536}' from endpoint 1 with groupID 0
Debug 2022-12-05 10:44:39Datapoint 15 not defined for '_TZE200_m9skfctm' with data {"dp":15,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,100]}}
Debug 2022-12-05 10:44:39Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":11}],"seq":1792}' from endpoint 1 with groupID 0
Debug 2022-12-05 10:44:39Datapoint 11 not defined for '_TZE200_m9skfctm' with data {"dp":11,"datatype":1,"data":{"type":"Buffer","data":[0]}}
Debug 2022-12-05 10:44:41Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":101}],"seq":2048}' from endpoint 1 with groupID 0
Debug 2022-12-05 10:44:41Datapoint 101 not defined for '_TZE200_m9skfctm' with data {"dp":101,"datatype":1,"data":{"type":"Buffer","data":[0]}}
Debug 2022-12-05 10:44:41Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":2304}' from endpoint 1 with groupID 0
Debug 2022-12-05 10:44:41Datapoint 1 not defined for '_TZE200_m9skfctm' with data {"dp":1,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Debug 2022-12-05 10:44:41Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,216],"type":"Buffer"},"datatype":2,"dp":2}],"seq":2560}' from endpoint 1 with groupID 0
Debug 2022-12-05 10:44:41Datapoint 2 not defined for '_TZE200_m9skfctm' with data {"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,216]}}
Debug 2022-12-05 10:44:41Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15}],"seq":2816}' from endpoint 1 with groupID 0
Debug 2022-12-05 10:44:41Datapoint 15 not defined for '_TZE200_m9skfctm' with data {"dp":15,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,100]}}
Debug 2022-12-05 10:44:42Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":11}],"seq":3072}' from endpoint 1 with groupID 0
Debug 2022-12-05 10:44:42Datapoint 11 not defined for '_TZE200_m9skfctm' with data {"dp":11,"datatype":1,"data":{"type":"Buffer","data":[0]}}
Debug 2022-12-05 10:44:44Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":101}],"seq":3328}' from endpoint 1 with groupID 0
Debug 2022-12-05 10:44:44Datapoint 101 not defined for '_TZE200_m9skfctm' with data {"dp":101,"datatype":1,"data":{"type":"Buffer","data":[0]}}
Debug 2022-12-05 10:44:44Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":3584}' from endpoint 1 with groupID 0
Debug 2022-12-05 10:44:44Datapoint 1 not defined for '_TZE200_m9skfctm' with data {"dp":1,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Debug 2022-12-05 10:44:44Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,33],"type":"Buffer"},"datatype":2,"dp":2}],"seq":3840}' from endpoint 1 with groupID 0
Debug 2022-12-05 10:44:44Datapoint 2 not defined for '_TZE200_m9skfctm' with data {"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,33]}}
[... cut ...]
Debug 2022-12-05 10:45:00Datapoint 15 not defined for '_TZE200_m9skfctm' with data {"dp":15,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,100]}}
Debug 2022-12-05 10:45:00Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":11}],"seq":9472}' from endpoint 1 with groupID 0
Debug 2022-12-05 10:45:00Datapoint 11 not defined for '_TZE200_m9skfctm' with data {"dp":11,"datatype":1,"data":{"type":"Buffer","data":[0]}}
Debug 2022-12-05 10:45:46Saving state to file /config/zigbee2mqtt/state.json

REDKIY-RAK, 2022-12-05:
Hello. I will not create a new topic, I will write here so as not to clog. I'm trying to connect a floor heating thermostat from Moes.  Termostat model BHT-002-GCLZB. 

**External converter** (/config/zigbee2mqtt/TERMOSTAT_Moes.js)  This is my test version.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_aoclfnxz',
        },
    ],
    model: 'TS0601_new',
    vendor: 'TuYa',
    description: 'Moes BHT series Thermostat',
    fromZigbee: [fz.moes_thermostat],
    toZigbee: [tz.moes_thermostat_child_lock, tz.moes_thermostat_current_heating_setpoint, tz.moes_thermostat_mode, tz.moes_thermostat_standby],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [exposes.climate().withSystemMode(['heat', 'auto', 'off'], ea.STATE_SET)],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
        ],
    },
};

module.exports = definition;

```

**LOG Z2M**

```
Zigbee2MQTT:warn  2022-12-06 00:29:31: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
Zigbee2MQTT:warn  2022-12-06 00:29:31: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:debug 2022-12-06 00:29:31: Received Zigbee message from '0x04cd15fffe22750d', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":23560}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2022-12-06 00:29:31: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
Zigbee2MQTT:warn  2022-12-06 00:29:31: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:debug 2022-12-06 00:29:31: Received Zigbee message from '0x04cd15fffe22750d', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":23816}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2022-12-06 00:29:31: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
Zigbee2MQTT:warn  2022-12-06 00:29:31: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:debug 2022-12-06 00:29:41: Received Zigbee message from '0x04cd15fffe22750d', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,9],"type":"Buffer"},"datatype":2,"dp":24}],"seq":19464}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2022-12-06 00:29:41: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
Zigbee2MQTT:warn  2022-12-06 00:29:41: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:debug 2022-12-06 00:29:48: Received MQTT message on 'zigbee2mqtt/bridge/request/device/remove' with data '{"block":false,"force":true,"id":"0x04cd15fffe22750d","transaction":"rmbpb-1"}'
Zigbee2MQTT:info  2022-12-06 00:29:48: Removing device '0x04cd15fffe22750d' (block: false, force: true)
Zigbee2MQTT:debug 2022-12-06 00:29:48: Clearing Home Assistant discovery topic for '0x04cd15fffe22750d'
Zigbee2MQTT:info  2022-12-06 00:29:48: MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe22750d', payload ''
Zigbee2MQTT:info  2022-12-06 00:29:48: Successfully removed device '0x04cd15fffe22750d' (block: false, force: true)
Zigbee2MQTT:info  2022-12-06 00:29:48: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":true,"id":"0x04cd15fffe22750d"},"status":"ok","transaction":"rmbpb-1"}'
Zigbee2MQTT:debug 2022-12-06 00:31:06: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2022-12-06 00:36:06: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2022-12-06 00:41:06: Saving state to file /config/zigbee2mqtt/state.json
```
What am I doing wrong, why is it not supported in Z2M? Please help me )

seesrza, 2022-12-06:
Hello! I've only just started to study tmeme myself, so I can't help
  
>Вторник, 6 декабря 2022, 2:47 +05:00 от REDKIY-RAK ***@***.***>:
> 
> 
>Привет.  Новую тему создавать не буду, напишу здесь, чтобы не засорять.  Пытаюсь подключить термостат напольного отопления от Moes.  Модель термостата BHT-002-GCLZB.
>Внешний преобразователь (/config/zigbee2mqtt/TERMOSTAT_Moes.js) Это моя тестовая версия.
>const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
>const tz = require('zigbee-herdsman-converters/converters/toZigbee');
>const exposes = require('zigbee-herdsman-converters/lib/exposes');
>const report = require('zigbee-herdsman-converters/lib/reporting');
>const extend = require('zigbee-herdsman-converters/lib/extend');
>const e = exposes.presets;
>const ea = exposes.access;
>const tuya = require('zigbee-herdsman-converters/lib/tuya');
>
>константное определение = {
>    // Поскольку многие устройства TuYa используют один и тот же идентификатор модели, но используют разные точки данных
>    // необходимо предоставить отпечаток вместо zigbeeModel
>    отпечаток пальца: [
>        {
>            // Идентификатор модели от: Устройство с идентификатором модели «TS0601» не поддерживается
>            // В некоторых случаях вам может понадобиться добавить \u0000 в конце имени
>            идентификатор модели: «TS0601»,
>            // Название производителя из: Устройство с идентификатором модели «TS0601» не поддерживается.
>            имя производителя: '_TZE200_aoclfnxz',
>        },
>    ],
>    модель: 'TS0601_new',
>    продавец: 'ЮЯ',
>    описание: «Термостат серии Moes BHT»,
>    от Zigbee: [fz.moes_thermostat],
>    toZigbee: [tz.moes_thermostat_child_lock, tz.moes_thermostat_current_heating_setpoint, tz.moes_thermostat_mode, tz.moes_thermostat_standby],
>    onEvent: tuya.onEventSetTime, // Добавьте это, если вы не получаете конвертер для 'commandMcuSyncTime'
>    настроить: tuya.configureMagicPacket,
>    выставляет: [exposes.climate().withSystemMode(['heat', 'auto', 'off'], ea.STATE_SET)],
>    мета: {
>        // Все точки данных идут сюда
>        tuyaDatapoints: [
>        ],
>    },
>};
>
>модуль.экспорт = определение;
>
>
>ЛОГ Z2M
>Zigbee2MQTT:warn 06.12.2022, 00:29:31: получено сообщение от неподдерживаемого устройства с моделью Zigbee «не определено» и названием производителя «не определено».
>Zigbee2MQTT: предупреждение 06.12.2022, 00:29:31: см.: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
>Zigbee2MQTT: отладка 2022-12-06 00:29:31: получено сообщение Zigbee от «0x04cd15fffe22750d», тип «commandDataResponse», кластер «manuSpecificTuya», данные «{"dpValues":[{"data":{"data": [1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":23560}' из конечной точки 1 с groupID 0
>Zigbee2MQTT:warn 06.12.2022, 00:29:31: получено сообщение от неподдерживаемого устройства с моделью Zigbee «не определено» и названием производителя «не определено».
>Zigbee2MQTT: предупреждение 06.12.2022, 00:29:31: см.: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
>Zigbee2MQTT: отладка 2022-12-06 00:29:31: получено сообщение Zigbee от «0x04cd15fffe22750d», тип «commandDataResponse», кластер «manuSpecificTuya», данные «{"dpValues":[{"data":{"data": [1],"type":"Buffer"},"datatype":4,"dp":36}],"seq":23816}' из конечной точки 1 с groupID 0
>Zigbee2MQTT:warn 06.12.2022, 00:29:31: получено сообщение от неподдерживаемого устройства с моделью Zigbee «не определено» и названием производителя «не определено».
>Zigbee2MQTT: предупреждение 06.12.2022, 00:29:31: см.: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
>Zigbee2MQTT: отладка 2022-12-06 00:29:41: получено сообщение Zigbee от «0x04cd15fffe22750d», тип «commandDataResponse», кластер «manuSpecificTuya», данные «{"dpValues":[{"data":{"data": [0,0,1,9],"type":"Buffer"},"datatype":2,"dp":24}],"seq":19464}' из конечной точки 1 с groupID 0
>Zigbee2MQTT:warn 06.12.2022, 00:29:41: получено сообщение от неподдерживаемого устройства с моделью Zigbee «не определено» и названием производителя «не определено».
>Zigbee2MQTT: предупреждение 06.12.2022, 00:29:41: см.: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
>Zigbee2MQTT: отладка 2022-12-06 00:29:48: получено сообщение MQTT на 'zigbee2mqtt/bridge/request/device/remove' с данными '{"block":false,"force":true,"id":" 0x04cd15fffe22750d","транзакция":"rmbpb-1"}'
>Zigbee2MQTT:info 06.12.2022, 00:29:48: Удаление устройства 0x04cd15fffe22750d (блокировка: ложь, принудительное выполнение: истина)
>Zigbee2MQTT: отладка 2022-12-06 00:29:48: Очистка темы обнаружения Home Assistant для «0x04cd15fffe22750d»
>Zigbee2MQTT:info 2022-12-06 00:29:48: публикация MQTT: тема 'zigbee2mqtt/0x04cd15fffe22750d', полезная нагрузка ''
>Zigbee2MQTT:info 2022-12-06 00:29:48: Успешно удалено устройство 0x04cd15fffe22750d (блокировка: ложь, сила: истина)
>Zigbee2MQTT:info 2022-12-06 00:29:48: Публикация MQTT: тема 'zigbee2mqtt/bridge/response/device/remove', полезная нагрузка '{"data":{"block":false,"force":true, "id": "0x04cd15fffe22750d"}, "status": "ok", "транзакция": "rmbpb-1"}'
>Zigbee2MQTT:debug 2022-12-06 00:31:06: Сохранение состояния в файл /config/zigbee2mqtt/state.json
>Zigbee2MQTT:debug 2022-12-06 00:36:06: Сохранение состояния в файл /config/zigbee2mqtt/state.json
>Zigbee2MQTT:debug 2022-12-06 00:41:06: Сохранение состояния в файл /config/zigbee2mqtt/state.json
>
>Что я делаю не так, почему это не поддерживается в Z2M?  Помогите мне, пожалуйста )
>— 
>Ответьте на это письмо напрямую,  просмотрите его на GitHub или  отмените подписку . 
>Вы получаете это, потому что вы создали тему. Идентификатор сообщения:  <Koenkk/zigbee2mqtt/issues/15349/1338219751  @  github  .  ком> 
 
 
С уважением,
Главный инженер Салехардского РЭС
Пушкарев Андрей Николаевич
тел. +79924040014

seesrza, 2022-12-06:
> Можете ли вы попробовать с этим конвертером:
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const utils = require('zigbee-herdsman-converters/lib/utils');
> const globalStore = require('zigbee-herdsman-converters/lib/store');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const definition = {
>     fingerprint: tuya.fingerprint('TS0601', ['_TZE200_ntcy3xu1', '_TZE200_m9skfctm']),
>     model: 'TS0601_smoke_1',
>     vendor: 'TuYa',
>     description: 'Smoke sensor',
>     fromZigbee: [tuya.fz.datapoints],
>     toZigbee: [tuya.tz.datapoints],
>     configure: tuya.configureMagicPacket,
>     exposes: [e.smoke(), e.tamper(), e.battery_low()],
>     meta: {
>         tuyaDatapoints: [
>             [1, 'smoke', tuya.valueConverter.true0ElseFalse],
>             [4, 'tamper', tuya.valueConverter.raw],
>             [14, 'battery_low', tuya.valueConverter.true0ElseFalse],
>         ],
>     },
> };
> 
> module.exports = definition;
> ```

Thanks a lot!!!

seesrza, 2022-12-07:
> Может ли вы попробовать это с конвертером:
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const utils = require('zigbee-herdsman-converters/lib/utils');
> const globalStore = require('zigbee-herdsman-converters/lib/store');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const definition = {
>     fingerprint: tuya.fingerprint('TS0601', ['_TZE200_ntcy3xu1', '_TZE200_m9skfctm']),
>     model: 'TS0601_smoke_1',
>     vendor: 'TuYa',
>     description: 'Smoke sensor',
>     fromZigbee: [tuya.fz.datapoints],
>     toZigbee: [tuya.tz.datapoints],
>     configure: tuya.configureMagicPacket,
>     exposes: [e.smoke(), e.tamper(), e.battery_low()],
>     meta: {
>         tuyaDatapoints: [
>             [1, 'smoke', tuya.valueConverter.true0ElseFalse],
>             [4, 'tamper', tuya.valueConverter.raw],
>             [14, 'battery_low', tuya.valueConverter.true0ElseFalse],
>         ],
>     },
> };
> 
> module.exports = definition;
> ```
the report is like this:

Info 2022-12-07 10:38:19MQTT publish: topic 'zigbee2mqtt/Датчик дыма кухня', payload '{"battery_low":false,"fault_alarm":"0","linkquality":188,"self_test":true,"smoke":false,"tamper":null}'
Debug 2022-12-07 10:38:19Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,112],"type":"Buffer"},"datatype":2,"dp":2}],"seq":48896}' from endpoint 1 with groupID 0
Debug 2022-12-07 10:38:19Datapoint 2 not defined for '_TZE200_m9skfctm' with data {"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,112]}}
Debug 2022-12-07 10:38:20Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15}],"seq":49152}' from endpoint 1 with groupID 0
Debug 2022-12-07 10:38:20Datapoint 15 not defined for '_TZE200_m9skfctm' with data {"dp":15,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,100]}}
Debug 2022-12-07 10:38:20Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":11}],"seq":49408}' from endpoint 1 with groupID 0
Debug 2022-12-07 10:38:20Datapoint 11 not defined for '_TZE200_m9skfctm' with data {"dp":11,"datatype":1,"data":{"type":"Buffer","data":[0]}}

Koenkk, 2022-12-07:
I don't know what these datapoints mean, to figure this out see: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

seesrza, 2022-12-17:
> I don't know what these datapoints mean, to figure this out see: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

Made a new converter. I get all the necessary information from the sensor.
But as a result, that I have no experience, I get a lot of errors in the log. Can you see where I went wrong?

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [ {   modelID: 'TS0601', manufacturerName: '_TZE200_m9skfctm'},],
    model: 'TS0601_new',
    vendor: 'TuYa',
    description: 'Photoelectric Smoke Detector',
    fromZigbee: [tuya.fz.datapoints, fz.tuya_air_quality],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, 
    configure: tuya.configureMagicPacket,
    exposes: [e.smoke(), e.battery(), e.co2(), e.tamper(), e.battery_low()],
    meta: {
        tuyaDatapoints: [
          [1, 'smoke', tuya.valueConverter.true0ElseFalse],
          [101, 'co2', tuya.valueConverter.raw],
          [15,'battery', tuya.valueConverter.raw],
          [11,'tamper', tuya.valueConverter.true0ElseFalse],
          [2,'battery_low', tuya.valueConverter.true0ElseFalse],
          
        ],
    },
};

module.exports = definition;

seesrza, 2022-12-17:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [ {   modelID: 'TS0601', manufacturerName: '_TZE200_m9skfctm'},],
    model: 'TS0601_new',
    vendor: 'TuYa',
    description: 'Photoelectric Smoke Detector',
    fromZigbee: [tuya.fz.datapoints, fz.tuya_air_quality],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, 
    configure: tuya.configureMagicPacket,
    exposes: [e.smoke(), e.battery(), e.co2(), e.tamper(), e.battery_low()],
    meta: {
        tuyaDatapoints: [
          [1, 'smoke', tuya.valueConverter.true0ElseFalse],
          [101, 'co2', tuya.valueConverter.raw],
          [15,'battery', tuya.valueConverter.raw],
          [11,'tamper', tuya.valueConverter.true0ElseFalse],
          [2,'battery_low', tuya.valueConverter.true0ElseFalse],
          
        ],
    },
};

module.exports = definition;

seesrza, 2022-12-17:
Debug 2022-12-17 18:07:53Received Zigbee message from '0xa4c13879b8a471c2', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":101}],"seq":2560}' from endpoint 1 with groupID 0
Warning 2022-12-17 18:07:53zigbee-herdsman-converters:TuyaSmartAirBox: Unrecognized DP #101 with data {"dp":101,"datatype":1,"data":{"type":"Buffer","data":[0]}}
Info 2022-12-17 18:07:53MQTT publish: topic 'zigbee2mqtt/0xa4c13879b8a471c2', payload '{"battery":90,"battery_low":false,"co2":false,"linkquality":236,"smoke":false,"tamper":false}'
Debug 2022-12-17 18:07:53Received Zigbee message from '0xa4c13879b8a471c2', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":2816}' from endpoint 1 with groupID 0
Warning 2022-12-17 18:07:53zigbee-herdsman-converters:TuyaSmartAirBox: Unrecognized DP #1 with data {"dp":1,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Info 2022-12-17 18:07:53MQTT publish: topic 'zigbee2mqtt/0xa4c13879b8a471c2', payload '{"battery":90,"battery_low":false,"co2":false,"linkquality":255,"smoke":true,"tamper":false}'
Debug 2022-12-17 18:07:53Received Zigbee message from '0xa4c13879b8a471c2', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,250],"type":"Buffer"},"datatype":2,"dp":2}],"seq":3072}' from endpoint 1 with groupID 0
Info 2022-12-17 18:07:53MQTT publish: topic 'zigbee2mqtt/0xa4c13879b8a471c2', payload '{"battery":90,"battery_low":false,"co2":250,"linkquality":208,"smoke":true,"tamper":false}'
Debug 2022-12-17 18:07:53Received Zigbee message from '0xa4c13879b8a471c2', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,92],"type":"Buffer"},"datatype":2,"dp":15}],"seq":3328}' from endpoint 1 with groupID 0
Warning 2022-12-17 18:07:53zigbee-herdsman-converters:TuyaSmartAirBox: Unrecognized DP #15 with data {"dp":15,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,92]}}
Info 2022-12-17 18:07:53MQTT publish: topic 'zigbee2mqtt/0xa4c13879b8a471c2', payload '{"battery":92,"battery_low":false,"co2":250,"linkquality":148,"smoke":true,"tamper":false}'
Debug 2022-12-17 18:07:53Received Zigbee message from '0xa4c13879b8a471c2', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":11}],"seq":3584}' from endpoint 1 with groupID 0
Warning 2022-12-17 18:07:53zigbee-herdsman-converters:TuyaSmartAirBox: Unrecognized DP #11 with data {"dp":11,"datatype":1,"data":{"type":"Buffer","data":[0]}}
Info 2022-12-17 18:07:53MQTT publish: topic 'zigbee2mqtt/0xa4c13879b8a471c2', payload '{"battery":92,"battery_low":false,"co2":250,"linkquality":152,"smoke":true,"tamper":false}'
Debug 2022-12-17 18:07:55Received Zigbee message from '0xa4c13879b8a471c2', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":101}],"seq":3840}' from endpoint 1 with groupID 0
Warning 2022-12-17 18:07:56zigbee-herdsman-converters:TuyaSmartAirBox: Unrecognized DP #101 with data {"dp":101,"datatype":1,"data":{"type":"Buffer","data":[0]}}
Info 2022-12-17 18:07:56MQTT publish: topic 'zigbee2mqtt/0xa4c13879b8a471c2', payload '{"battery":92,"battery_low":false,"co2":false,"linkquality":188,"smoke":true,"tamper":false}'
Debug 2022-12-17 18:07:56Received Zigbee message from '0xa4c13879b8a471c2', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":4096}' from endpoint 1 with groupID 0
Warning 2022-12-17 18:07:56zigbee-herdsman-converters:TuyaSmartAirBox: Unrecognized DP #1 with data {"dp":1,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Info 2022-12-17 18:07:56MQTT publish: topic 'zigbee2mqtt/0xa4c13879b8a471c2', payload '{"battery":92,"battery_low":false,"co2":false,"linkquality":192,"smoke":true,"tamper":false}'
Debug 2022-12-17 18:07:56Received Zigbee message from '0xa4c13879b8a471c2', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,253],"type":"Buffer"},"datatype":2,"dp":2}],"seq":4352}' from endpoint 1 with groupID 0
Info 2022-12-17 18:07:56MQTT publish: topic 'zigbee2mqtt/0xa4c13879b8a471c2', payload '{"battery":92,"battery_low":false,"co2":253,"linkquality":196,"smoke":true,"tamper":false}'
Debug 2022-12-17 18:07:56Received Zigbee message from '0xa4c13879b8a471c2', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,92],"type":"Buffer"},"datatype":2,"dp":15}],"seq":4608}' from endpoint 1 with groupID 0
Warning 2022-12-17 18:07:56zigbee-herdsman-converters:TuyaSmartAirBox: Unrecognized DP #15 with data {"dp":15,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,92]}}
Info 2022-12-17 18:07:56MQTT publish: topic 'zigbee2mqtt/0xa4c13879b8a471c2', payload '{"battery":92,"battery_low":false,"co2":253,"linkquality":196,"smoke":true,"tamper":false}'
Debug 2022-12-17 18:07:56Received Zigbee message from '0xa4c13879b8a471c2', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":11}],"seq":4864}' from endpoint 1 with groupID 0
Warning 2022-12-17 18:07:56zigbee-herdsman-converters:TuyaSmartAirBox: Unrecognized DP #11 with data {"dp":11,"datatype":1,"data":{"type":"Buffer","data":[0]}}
[... cut ...]
Debug 2022-12-17 18:08:15Received Zigbee message from '0xa4c13879b8a471c2', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":11}],"seq":9984}' from endpoint 1 with groupID 0
Warning 2022-12-17 18:08:15zigbee-herdsman-converters:TuyaSmartAirBox: Unrecognized DP #11 with data {"dp":11,"datatype":1,"data":{"type":"Buffer","data":[0]}}
Info 2022-12-17 18:08:15MQTT publish: topic 'zigbee2mqtt/0xa4c13879b8a471c2', payload '{"battery":92,"battery_low":false,"co2":27,"linkquality":164,"smoke":false,"tamper":false}'

Koenkk, 2022-12-18:
`fz.tuya_air_quality` should not be there, does it work without errors after removing it?

seesrza, 2022-12-18:
> `fz.tuya_air_quality`там не должно быть, работает без ошибок после удаления?

without it, it does not display the CO level. with it displays an almost exact value
[image]

seesrza, 2022-12-18:
[image]

Koenkk, 2022-12-19:
Try adding the following to `tuyaDatapoints`: 

- `[2, 'co2', tuya.valueConverter.raw],`
- `[22, 'co2', tuya.valueConverter.raw],`

fulup-bzh, 2022-12-21:
MOES smoke detector signature TS0601_TZE200_dnz6yvl2 works with following converter definition
```
const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_dnz6yvl2']),
    model: 'MOES_smoke',
    vendor: 'TuYa',
    description: 'Smoke Detector',
    fromZigbee: [tuya.fz.datapoints,fz.tuya_smoke],
    toZigbee: [],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.smoke(), e.battery_low(), e.battery()
    ],
    meta: {
        // All datapoints go in here
         tuyaDatapoints: [
                [1, 'smoke', tuya.valueConverter.true0ElseFalse],
                [15,'battery', tuya.valueConverter.raw],
                [14,'battery_low', tuya.valueConverter.true0ElseFalse],
            ],
    },
};
```
Note: following DPs are still not handled #2 could the be % of smoke but more test should be done to make sure about it.
```
2022-12-21 21:25:21zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #2 with data {"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
2022-12-21 21:25:21zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #11 with data {"dp":11,"datatype":5,"data":{"type":"Buffer","data":[0]}}
2022-12-21 21:25:27zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #17 with data {"dp":17,"datatype":1,"data":{"type":"Buffer","data":[0]}}
2022-12-21 21:25:32zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #21 with data {"dp":21,"datatype":4,"data":{"type":"Buffer","data":[0]}}
```

Koenkk, 2022-12-22:
@fulup-bzh thanks, added `_TZE200_dnz6yvl2`. I believe dp 14 is the battery_state (where 0 = low, 1 = med, 2 = high)

fulup-bzh, 2022-12-22:
I unfortunately have no simple way to test if this device provides 2 or 3 battery states. I checked within tuya.js and all tuya 'battery_low' seems to be mapped on true0ElseFalse. Nevertheless you still might be right :)

Jonno12345, 2022-12-24:
Hi all,

I've got three _TZE200_m9skfctm - however I've ended up with quite a different converter, as follows, to get mine to work:

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const numeric = exposes.numeric;

const definition = {
fingerprint: [ { modelID: 'TS0601', manufacturerName: '_TZE200_m9skfctm'},],
model: 'TS0601_new',
vendor: 'TuYa',
description: 'Photoelectric Smoke Detector',
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
onEvent: tuya.onEventSetTime,
configure: tuya.configureMagicPacket,
exposes: [e.smoke(), numeric('smoke_level', ea.STATE).withUnit("ppm").withDescription("Parts per million of smoke detected"), e.tamper(), e.battery(), e.test()],
meta: {
tuyaDatapoints: [
            [1, 'smoke', tuya.valueConverter.true0ElseFalse],
            [2, 'smoke_level', tuya.valueConverter.raw],
            [11,'tamper', tuya.valueConverter.true0ElseFalse],
            [15,'battery', tuya.valueConverter.raw],
            [101,'test', tuya.valueConverter.true0ElseFalse]
        ],
    },
};

module.exports = definition;
```

I used the Tuya gateway to find out the datapoints. 2 is actually the PPM level, which I don't believe to be CO/CO2 as it's a photoelectric sensor. As such, I've added a custom smoke_level for this, but I am no expert in smoke alarms so might have mislabelled this. I am a little concerned it's so different to the original though, presumably there aren't multiple models with the same manufacturerName..?

[image]

Koenkk, 2022-12-25:
@Jonno12345 looks good to me, can you make a PR?

kjeisa, 2022-12-25:
Hello, I also have _TZE200_m9skfctm and have used "my own" converter but tried to change to this one. The "test" didn report anything with tuya.valueConverter.true0ElseFalse I have to use tuya.valueConverter.raw. And my "smoke" value is 1 when no smoke and 0 when smoke is detected, can I get it to show Clear/detected instead? Or are there more models with the same name out there?

Jonno12345, 2022-12-25:
> 

Was that using my converter? I'm concerned there are two versions of this as my converter is really different to the original posted.

kjeisa, 2022-12-25:
> > 
> 
> Was that using my converter? I'm concerned there are two versions of this as my converter is really different to the original posted.
> 
> > Hello, I also have _TZE200_m9skfctm and have used "my own" converter but tried to change to this one. The "test" didn report anything with tuya.valueConverter.true0ElseFalse I have to use tuya.valueConverter.raw. And my "smoke" value is 1 when no smoke and 0 when smoke is detected, can I get it to show Clear/detected instead? Or are there more models with the same name out there?

Yes I tried with this one:

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const numeric = exposes.numeric;

const definition = {
fingerprint: [ { modelID: 'TS0601', manufacturerName: '_TZE200_m9skfctm'},],
model: 'TS0601_new',
vendor: 'TuYa',
description: 'Photoelectric Smoke Detector',
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
onEvent: tuya.onEventSetTime,
configure: tuya.configureMagicPacket,
exposes: [e.smoke(), numeric('smoke_level', ea.STATE).withUnit("ppm").withDescription("Parts per million of smoke detected"), e.tamper(), e.battery(), e.test()],
meta: {
tuyaDatapoints: [
            [1, 'smoke', tuya.valueConverter.true0ElseFalse],
            [2, 'smoke_level', tuya.valueConverter.raw],
            [11,'tamper', tuya.valueConverter.true0ElseFalse],
            [15,'battery', tuya.valueConverter.raw],
            [101,'test', tuya.valueConverter.raw]
        ],
    },
};

module.exports = definition;

and before I have used this one:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_m9skfctm',
        },
    ],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'Smokedetector',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [e.smoke(), e.tamper(), e.battery(), tuya.exposes.faultAlarm(), tuya.exposes.gasValue()],
    meta: {
        tuyaDatapoints: [
            [1, 'smoke', tuya.valueConverter.raw],
            [2, 'gas_value', tuya.valueConverter.raw],
            [11, 'fault_alarm', tuya.valueConverter.trueFalse],
            [15, 'battery', tuya.valueConverter.raw],
            [101, 'tamper', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;

but this is my first time trying to make a converter since there was no official so I dont know whats wrong and not, but atleast the test button only went from N/A to false when I tried yours. So I changed it to tuya.valueConverter.raw on the test button and it now works on true/false. I also made/translated the datapoints from the info I got my hands on, maybe it makes any sense?

[image]

Jonno12345, 2022-12-25:
Odd, your mappings look the same as mine. Not at my house to test right now, but I'm surprised it doesn't work with my converter if that's the data points you have as it's identical. What isn't working sorry? 1 when smoke and 0 no smoke is correct and should be the same as my converter

kjeisa, 2022-12-25:
Yes, i tried the Test button on the smoke detector and it did not change from false to true in the output. (With [101,'test', tuya.valueConverter.true0ElseFalse]) I must use [101,'test', tuya.valueConverter.raw] for it to work. Strange

kjeisa, 2022-12-25:
I tried with smoke now, and the smoke detection works good, thanks.
So looks like the correct setup for mine DP's and valueConverters are:
            [1, 'smoke', tuya.valueConverter.true0ElseFalse],
            [2, 'smoke_level', tuya.valueConverter.raw],
            [11,'tamper', tuya.valueConverter.true0ElseFalse],
            [15,'battery', tuya.valueConverter.raw],
            [101,'test', tuya.valueConverter.raw]

I dont know how to test dp:11 (tamper) or fault but I guess that's a value thats not needed to check.
Guess you check if your test button reports correct value when you have time, if it does then there may be 2 types of this detector?

Jonno12345, 2022-12-26:
> 

So yeah, I've just confirmed this. Using 'raw' converts to a true/false value automatically based on the datapoint type, which in this case is a bool, so 'raw' works as intended. 'Smoke present' however is considered an enum so as raw would just pass through 0 or 1, 0 for smoke present, 1 for no smoke present, so the converter works as intended. Sorry for any confusion, I was testing the smoke presence primarily as honestly that was the important one for me 😁.

Unfortunately I have no idea how to test the device fault either, presumably I'd have to break the power to the photoelectric sensor in some way but I haven't got one I can try that with (or certain it would actually work).

If you're happy with the below on your device, I'll submit as a PR?

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const numeric = exposes.numeric;

const definition = {
fingerprint: [ { modelID: 'TS0601', manufacturerName: '_TZE200_m9skfctm'},],
model: 'TS0601_new',
vendor: 'TuYa',
description: 'Photoelectric Smoke Detector',
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
onEvent: tuya.onEventSetTime,
configure: tuya.configureMagicPacket,
exposes: [e.smoke(), numeric('smoke_level', ea.STATE).withUnit("ppm").withDescription("Parts per million of smoke detected"), e.tamper(), e.battery(), e.test()],
meta: {
    tuyaDatapoints: [
            [1, 'smoke', tuya.valueConverter.true0ElseFalse],
            [2, 'smoke_level', tuya.valueConverter.raw],
            [11,'tamper', tuya.valueConverter.true0ElseFalse],
            [15,'battery', tuya.valueConverter.raw],
            [101,'test', tuya.valueConverter.raw]
        ],
    },
};

module.exports = definition;
```

kjeisa, 2022-12-27:
@Jonno12345 Okay, yes just submit that now looks good 😁

seesrza, 2022-12-27:
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const e = exposes.presets;
> const ea = exposes.access;
> const numeric = [...]

with this converter gives such errors

Info 2022-12-27 19:50:43MQTT publish: topic 'zigbee2mqtt/Датчик дыма кухня', payload '{"battery":88,"linkquality":52,"smoke":false,"smoke_level":28,"tamper":false,"test":false}'
Debug 2022-12-27 19:50:43Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,88],"type":"Buffer"},"datatype":2,"dp":15}],"seq":42496}' from endpoint 1 with groupID 0
Info 2022-12-27 19:50:43MQTT publish: topic 'zigbee2mqtt/Датчик дыма кухня', payload '{"battery":88,"linkquality":72,"smoke":false,"smoke_level":28,"tamper":false,"test":false}'
Debug 2022-12-27 19:50:43Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":101}],"seq":42752}' from endpoint 1 with groupID 0

Jonno12345, 2022-12-27:
> with this converter gives such errors
>
> Info 2022-12-27 19:50:43MQTT publish: topic 'zigbee2mqtt/Датчик дыма кухня', payload [...]

Whats the error? This is just an info and debug output from waht I can see which look perfectly normal?

seesrza, 2022-12-27:
> 

Good afternoon! I'm new to this, so I'm wondering if this is not a bug?:
Debug 2022-12-27 20:40:21Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,91],"type":"Buffer"},"datatype":2,"dp":15}],"seq":52480}' from endpoint 1 with groupID 0

Jonno12345, 2022-12-27:
> Good afternoon! I'm new to this, so I'm wondering if this is not a bug?: Debug 2022-12-27 20:40:21Received Zigbee message from 'Датчик дыма кухня', type 'commandDataReport', cluster [...]

That's just the smoke alarm reporting a battery level of 91, this is before the converter has made it into something usable by Z2M, so no, not an issue. You've got debug logging turned on so you'll see all the raw ZigBee messages coming in.

seesrza, 2022-12-27:
> > > 
> > 
> > 
> > Добрый день! Я новичок в этом, поэтому мне интересно, не ошибка ли это?: Отладка 2022-12-27 20:40:21Получено сообщение Zigbee от «Датчик дыма кухня», тип «commandDataReport», кластер «manuSpecificTuya», data '{"dpValues":[{"data":{"data":[0,0,0,91],"type":"Buffer"},"datatype":2,"dp":15}] ,"seq":52480}' из конечной точки 1 с идентификатором группы 0
> 
> Это просто дымовая сигнализация, сообщающая об уровне заряда батареи 91, это до того, как преобразователь превратил ее во что-то, пригодное для использования Z2M, так что нет, не проблема. У вас включено ведение журнала отладки, поэтому вы будете видеть все поступающие необработанные сообщения ZigBee.

Thank you so much! Now I understand.

Regarding tamper - I opened the sensor, it physically does not have such a function.

Jonno12345, 2022-12-27:
> Thank you so much! Now I understand.
>
> Regarding tamper - I opened the sensor, it physically does not have such a function.

Yeah this should actually be 'device_fault'. I'll review whether I can get this one working correctly but I have no way to actually test it as I don't know what it considers to be a fault. I'd assume if it tries to pulse the sensor but it doesn't work, however I don't have one to sacrifice to test this.

Jonno12345, 2022-12-28:
I have submitted a PR at https://github.com/Koenkk/zigbee-herdsman-converters/pull/5218 now, with the correct value for 'device_fault'. As said before I can't test it properly without damaging one, but I've identified the correct datapoint now.

kjeisa, 2022-12-28:
Nice work:)

closed 2023-02-04 by github-actions[bot]

DennisJohnsen, 2023-02-07:
@Jonno12345 
Hey.
I just received 3 of these smoke detectors, and hoped that i would be able to link them together with NodeRed (i.e. if one goes off i can send a state to the others to make them be on too).

Do you have any chance of figuring out if we're missing features for this device in order to use `/set` to "trigger" the smoke detector? Ala what you see here: https://www.zigbee2mqtt.io/devices/SD-8SCZBS.html#warning-composite or https://www.zigbee2mqtt.io/devices/SA12IZL.html#alarm-binary

doreks, 2023-10-23:
any progress with this issue? I have peering smoke detector but not all values detected. Here the cnverted used:

`const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const numeric = exposes.numeric;

const definition = {
fingerprint: [ { modelID: 'TS0601', manufacturerName: '_TZE200_m9skfctm'},],
model: 'TS0601_new',
vendor: 'TuYa',
description: 'Photoelectric Smoke Detector',
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
onEvent: tuya.onEventSetTime,
configure: tuya.configureMagicPacket,
exposes: [e.smoke(), numeric('smoke_level', ea.STATE).withUnit("ppm").withDescription("Parts per million of smoke detected"), e.tamper(),
 e.battery(), e.test()],
meta: {
tuyaDatapoints: [
            [1, 'smoke', tuya.valueConverter.true0ElseFalse],
            [2, 'smoke_level', tuya.valueConverter.raw],
            [11,'tamper', tuya.valueConverter.true0ElseFalse],
            [15,'battery', tuya.valueConverter.raw],
            [101,'test', tuya.valueConverter.true0ElseFalse]
        ],
    },
};

module.exports = definition;`

and logs with error:

`info  2023-10-23 11:36:21: MQTT publish: topic 'zigbee2mqtt-herobox/0xa4c138aa659be6ef', payload '{"battery":94,"device":{"applicationVersion":72,"dateCode":"","friendlyName":"0xa4c138aa659be6ef","hardwareVersion":1,"ieeeAddr":"0xa4c138aa659be6ef","manufacturerID":4417,"manufacturerName":"_TZE200_m9skfctm","model":"TS0601_new","networkAddress":28371,"powerSource":"Battery","stackVersion":0,"type":"EndDevice","zclVersion":3},"linkquality":108,"smoke":null,"smoke_level":0,"tamper":null,"test":null}'`

any idea?