Koenkk/zigbee2mqtt #15350: Can't connect to MQTT broker after updating to 1.28.4

rjmarques, 2022-12-04:
### What happened?

After updating to `1.28.4` the service stopped being able to connect to the MQTT broker. Both `zigbee2mqtt` and the broker (`mosquitto`) are containers on the same docker network, and so far `zigbee2mqtt` had always been able to reach `mqtt://localhost:1883` without issues.

### What did you expect to happen?

It being able to connect to the MQTT broker as usual, as nothing has changed in the config anywhere.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20211210

### Adapter

CS2531

### Debug log

```
Zigbee2MQTT:info  2022-12-04 13:25:30: Connecting to MQTT server at mqtt://localhost
Zigbee2MQTT:debug 2022-12-04 13:25:30: Using MQTT login with username: root
Zigbee2MQTT:error 2022-12-04 13:25:30: MQTT error: connect ECONNREFUSED ::1:1883
Zigbee2MQTT:error 2022-12-04 13:25:30: MQTT failed to connect, exiting...
Zigbee2MQTT:info  2022-12-04 13:25:30: Stopping zigbee-herdsman...
Zigbee2MQTT:info  2022-12-04 13:25:31: Stopped zigbee-herdsman
```

closed 2022-12-04 by rjmarques

rodrigogbs, 2022-12-04:
same here:

```
[10:45:52] INFO: Preparing to start...
[10:45:53] INFO: Socat not enabled
[10:45:55] INFO: Starting Zigbee2MQTT...
/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
```

GsakuL, 2022-12-11:
i'm sorry. I don't want to "bump necros". But just as an FYI for someone finding this, if it's not too late:
if you configure your Broker yourself (like with HA Container), you could configure it to listen to IPv6.
eg mosquitto:
```txt
listener 1883 ::1
```