Koenkk/zigbee2mqtt #15351: Add pushbutton functionality for

cromefire, 2022-12-04:
### What happened?

Got a [Kobold](https://www.zigbee2mqtt.io/devices/BN-600110.html) which also [claims to support](https://phoscon.de/kobold/img/kobold_case1.jpg) pushbutton functionality (`action` ?), but the current converter doesn't taking advantage of that and only supports the output. I've tried adding it myself, but I failed at that

### What did you expect to happen?

The pushbutton functionality to be available through zigbee2mqtt.

### How to reproduce it (minimal and precise)

Just connect one or have a look on the code (in herdsman).

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Debug log

debug 2022-12-04 14:35:53: Loaded state from file /opt/zigbee2mqtt/data/state.json
info  2022-12-04 14:35:53: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-12-04.14-35-53' filename: log.txt
info  2022-12-04 14:35:53: Starting Zigbee2MQTT version 1.28.0 (commit #9b9f200)
info  2022-12-04 14:35:53: Starting zigbee-herdsman (0.14.62)
debug 2022-12-04 14:35:53: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
info  2022-12-04 14:35:57: zigbee-herdsman started (resumed)
info  2022-12-04 14:35:57: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":114,"product":0,"revision":"0x26720700","transportrev":0},"type":"ConBee2/RaspBee2"}'
debug 2022-12-04 14:35:57: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
info  2022-12-04 14:35:57: Currently 1 devices are joined:
info  2022-12-04 14:35:57: test-sw (0x00212effff0a6436): BN-600110 - Dresden Elektronik Zigbee 3.0 dimm actuator (Router)
warn  2022-12-04 14:35:57: `permit_join` set to  `true` in configuration.yaml.
warn  2022-12-04 14:35:58: Allowing new devices to join.
warn  2022-12-04 14:35:58: Set `permit_join` to `false` once you joined all devices.
info  2022-12-04 14:35:58: Zigbee: allowing new devices to join.
info  2022-12-04 14:35:59: Connecting to MQTT server at mqtts://<censored>:1883
debug 2022-12-04 14:35:59: MQTT SSL/TLS: Path to CA certificate = /opt/zigbee2mqtt/data/ca.crt
debug 2022-12-04 14:35:59: Using MQTT login with username: zigbee-coordinator
debug 2022-12-04 14:35:59: Using MQTT client ID: 'zigbee-coordinator'
info  2022-12-04 14:36:00: Connected to MQTT server
info  2022-12-04 14:36:00: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
debug 2022-12-04 14:36:00: Received MQTT message on 'zigbee2mqtt/bridge/info' with data '{"commit":"9b9f200","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"last_seen":"disable","legacy_api":true,"legacy_availability_payload":true,"log_directory":"/opt/zigbee2mqtt/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_symlink_current":false,"log_syslog":{},"output":"json","pan_id":6754,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"blocklist":[],"device_options":{},"devices":{"0x00212effff0a6436":{"friendly_name":"test-sw"}},"external_converters":[],"frontend":{"host":"localhost","port":8080,"url":"https://<censored>"},"groups":{},"homeassistant":{"discovery_topic":"homeassistant/discovery","legacy_entity_attributes":false,"legacy_triggers":false,"status_topic":"homeassistant/status"},"map_options":{"graphviz":{"colors":{"fill":{"coordinator":"#e04e5d","enddevice":"#fff8ce","router":"#4ea3e0"},"font":{"coordinator":"#ffffff","enddevice":"#000000","router":"#ffffff"},"line":{"active":"#009900","inactive":"#994444"}}}},"mqtt":{"base_topic":"zigbee2mqtt","ca":"/opt/zigbee2mqtt/data/ca.crt","client_id":"zigbee-coordinator","force_disable_retain":false,"include_device_information":true,"server":"mqtts://<censored>:1883","user":"zigbee-coordinator","version":5},"ota":{"disable_automatic_update_check":false,"update_check_interval":1440},"passlist":[],"permit_join":true,"serial":{"disable_led":false,"port":"/dev/ttyACM0"}},"config_schema":{"definitions":{"device":{"properties":{"debounce":{"description":"Debounces messages of this device","title":"Debounce","type":"number"},"debounce_ignore":{"description":"Protects unique payload values of specified payload properties from overriding within debounce time","examples":["action"],"items":{"type":"string"},"title":"Ignore debounce","type":"array"},"filtered_attributes":{"description":"Filter attributes with regex from published payload.","examples":["^temperature$","^battery$","^action$"],"items":{"type":"string"},"title":"Filtered publish attributes","type":"array"},"filtered_cache":{"description":"Filter attributes with regex from being added to the cache, this prevents the attribute from being in the published payload when the value didn't change.","examples":["^input_actions$"],"items":{"type":"string"},"title":"Filtered attributes from cache","type":"array"},"filtered_optimistic":{"description":"Filter attributes with regex from optimistic publish payload when calling /set. (This has no effect if optimistic is set to false).","examples":["^color_(mode|temp)$","color"],"items":{"type":"string"},"title":"Filtered optimistic attributes","type":"array"},"friendly_name":{"description":"Used in the MQTT topic of a device. By default this is the device ID","readOnly":true,"title":"Friendly name","type":"string"},"homeassistant":{"properties":{"name":{"description":"Name of the device in Home Assistant","title":"Home Assistant name","type":"string"}},"title":"Home Assistant","type":["object","null"]},"icon":{"description":"The user-defined device icon for the frontend. It can be a link to an image (not a path to a file) or base64 encoded data URL like: image/svg+xml;base64,PHN2ZyB3aW....R0aD","title":"Icon","type":"string"},"optimistic":{"default":true,"description":"Publish optimistic state after set","title":"Optimistic","type":"boolean"},"qos":{"description":"QoS level for MQTT messages of this device","title":"QoS","type":"number"},"retain":{"description":"Retain MQTT messages of this device","title":"Retain","type":"boolean"},"retention":{"description":"Sets the MQTT Message Expiry in seconds, Make sure to set mqtt.version to 5","title":"Retention","type":"number"}},"required":["friendly_name"],"type":"object"},"group":{"properties":{"devices":{"items":{"type":"string"},"type":"array"},"filtered_attributes":{"items":{"type":"string"},"type":"array"},"friendly_name":{"type":"string"},"off_state":{"default":"auto","description":"Control when to publish state OFF for a group. 'all_members_off': only publish state OFF when all group memebers are in state OFF, 'last_member_state': publish state OFF whenever one of its members changes to OFF","enum":["all_members_off","last_member_state"],"requiresRestart":true,"title":"Group off state","type":["string"]},"optimistic":{"type":"boolean"},"qos":{"type":"number"},"retain":{"type":"boolean"}},"required":["friendly_name"],"type":"object"}},"properties":{"advanced":{"properties":{"adapter_concurrent":{"description":"Adapter concurrency (e.g. 2 for CC2531 or 16 for CC26X2R1) (default: null, uses recommended value)","requiresRestart":true,"title":"Adapter concurrency","type":["number","null"]},"adapter_delay":{"description":"Adapter delay","requiresRestart":true,"title":"Adapter delay","type":["number","null"]},"cache_state":{"default":true,"description":"MQTT message payload will contain all attributes, not only changed ones. Has to be true when integrating via Home Assistant","title":"Cache state","type":"boolean"},"cache_state_persistent":{"default":true,"description":"Persist cached state, only used when cache_state: true","title":"Persist cache state","type":"boolean"},"cache_state_send_on_startup":{"default":true,"description":"Send cached state on startup, only used when cache_state: true","title":"Send cached state on startup","type":"boole [... cut ...]
info  2022-12-04 14:42:42: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"xt728-1"}'
debug 2022-12-04 14:42:44: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"xt728-2","value":true}'
info  2022-12-04 14:42:44: Zigbee: allowing new devices to join.
info  2022-12-04 14:42:45: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"xt728-2"}'

cromefire, 2022-12-04:
What I've tried so far is reading `OnOff` of endpoint 2, but it's apparently unsupported, so there's probably some other attribute that has to be read.

Danit2, 2022-12-04:
Can you show your external converter? Or what have you alrady tried?

cromefire, 2022-12-05:
Sadly I think I discarded my converter, because it wasn't working, but I tried changing the converter to be similar to this: https://github.com/Koenkk/zigbee-herdsman-converters/blob/2636217a818d08ec3a2ec2fe4c84a9c1afa99041/devices/sunricher.js#L189 (with endpoint set to 2)

And also tried to use the Dev Console to manually read stuff. I did also create that binding manually, but I'm not sure where I would even check if that binding is working and doing something? Sorry I'm just starting with Zigbee and haven't got every feature yet.

Danit2, 2022-12-05:
Can you try this external converter:

```
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const herdsman = require('zigbee-herdsman');

module.exports = [
    {
        zigbeeModel: ['Kobold'],
        model: 'BN-600110',
        vendor: 'Dresden Elektronik',
        description: 'Zigbee 3.0 dimm actuator',
        fromZigbee: [fz.on_off, fz.brightness, fz.level_config, fz.ignore_basic_report, fz.command_toggle, fz.command_move, fz.command_stop],
        toZigbee: [tz.light_onoff_brightness, tz.ignore_rate, tz.level_config, tz.light_brightness_move, tz.light_brightness_step],
        exposes: [e.action(['toggle', 'brightness_move_up', 'brightness_move_down', 'brightness_stop']),
            e.light_brightness(), 
    },
];
```

and for the binding it must look something like this  in the UI:
[image]

to check if it works, you can press the pushbutton and the remote light must go on and off.

cromefire, 2022-12-05:
[image]

Had to slightly modify it and it now does toggle itself, I don't seem to receive anything in home assistant or zigbee2mqtt though...

Danit2, 2022-12-05:
Yes it looks good. Now you have bound the push button to de device directly. 

Do you use my external converter? 
Then you have now a action expose. 
Like this: 
[image]

This you can use in home assistant or node-red.

cromefire, 2022-12-05:
Okay that works now, but it automatically does brightness up/down and does not work with a pushbutton or switch, where it should normally send a trigger or on/off signal. Is it possible to like parameterize that, so I can upstream it to herdsman and it's working like a generic button would?

Danit2, 2022-12-06:
Sorry this i dont understand. 
I dont know what you meen with this text.

cromefire, 2022-12-06:
Current behavior:
Short push -> no action

Long push start -> brightness_move_up
Long push stop -> brightness_stop
Long push start -> brightness_mode_down
Long push stop -> brightness_stop

Behavior I'd expect from a button:
Short push -> toggle

Long push start -> on
Long push stop -> off

Or is this just an inherent limitation of Zigbee and the button just works at a light button? I my case for example I don't want to switch the light directly with the switch, but I want to route it through some home assistant/NodeRED logic

cromefire, 2022-12-08:
> This what you want is not possible with your device.
> because this is defined in the firmware of the device.
> 
> There are devices that do this the way you want it. but not this one.
> 
> Usually it's always like you described:
> Short push -> toggle
> Long push start -> brightness_move_up
> Long push stop -> brightness_stop
> Long push start -> brightness_mode_down
> Long push stop -> brightness_stop
> 
> but if you let it run via logic, the command doesn't matter.

Yeah okay, then I'll just translate it in NodeRed.

> are you sure that there was no toggle action during the short press?

I couldn't see one in the zigbee2mqtt UI at least maybe it's too short to show? Maybe I have to check home assistant/the mqtt bus. Toggle did work when directly connected to the light...

cromefire, 2023-01-08:
Bump, still have to test it again, just haven't come around to it yet.

closed 2023-02-16 by github-actions[bot]