Koenkk/zigbee2mqtt #15359: [New device support]:  tuya/Matsee plus current clamp meter

uowis, 2022-12-04:
### Link

https://fr.aliexpress.com/item/1005004948381800.html?spm=a2g0o.productlist.0.0.31a631410zhHLp&algo_pvid=def1e34d-7ff1-4539-b467-eeeaa6c164ba&algo_exp_id=def1e34d-7ff1-4539-b467-eeeaa6c164ba-12&pdp_ext_f=%7B%22sku_id%22%3A%2212000031119654070%22%7D&pdp_npi=2%40dis%21EUR%2157.81%2127.17%21%21%21%21%21%402100bb4a16701892996707394e5162%2112000031119654070%21sea&curPageLogUid=kahBvYoaamVy

### Database entry

{"id":27,"type":"Router","ieeeAddr":"0xe0798dfffede3b6d","nwkAddr":25319,"manufId":4098,"manufName":"_TZE204_cjbofhxw","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000f\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_cjbofhxw","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00212effff07b952","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1670193766558,"defaultSendRequestWhen":"immediate"}

### Comments

 fingerprint: tuya.fingerprint('TS0601', ['_TZE204_cjbofhxw']),

I have try to make an external converter but i have this error :
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;
const {Buffer} = require('buffer');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_cjbofhxw']),
    model: 'TS0601_1_phase_clamp_meter',
    vendor: 'TuYa',
    description: '1 phase clamp power meter',
    fromZigbee: [tuya.fzDataPoints],
    toZigbee: [tuya.tzDataPoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [],
    meta: {
        tuyaDatapoints: [
           
        ],
    },
};

module.exports = definition;

```

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;
const {Buffer} = require('buffer');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_cjbofhxw']),
    model: 'TS0601_1_phase_clamp_meter',
    vendor: 'TuYa',
    description: '1 phase clamp power meter',
    fromZigbee: [tuya.fzDataPoints],
    toZigbee: [tuya.tzDataPoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [],
    meta: {
        tuyaDatapoints: [
           
        ],
    },
};

module.exports = definition;
```

Clem-, 2022-12-05:
Hi, I've just received the same device as yours and didn't manage to make it works by using an existing Tuya Meter device adapter.

If this can help :
```
[9,114,37,170,0] // data when clamp **not** connected on live wire
[9,58,37,56,0] // data when current running trough clamp:
```

lugiis, 2022-12-05:
https://github.com/Koenkk/zigbee2mqtt/issues/14790 
It's still useful, it's from the same manufacturer

Clem-, 2022-12-05:
See also [this issue](https://github.com/zigpy/zha-device-handlers/issues/1973) from zigpy/zha for the same device.

mrbluebrett, 2022-12-05:
I received the same today. But i have no tuya zigbee gateway to help with the endpoints… :(

Clem-, 2022-12-06:
Seller has provided those information:

> //current(only report)
> //备注:
> #define DPID_CUR_CURRENT 18
> //power(only report)
> //备注:
> #define DPID_CUR_POWER 19
> //voltage(only report)
> //备注:
> #define DPID_CUR_VOLTAGE 20
> // add electric (only report)
> //备注:
> #define DPID_ELECTRIC_ADD 101

And also a link to this [blog post](https://blog.ravindra-job.com/iot/tuya_meter/).

uowis, 2022-12-06:
> [9,114,37,170,0] // data when clamp **not** connected on live wire
> [9,58,37,56,0] // data when current running trough clamp:

I have same value in [9,xx,37, xx, 00] 9 and 37 are static and the xx change every time,... i thing this array have no the value.

Clem, after device apaired ? Is the red light stay on ?
and in Zigbee2mqtt log have could find this error :
```
zigbee-herdsman:deconz:frameParser response payload: 9,32,37,44,0
2022-12-06T08:40:26.827Z zigbee-herdsman:deconz:adapter could not parse zclFrame: Error: Cluster 'manuSpecificTuya' has no command response '37'
```

Clem-, 2022-12-06:
Light is steady red after pairing, yes.
I have the same log error.

What is weird is that there isn't any Datapoint reported in the log. Could it be due to a bad pairing of the device?

uowis, 2022-12-06:
I could try many converter model like this one https://github.com/Koenkk/zigbee2mqtt/issues/14790
or this [blog](https://blog.ravindra-job.com/iot/tuya_meter/)
I have ever the same error and the same static value  energy : 9 and power : 37

uowis, 2022-12-06:
I'm a newbie and i don"t understand all detail but it seems to be a similar issue here :
#14519

lugiis, 2022-12-06:
One question since you're talking to Sophie de Matsee, she can't provide us with the necessary information about you. I don't have a port compatible with tuya either.

guillermohor, 2022-12-06:
Dame device,same problem here.
'_TZE204_cjbofhxw' with cluster 'manuSpecificTuya' and type 'raw' and data '{"data":[9,96,37,49,0],"type":"Buffer"}'

Koenkk, 2022-12-06:
Please check with https://gist.github.com/Koenkk/8fc146f998e2bd9986df783164f4f3df (requires z2m 1.28.4). Double check that the reported values are correct (e.g. could be possible that they have to be divided by 100)

guillermohor, 2022-12-06:
current
Instantaneous measured electrical current
N/AA
power
Instantaneous measured power
N/AW
voltage
Measured electrical potential value
N/AV
linkquality
Link quality (signal strength)
255lqi

guillermohor, 2022-12-06:
Zigbee2MQTT:debug 2022-12-06 18:54:37: Received Zigbee message from 'Medidor', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,100,37,251,0],"type":"Buffer"}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2022-12-06 18:54:37: No converter available for 'TS0601_clamp_meter' with cluster 'manuSpecificTuya' and type 'raw' and data '{"data":[9,100,37,251,0],"type":"Buffer"}'

guillermohor, 2022-12-06:
[image]

guillermohor, 2022-12-06:
[image]

Clem-, 2022-12-06:
It might be due to something going wrong during the pairing, like an operation missing in the `configure` attribute?

The red light after pairing isn’t right according to the seller and the only message sent by the device seems to be something like a clock or heartbeat.

mrbluebrett, 2022-12-07:
Same here...
Debug 2022-12-07 10:11:52Received Zigbee message from '0xe0798dfffede41bc', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,110,37,98,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug 2022-12-07 10:11:52No converter available for 'TS0601_clamp_meter' with cluster 'manuSpecificTuya' and type 'raw' and data '{"data":[9,110,37,98,0],"type":"Buffer"}'

In the log it only shows endpoint 1. Light stays red... so i guess it's not completely paired...

uowis, 2022-12-07:
> Please check with https://gist.github.com/Koenkk/8fc146f998e2bd9986df783164f4f3df (requires z2m 1.28.4). Double check that the reported values are correct (e.g. could be possible that they have to be divided by 100)

```
2022-12-07T10:34:25.556Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-12-07T10:34:25.727Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2022-12-07T10:34:25.727Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-12-07T10:34:25.729Z zigbee-herdsman:deconz:driver query aps data indication
2022-12-07T10:34:25.734Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 16
2022-12-07T10:34:25.738Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 16 srcAddr: 0x3384 destAddr: 0x0 profile id: 0x104 cluster id: 0xef00 lqi: 255
2022-12-07T10:34:25.739Z zigbee-herdsman:deconz:frameParser response payload: 9,12,37,111,0
2022-12-07T10:34:25.740Z zigbee-herdsman:deconz:adapter could not parse zclFrame: Error: Cluster 'manuSpecificTuya' has no command response '37'
2022-12-07T10:34:25.742Z zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":61184,"data":{"type":"Buffer","data":[9,12,37,111,0]},"address":13188,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2022-12-07 11:34:25: Received Zigbee message from '0xe0798dfffede3b6d', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,12,37,111,0],"type":"Buffer"}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2022-12-07 11:34:25: No converter available for 'TS0601_clamp_meter' with cluster 'manuSpecificTuya' and type 'raw' and data '{"data":[9,12,37,111,0],"type":"Buffer"}'
```

uowis, 2022-12-07:
Is it possible that device need some 'response' to finalise apairing ?

Koenkk, 2022-12-07:
Can you provide a herdsman debug log when receiving a message from this device?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

neooffline, 2022-12-07:
@Koenkk I sent log with herdsman debug on in message [tap](https://github.com/Koenkk/zigbee2mqtt/issues/15359#issuecomment-1339805026)

uowis, 2022-12-08:
> Pouvez-vous fournir un journal de débogage Herdsman lorsque vous recevez un message de cet appareil ?
> 
> Voir https://www.zigbee2mqtt.io/guide/usage/debug.html pour savoir comment activer la journalisation de débogage du bouvier. Notez que cela n'est enregistré que dans STDOUT et non dans les fichiers journaux.

STDOUT with herdsman debug on : 
```
Debug 2022-12-08 11:54:54Received Zigbee message from '0xe0798dfffede3b6d', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,89,37,205,0],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2022-12-08 11:54:54No converter available for 'TS0601_clamp_meter' with cluster 'manuSpecificTuya' and type 'raw' and data '{"data":[9,89,37,205,0],"type":"Buffer"}'
Debug 2022-12-08 11:54:58Received Zigbee message from '0xe0798dfffede3b6d', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,90,37,207,0],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2022-12-08 11:54:58No converter available for 'TS0601_clamp_meter' with cluster 'manuSpecificTuya' and type 'raw' and data '{"data":[9,90,37,207,0],"type":"Buffer"}'
Debug 2022-12-08 11:54:58Received Zigbee message from '0xe0798dfffede3b6d', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,90,37,207,0],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2022-12-08 11:54:58No converter available for 'TS0601_clamp_meter' with cluster 'manuSpecificTuya' and type 'raw' and data '{"data":[9,90,37,207,0],"type":"Buffer"}'
```

uowis, 2022-12-08:
wow i have recieve this message from sophie : 

> This device requires it must connect with cloud.

> The MCU polls the value periodically. Only when the value feedback is normal then can  be considered the pairing successful, and this indicates whether the gateway is connected to the cloud

> Only after the mcu confirms that the gateway is connected to the cloud, can it report the dp id normally. Otherwise, it is saved on the local device

`#define    CHACKE8ZIGBEE8GATEWAY8STATUS8CMD    0x25`

uowis, 2022-12-08:
i don"'t know,... perahps is it tz.datapoint ?
I try to read this doc : 
https://developer.tuya.com/en/docs/iot/zigbee-door-lock-universal-docking-agreement?id=K9bjcnh616mg5
But it's 'foggy' for me

Koenkk, 2022-12-08:
I see the `CHECK_ZIGBEE_GATEWAY_STATUS_CMD 0x25` command in the log from @guillermohor 

```
2022-12-08T10:25:47.194Z zigbee-herdsman:deconz:adapter could not parse zclFrame: Error: Cluster 'manuSpecificTuya' has no command response '37'
```

But we need to know what to send back, can somebody sniff the traffic when pairing this device with the original TuYa gateway?

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

uowis, 2022-12-08:
i don't have tuya gateway, i ask it to Sophie.. 
I see this is why we have heartbeats vaule in the arraw :
https://developer.tuya.com/en/docs/iot/advanced-features?id=Kav0qqgzfj95f#title-27-Commands
https://developer.tuya.com/en/docs/iot/tuya-zigbee-module-uart-communication-protocol?id=K9ear5khsqoty#title-32-MCU%20requests%20gateway%20connection%20status%20(0x25)

guillermohor, 2022-12-08:
> i don't have tuya gateway, i ask it to Sophie.. I see this is why we have heartbeats vaule in the arraw : https://developer.tuya.com/en/docs/iot/advanced-features?id=Kav0qqgzfj95f#title-27-Commands [...]

Is it not enought information ? Do we still need to sniff traffic with the tuya gateway?

uowis, 2022-12-08:
we need to know the sequence number ?
how we can send it ? in the converter file ?

guillermohor, 2022-12-08:
Let's wait to @Koenkk . Perhaps  a function in the z2mqtt code could do it.

mrbluebrett, 2022-12-08:
> I see the `CHECK_ZIGBEE_GATEWAY_STATUS_CMD 0x25` command in the log from @guillermohor
>
> ```
> 2022-12-08T10:25:47.194Z zigbee-herdsman:deconz:adapter could not parse zclFrame: Error: Cluster [...]

This is the documentation in the link:
<p style="font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none; box-sizing: border-box; padding: 0px; margin-block: 0px; border: 0px; font-family: &quot;Chinese Quote&quot;, -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, Roboto, &quot;PingFang SC&quot;, &quot;Hiragino Sans GB&quot;, &quot;Microsoft YaHei&quot;, &quot;Helvetica Neue&quot;, Helvetica, Arial, sans-serif; margin: 16px 0px 0px; color: rgb(96, 98, 102); font-size: 14px; line-height: 24px; -webkit-tap-highlight-color: rgba(0, 0, 0, 0); -webkit-text-size-adjust: 100%;"><strong style="box-sizing: border-box; margin: 0px; padding: 0px; font-weight: 700; border: 0px; font-family: &quot;Chinese Quote&quot;, -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, Roboto, &quot;PingFang SC&quot;, &quot;Hiragino Sans GB&quot;, &quot;Microsoft YaHei&quot;, &quot;Helvetica Neue&quot;, Helvetica, Arial, sans-serif;">The module returns the following data.</strong></p>

Field | Bytes | Description
-- | -- | --
Header | 2 | 0x55aa
Version | 1 | 0x02
Sequence number | 2 | N
Command | 1 | 0x025
Data length | 2 | 0x0001
Data | 1 | 0x00: The gateway is not connected to the internet.0x01: The gateway is connected to the internet.0x02: The request timed out after three seconds.
Checksum | 1 | Start from the header, add up all the bytes, and then divide the sum by 256 to get the remainder.

<p style="font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none; box-sizing: border-box; padding: 0px; margin-block: 0px; border: 0px; font-family: &quot;Chinese Quote&quot;, -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, Roboto, &quot;PingFang SC&quot;, &quot;Hiragino Sans GB&quot;, &quot;Microsoft YaHei&quot;, &quot;Helvetica Neue&quot;, Helvetica, Arial, sans-serif; margin: 16px 0px 0px; color: rgb(96, 98, 102); font-size: 14px; line-height: 24px; -webkit-tap-highlight-color: rgba(0, 0, 0, 0); -webkit-text-size-adjust: 100%;"><strong style="box-sizing: border-box; margin: 0px; padding: 0px; font-weight: 700; border: 0px; font-family: &quot;Chinese Quote&quot;, -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, Roboto, &quot;PingFang SC&quot;, &quot;Hiragino Sans GB&quot;, &quot;Microsoft YaHei&quot;, &quot;Helvetica Neue&quot;, Helvetica, Arial, sans-serif;">Example</strong></p><p style="font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none; box-sizing: border-box; padding: 0px; margin-block: 0px; border: 0px; font-family: &quot;Chinese Quote&quot;, -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, Roboto, &quot;PingFang SC&quot;, &quot;Hiragino Sans GB&quot;, &quot;Microsoft YaHei&quot;, &quot;Helvetica Neue&quot;, Helvetica, Arial, sans-serif; margin: 16px 0px 0px; color: rgb(96, 98, 102); font-size: 14px; line-height: 24px; -webkit-tap-highlight-color: rgba(0, 0, 0, 0); -webkit-text-size-adjust: 100%;"><code style="box-sizing: border-box; margin: 0px 4px; padding: 2px 4px; font-family: Menlo, Monaco, Consolas, &quot;Courier New&quot;, monospace; font-size: 12.6px; border: 0px; background-color: rgba(0, 0, 0, 0.08); border-radius: 2px; color: rgb(48, 49, 51);">55 aa 02 N 25 00 01 01 xx</code></p>

It seems we need to send back a 0x01 with all the other packet data, right?

uowis, 2022-12-09:
at first the zigbee module (device) send to gateway the request with the sequence number ? that right ?
So can we saw it with zigbee2mqtt ?

uowis, 2022-12-09:
me : "It's seems we need a sequence number that right ?   a four digit number ?"
sophie: "Actually, for MCU, it doesn't check this value right or not"

Somebody can tell us how send this return request ?

mrbluebrett, 2022-12-09:
I think that even if the sequence number is not checked, we can copy it from the request.
I don't know how to develop this though.

Koenkk, 2022-12-09:
Thanks for the links, I've did an attempt to add the command  and answer it accordingly.

- Update to the latest dev first
- Try https://gist.github.com/Koenkk/8fc146f998e2bd9986df783164f4f3df

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

kkossev, 2022-12-09:
Here is a temporary link for downloading the PCAP file , paired to Tuya GW : https://drive.google.com/file/d/1GDxJrCZVoItQGcIvaBH2qqlSykkXlbcP/view?usp=sharing 
filter by zbee_nwk.addr eq 0xddf7
Hope this will help.

I've made it work in Hubitat, but this device is rather chatty .. :(

neooffline, 2022-12-09:
It works, but values...
[image]

guillermohor, 2022-12-09:
Thanks to everyone!

mrbluebrett, 2022-12-09:
Cool! Probably energy also needs to be divided by 10.

guillermohor, 2022-12-10:
I had 100kwh in energy and after starting again 22. Can it go down?

guillermohor, 2022-12-10:
Energy is dividedby1000

uowis, 2022-12-10:
Same value for energy... State Can go down..

Koenkk, 2022-12-10:
Updated https://gist.github.com/Koenkk/8fc146f998e2bd9986df783164f4f3df

guillermohor, 2022-12-10:
Yes. There is a tab in ZigBee2mqtt for adjusting that. 
[image]

guillermohor, 2022-12-10:
Mine is a sensor.

mrbluebrett, 2022-12-10:
[image]
I mean in the diagnostics class.

lugiis, 2022-12-10:
Thank you very much now it is only 0.3 difference and I keep trying to make it perfect

guillermohor, 2022-12-10:
> [image] I mean in the diagnostics class.

You are right . They are these class to me too.

Koenkk, 2022-12-11:
That is expected behaviour, they are defined as such in https://github.com/Koenkk/zigbee2mqtt/blob/52e545f900c799377a69b1a991577c7ccdf641ec/lib/extension/homeassistant.ts#L684 (see https://github.com/Koenkk/zigbee2mqtt/pull/9293 for the rationale).

Can I add this device? (everything seems to work?)

uowis, 2022-12-11:
We don't know how interpret the energy value,...

mrbluebrett, 2022-12-11:
It looks like its sending delta values for energy.
I dont know if this can be fixed changing the last reset timestamp value to everytime it sends data.

Koenkk, 2022-12-11:
Maybe `energy` is send through a different datapoint, when activating the debug log, do you see any `Datapoint XX not defined for...`?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

mrbluebrett, 2022-12-11:
We can also fix this in home assistant with an utility meter with option delta set up.
But it would be nice have it fixed from origin.

uowis, 2022-12-11:
utilitymeter don"t make the job ,... : 
[image]

uowis, 2022-12-11:
But i'm agree,
a sum of the positive variation could perahps work

mrbluebrett, 2022-12-11:
Can you share how tou defined the utility meter?
Another option would be to create an integration sensor of the power entity, but it’s less precise.

uowis, 2022-12-11:
Can we make in the converter something like this : 
if (old_energy < new_energy)
  sum_energy= sum_energy + abs(new_energy - old_energy)

uowis, 2022-12-11:
In this case reinman integration seems to be more precise no ? The energy value don't update enough quickly : 
[image]
[image]

uowis, 2022-12-11:
I don't like Hass helper entry like the reimann integration because  they are not real entity 'device_class: energy'
perahps Zigbee2mqtt can make the integration ?

Koenkk, 2022-12-12:
I would expect there is a seperate `energy` datapoint (or this would be the first device that doesn't). To find it out: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

[62 more replies]

closed 2023-05-15 by github-actions[bot]