Koenkk/zigbee2mqtt #15361: [New device support]:

ArcadeMachinist, 2022-12-05:
### Link

https://www.control4.com/solutions/products/wireless-keypads/

### Database entry

{"id":7,"type":"Router","ieeeAddr":"0x000fff00002786f1","nwkAddr":9525,"manufId":43981,"epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[1,0],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{"genPowerCfg":{"attributes":{"2":300,"3":0,"4":"03.22.41","5":5,"6":1736,"7":"c4:control4_keypad:C4-KP6-Z","11":300,"12":11,"20":255,"mainsVoltage":2,"mainsFrequency":10,"mainsVoltageDwellTripPoint":-77,"mainsVoltMaxThres":{"type":"Buffer","data":[0,0]}}}},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1670212650553,"defaultSendRequestWhen":"immediate"}

### Comments

Control4 C4-KP6-Z 6 button mains powered keypad with individual RGB leds for each button.
Product page, linked above, does not contain this model anymore. eBay does.

[image]

This is Ember EM250 based device. 
It can come both with Embernet and ZPro firmware, so I specifically mention this one is ZPro.

While the keypad can join the network and pass interview, I can't get any button presses from it.
(While I can with Tasmota ZbBridge, allthru incorrectly parsed)

Join log:

> Device '0x000fff00002abcde' joined
> Starting interview of '0x000fff00002abcde'
> Device '0x000fff00002abcde' announced itself
> Received Zigbee message from '0x000fff00002abcde', type 'attributeReport', cluster 'genPowerCfg', data '{"4":"03.22.41","5":5,"6":1720,"7":"c4:control4_keypad:C4-KP6-Z"}' from endpoint 2 with groupID 0
> Received Zigbee message from '0x000fff00002abcde, type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
> Received Zigbee message from '0x000fff00002abcde', type 'attributeReport', cluster 'genPowerCfg', data '{"11":300,"12":11,"2":300,"20":255,"3":0,"mainsFrequency":10,"mainsVoltage":2,"mainsVoltageDwellTripPoint":-77}' from endpoint 2 with groupID 0
> Received Zigbee message from '0x000fff00002abcde', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
> Successfully interviewed '0x000fff00002abcde', device has successfully been paired

Looking at the database entry, I see endpoints 1 and 2, however from what I know from previous experiments - the keypress messages come with srcEndpoint: 197 (0xc5), dstEndpoint 197, cluster 0x0001, profile 0xc25c

My question how do I catch that with Z2M?
How to define and endpoint from the converter and make the coordinator process incoming data?

For now on keypress I see only these:

>   zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,53,37,1,127,152,114] +2s
>   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,53,37,1,127,152,114] +0ms
>   zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [53,37,1,127,152] - 114 +1ms
>   zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":9525,"relaycount":1,"relaylist":[39039]} +2

same without relaying

>   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
>   zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,53,37,0,146] +75ms
>   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,53,37,0,146] +2ms
>   zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [53,37,0] - 146 +1ms
>   zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":9525,"relaycount":0,"relaylist":[]} +81ms

If I setup a ZbBridge (with Tasmota fw) based coordinator - I do get the keypress messages too, they are then incorrectly parsed (first 2 bytes of payload are parsed as cmd_id), lacking custom converter, but at least they come to the level where I can see them.

Running original ZMAN from an old C4 controller box - I have collected all the data needed to parse button press info and to control LEDs, now just need to figure out how to get the payload data parseable with Z2M.

### External converter

```shell
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const c4Fz = {
        c4config: {
                cluster: 'genPowerCfg',
                type: ['attributeReport'],
                convert: (model, msg, publish, options, meta) => {
                        const result = {};
                        if (msg.data.hasOwnProperty('4')) {
                                result['update'] = {};
                                result['update']['installed_version'] = msg.data['4'];
                                result['update']['state'] = 'idle';
                        }
                        if (msg.data.hasOwnProperty('7')) {
                                console.log('## SET! #');
                                let c4model = msg.data['7'].split(':');
                                if (c4model[0] == 'c4') {
//                                      meta.device._manufacturerName = 'Control4';
//                                      meta.device._modelID = c4model[2];
                                }
                        }
                        return result;
                },
        },
        c4catchall: {
                cluster: 'genPowerCfg',
                type: 'raw',
                convert: (model, msg, publish, options, meta) => {
                        const result = {};
                        console.log('##### C4 CATCH ALL #############');
                        console.log(msg.data);
                },
        }
}
module.exports = [
    {
        fingerprint: [{
                type: "Router",
                manufId: 43981,
                ieeeAddr: "0x000fff00002abcde",
        }],
        model: 'C4-KP6',
        vendor: 'Control4',
        description: 'Control4 6 button keypad',
        supports: '',
        fromZigbee: [c4Fz.c4config, c4Fz.c4catchall],
        toZigbee: [],
        meta: {configureKey: 1, timeout: 5000},
        exposes: [],
    }
]
```

ArcadeMachinist, 2022-12-07:
Meanwhile some info for others, who might want to play with these switches:

Factory reset and leave previous mesh: 13-4-13
That means press upper left button 13 times, bottom left 4 times, upper left 13 times.
The keypad would reset and light all LEDs green.

Then "Identify" command can be used. 4 times upper left button.
Keypad would blink all LEDs yellow and then all LEDs would become blue, when it joins.

Then, given we would manage to write a converter:
It sends keypress data to endpoint 0xc5, profile 0xc25c, cluster 0x0001 with a hex-encoded ASCII string like:

> 0t3ab2 sa c4.kp.bb 01

Where "0t" is packet type and "3ab2" is a sequence number, which can stard with a random number, it is not verified in anyway, hust should not be the same between neighbouring packets.

0t1234 sa c4.kp.bb # = button # down 
0t1234 sa c4.kp.bc # = button # up immediately after "down"
0t1234 sa c4.kp.bh # = button # being held after "down"
0t1234 sa c4.kp.be # = button # released after hold

If the button was not held, but clicked once or more it would also send final number of clicks after last "up" event
0t1234 sa c4.kp.cc # # = detected button # clicked # times

Button numbering starts with 0, button numbers are sent (and expected to be received) with the leading "0", eg "00",  etc.

**Now for sending commands from coordinator to the keypad: **

Commands to keypad are sent out to profile 0xc25c, cluster 0x0001, endpoint 0x1, the payload is hex encoded ASCII string, like:
0i1257 c4.kp.rb
with "0i" being packet type, "257" sequence.
c4.kp.rb tells keypad to reboot, it would reply with "0r1257 000"

The following 3 commands should be send with "0s" packet type, eg. "0s1234 c4.kp.lv 01 ff00ff"

c4.kp.lv # aabbcc = set color for button # to aabbcc RGB value, any click resets this to the value set by c4.kp.lf
c4.kp.lf  # aabbcc = set "off" color for button # to aabbcc RGB value, when button is not depressed it would be in this color
c4.kp.lo  # aabbcc = set "on" color for button # to aabbcc RGB value, when button is depressed it would be in this color

The above 3 commands also be used to set color for all 6 buttons at once, example: "c4.kp.lo ff ff aabbcc"

There are also things like managed states for buttons, but I didn't get that far.
I can run sniff on the original mesh, if there would be interest.
I do not have a complete C4 setup, but I can send messages to the keypad using H250 shell and builtin ZMan utility.

ArcadeMachinist, 2022-12-10:
I have poked C4 thermostat (C4-WT100-WH), which is like $10 new on eBay now.
And it does follow the same pattern.
Data sent to controller from endpoint 0xc5, to endpoint 0xc5, profile 0xc25c, cluster 0x0001

I have not digged deep into zigbee-herdsman yet.
From what I understand I need to somehow write a custom processor for ZCL frames for C4 devices.

Can it be done in the "converter", or requires deeper code changes?

ArcadeMachinist, 2022-12-19:
Ok. Changes to the underlying **zigbee-herdsman** are required.

My adapter (Sonoff USB stick -P model) uses ZStack, so **dist/adapter/z-stack/adapter/endpoints.js** should be modified to include:
`{ ...EndpointDefaults, endpoint: 0xc5, appprofid: 0xc25c },`

Now I can see ZCL frames forwarded to Z2M and my "C4 CATCHALL" (from previous post) now gets the frame, which I can parse.

So Control4 keypads support is doable.
Would post when parser done.

ArcadeMachinist, 2022-12-19:
Paging @Koenkk 

Is there a way to send raw data to the device?
Like send custom raw data instead of whole ZCL frames.

What would be the best approach? 
What is the best way to override profileID (from default 0x0104)?

ArcadeMachinist, 2022-12-25:
Ok, I have done some more research, source code surfing and seems like it not possible to send custom profileID for endpoint 1 with TI's ZNP api (without breaking everything else for all other devices)

Ref: https://github.com/Koenkk/zigbee-herdsman-converters/commit/d0fb06c2429171f327950484ea3dec80864637cc#commitcomment-51640096

So it is impossible to send required commands with Ti's CC2652 based devices.

However it **seems to be very possible** with EZSP talking sticks, as the profileID is handled not by firmware, but EZSP driver in https://github.com/Koenkk/zigbee-herdsman/

I guess I'm getting Silicon Labs' based stick and coming back, or may be I have a Mighty Gecko board somewhere.

Koenkk, 2022-12-25:
This is currently not well supported indeed but should be possible since raw data is passed to the adapter functions (e.g. https://github.com/Koenkk/zigbee-herdsman/blob/eaf96968817d34bae4e984e2ee4e3b0e4a4e5e97/src/adapter/z-stack/adapter/zStackAdapter.ts#L389)

ArcadeMachinist, 2022-12-25:
> This is currently not well supported indeed but should be possible since raw data is passed to the adapter functions 

This solves only ZCL part of the puzzle, which I was able to overcome playing with reserved bits and options sent to extended Endpoint.write

However it seems that ProfileID, which is part of the APS layer, is handled by firmware and not arbitrary settable per call on TI's chips.

Koenkk, 2022-12-26:
You are right, TI ZNP api doesn't support this (checked the `AF_DATA_REQUEST_EXT`). It will always use the profile ID of the endpoint.

ArcadeMachinist, 2022-12-27:
I was able to make it work with EZSP talking stick (Sonoffv2).
However to include the converter into z2m - minor core changes are required to zigbee-herdsman, only you know best how to implement (my dirty hacks aren't good for production)

So I would just talk about changes I've made.
Converter:  [c4-kp6.zip](https://github.com/Koenkk/zigbee2mqtt/files/10311102/c4-kp6.zip)

Other questions I still have:
1. how to define 6 independent LEDs in converter and handle them?
2. is it possible to update DB with device model after converter receives (non standard) data from the device?

/adapter/ezsp/adapter/ezspAdapter.ts
in **sendZclFrameToEndpointInternal**
`if (ieeeAddr.includes("0x000fff0000")) frame.profileId = 0xC25C;`
this is pretty dirty and we probably want to do it in some other way, eg only when converter wants to specifically send Control4 ZCL frame

in /adapter/ezsp/adapter/endpoint.ts
introduced **writeC4**
this should be achieved by just extending normal write, instead new function 
also probably just an interface to send arbitrary buffer as ZCL frame.

`    async writeC4(clusterKey, xpayload, options) {
        var _a;
        const cluster = Zcl.Utils.getCluster(clusterKey);
        options = this.getOptionsWithDefaults(options, true, Zcl.Direction.CLIENT_TO_SERVER, cluster.manufacturerCode);
        const log = `WriteC4 ${this.deviceIeeeAddress}/${this.ID} `;
        debug.info(log);
        try {
            const payload = [];
            xpayload = xpayload.slice(3);
            payload.push({ attrId: 0, attrData: xpayload, dataType:'charStr'});
            const frame = Zcl.ZclFrame.create(Zcl.FrameType.GLOBAL,0,1,null,0x73,"c4cmd_s",0x0001,payload,0x1);

            const result = await this.sendRequest(async () => {
                return entity_1.default.adapter.sendZclFrameToEndpoint(this.deviceIeeeAddress, this.deviceNetworkAddress, this.ID, frame, options.timeout, options.disableResponse, options.disableRecovery, options.srcEndpoint);
            }, options.sendWhen);
            if (!options.disableResponse) {
                this.checkStatus(result.frame.Payload);
            }
        }
        catch (error) {
            error.message = `${log} failed (${error.message})`;
            debug.error(error.message);
            throw error;
        }
    }`

in /zcl/definition/foundation.js
`c4cmd_s: {
        ID: 49,
        parseStrategy: 'repetitive',
        parameters: [
            { name: 'attrData', type: buffaloZclDataType_1.default.USE_DATA_TYPE },
        ]
    }`

Koenkk, 2022-12-28:
> how to define 6 independent LEDs in converter and handle them?

Can you clarify this further?

> is it possible to update DB with device model after converter receives (non standard) data from the device?

Yes, see https://github.com/Koenkk/zigbee-herdsman-converters/blob/0d287c9b6cedd81207de4508f87e89ad314f28b6/converters/fromZigbee.js#L3146

I understand it's interesting to add support for this device given it's price tag, but I'm wondering if we should do it. This seems to be a very exotic device, all the additions in zigbee-herdsman will be specific to this device. Also, we cannot support this device with the most used adapter (the Texas Instruments ones). Therefore listing it on the supported devices page sounds misleading to me. Of course this device can be supported indirectly through an external converter. Then we can also avoid some changes in zigbee-herdsman since we can call the adapter code directly.

ArcadeMachinist, 2022-12-28:
> Can you clarify this further?
The device has 6 independent LEDs.
I'm not sure how to correctly define them in exposes section.
I can define them like this:
```
exps.push(exposes.light('led1_color').withColor(['xy']).withDescription('Key 1 LED color'));
exps.push(exposes.light('led2_color').withColor(['xy']).withDescription('Key 2 LED color'));
etc...
```
But I'm not sure how to detect which exact one triggers the "color" change handler.
e.g.
```
key: ['color'],
		convertSet: async (entity, key, value, meta) => {
```

> but I'm wondering if we should do it
I think changes should be as minimal as possible, if something can be done from the converter - it should be done from there. This keypad device probably can be "partially supported" as keypress/sequence detection works fine with ZNP too.
And mains powered (works as router too) 6 button keypad is quite rare thing.

ZNP requires just
```{ ...EndpointDefaults, endpoint: 197, appprofid: 0xC25C },``` 
in endpoints.js
Again, if this definition can be somehow redefined from the converter level - better do it there.

Koenkk, 2022-12-29:
- You can use the `withEndpoint`, [example](https://github.com/Koenkk/zigbee-herdsman-converters/blob/bd8d4a6972cbb12e9ef564260b3ccbbca5d8a9d9/devices/lonsonho.js#L128). make sure to map the endpoints like done in the `endpoint` function below. The `entity` in `toZigbee` will be the endpoint.
- That's an acceptable change 😄

closed 2023-02-06 by github-actions[bot]