Koenkk/zigbee2mqtt #15365: No OTA-Support Tuya - TS011F_plug_1 / TS011F_plug_3

GerdRuetten, 2022-12-05:
### What happened?

The device-documentation say, the Tuya-devices TS011F_plug_1 and TS011F_plug_3 supported OTA updates.

If i check for updates, the 

**plug_3 device** respond with error:
2022-12-05 14:47:02Failed to check if update available for 'Steckdose Spülmaschine' (No image available for imageType '5634')

**plug_1 device** respond with error:
2022-12-05 14:48:02Failed to check if update available for 'Rauchmelder Marvin' (Device didn't respond to OTA request)

### What did you expect to happen?

Check for Updates without error

### Zigbee2MQTT version

1.28.2-dev commit: 8510e99

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus,TI CC2652P + CP2102(N)

### Debug log

Debug 2022-12-05 14:53:47Got OTA request '{"fieldControl":0,"manufacturerCode":4098,"imageType":5634,"fileVersion":69}'
Debug 2022-12-05 14:53:47ZigbeeOTA: downloaded main index
Info 2022-12-05 14:53:47MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Failed to check if update available for 'Steckdose Kühl-Gefrierschränke' (No image available for imageType '5634')","meta":{"device":"Steckdose Kühl-Gefrierschränke","status":"check_failed"},"type":"ota_update"}'
Info 2022-12-05 14:53:47MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Steckdose Kühl-Gefrierschränke"},"error":"Failed to check if update available for 'Steckdose Kühl-Gefrierschränke' (No image available for imageType '5634')","status":"error","transaction":"jywik-3"}'
Error 2022-12-05 14:53:47Failed to check if update available for 'Steckdose Kühl-Gefrierschränke' (No image available for imageType '5634')
Debug 2022-12-05 14:53:47Error: No image available for imageType '5634' at getImageMeta (/app/node_modules/zigbee-herdsman-converters/lib/ota/zigbeeOTA.js:113:15) at runMicrotasks (<anonymous>) at processTicksAndRejections (node:internal/process/task_queues:96:5) at isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:265:18) at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:256:23) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:192:39)

Koenkk, 2022-12-06:
There are many versions of this plug and we don't have OTAs for all. To retrieve the OTA see: https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html

DrSamjames, 2022-12-13:
Hi, 

I'm having the same problem. Can I check I understand correctly, I'll need  a Tuyan or Xiaomi gateway in order to use this method to find out which firmware my plug needs? I can't do it with the Sonoff Zigbee dongle and pi that I'm running my Zigbee2Mqtt on?

Thanks for the great software from a newbe

Sam

closed 2023-02-22 by github-actions[bot]