Koenkk/zigbee2mqtt #15367: "Unknown scene was recalled" from Home Assistant Core

robertklep, 2022-12-05:
### What happened?

I was having some (unrelated) issues with a group with two scenes, when I decided to remove all scenes for the group and start over.

The scenes were available from Home Assistant Core 2022.11.5, and still are (even though they were removed from z2m) and recalling a scene now yields errors because the recall is for an undefined scene ID.

I can't find where the scene ID's are being stored that are getting recalled so unsure if this is a HA issue or a z2m issue.

### What did you expect to happen?

Scenes removed from z2m should also be removed from Home Assistant (I restarted both z2m and HA a few times between iterations of trying to get the scenes removed).

### How to reproduce it (minimal and precise)

(I can only explain how I got here)
* Create a z2m group, and add two scenes (in my case they were called "Woonkamer 1%" and "Woonkamer 100%")
* The scenes should appear in Home Assistant as scenes (`scene.woonkamer_1` and `scene.woonkamer_100`)
* Remove the scenes from the z2m frontend
* The scenes are still present in HA and recalling them yields an error in the z2m log

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

Sonoff ZBDongle-P

### Debug log

The group definition from `zigbee2mqtt/bridge/groups`:
```
{
    "friendly_name": "Sfeerverlichting Woonkamer",
    "id": 3,
    "members": [
      {
        "endpoint": 1,
        "ieee_address": "0xd0cf5efffe6d6cc7"
      },
      {
        "endpoint": 1,
        "ieee_address": "0x086bd7fffe3647aa"
      },
      {
        "endpoint": 1,
        "ieee_address": "0xd0cf5efffe718f48"
      },
      {
        "endpoint": 1,
        "ieee_address": "0x04cd15fffe62b190"
      },
      {
        "endpoint": 11,
        "ieee_address": "0x84fd27fffe9c76b9"
      },
      {
        "endpoint": 1,
        "ieee_address": "0x90fd9ffffe2ad6d3"
      }
    ],
    "scenes": []
  }
```

Trying to active a scene from HA:
```
zigbee2mqtt/Sfeerverlichting Woonkamer/set { "scene_recall" : 2 }
zigbee2mqtt/bridge/logging {"level":"warning","message":"Unknown scene was recalled for 0xd0cf5efffe6d6cc7, can't restore state."}
zigbee2mqtt/bridge/logging {"level":"warning","message":"Unknown scene was recalled for 0x086bd7fffe3647aa, can't restore state."}
zigbee2mqtt/bridge/logging {"level":"warning","message":"Unknown scene was recalled for 0xd0cf5efffe718f48, can't restore state."}
zigbee2mqtt/bridge/logging {"level":"warning","message":"Unknown scene was recalled for 0x04cd15fffe62b190, can't restore state."}
zigbee2mqtt/bridge/logging {"level":"warning","message":"Unknown scene was recalled for 0x84fd27fffe9c76b9, can't restore state."}
zigbee2mqtt/bridge/logging {"level":"warning","message":"Unknown scene was recalled for 0x90fd9ffffe2ad6d3, can't restore state."}
```

danieledwardgeorgehitchcock, 2022-12-06:
Hi,

The scenes may still be being cached by Z2M. Try following [this guide](https://www.zigbee2mqtt.io/guide/usage/scenes.html#remove-scene) to see if you have any better luck...

robertklep, 2022-12-06:
@danieledwardgeorgehitchcock thanks for the tip. Sadly, HA keeps on trying to recall the same non-existent scene ID (I restarted both HA and z2m). It's also not possible to remove the scenes from HA manually since they can't be managed from the UI (I'm fine with using the CLI to remove them from `.storage/core.restore_state` in HA but because it's not clear to me where the fault lies I don't want to use that method just yet)

closed 2022-12-07 by robertklep