Koenkk/zigbee2mqtt #15368: Popp 701721 interview fails

mjaneczek, 2022-12-05:
### What happened?

Popp 701721 interview fails, I believe the first time I tried to connect the device, the error displayed but next after a few minutes it dispersed. I am talking about the "interview failed" danger icon next to the device image. Maybe I did something, not sure.

I was doing some testing and disconnected the device and connected it again, but now it fails all the time. Not sure what I did that it passed one time. Is there another way to "reinterview the device" without reconnecting it?

So basically: 
- interview fails
- it fixes itself or I did something (not sure here)
- I have disconnected the device, and connected it again
- interview fails

And I tried to fix it somehow but nothing helped.

### Zigbee2MQTT version

1.28.4-1

### Adapter firmware version

20210708

### Adapter

SLAE.SH CC2652RB Type 4 SLAESHv4

### Debug log

```
Info 2022-12-05 16:14:11Starting interview of '0x943469fffe70c1b6'
Info 2022-12-05 16:14:11MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x943469fffe70c1b6","ieee_address":"0x943469fffe70c1b6","status":"started"},"type":"device_interview"}'
Debug 2022-12-05 16:14:11Received MQTT message on 'homeassistant/sensor/0x943469fffe70c1b6/battery/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x943469fffe70c1b6"],"manufacturer":"Popp","model":"Smart thermostat based on Danfoss Ally (014G2461) (701721)","name":"0x943469fffe70c1b6","sw_version":"01.12.0008 01.12"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"0x943469fffe70c1b6 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x943469fffe70c1b6","unique_id":"0x943469fffe70c1b6_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Debug 2022-12-05 16:14:11Received MQTT message on 'homeassistant/sensor/0x943469fffe70c1b6/keypad_lockout/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x943469fffe70c1b6"],"manufacturer":"Popp","model":"Smart thermostat based on Danfoss Ally (014G2461) (701721)","name":"0x943469fffe70c1b6","sw_version":"01.12.0008 01.12"},"enabled_by_default":false,"entity_category":"config","icon":"mdi:lock","name":"0x943469fffe70c1b6 keypad lockout","state_topic":"zigbee2mqtt/0x943469fffe70c1b6","unique_id":"0x943469fffe70c1b6_keypad_lockout_zigbee2mqtt","value_template":"{{ value_json.keypad_lockout }}"}'
Debug 2022-12-05 16:14:11Received MQTT message on 'homeassistant/select/0x943469fffe70c1b6/keypad_lockout/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x943469fffe70c1b6/set/keypad_lockout","device":{"identifiers":["zigbee2mqtt_0x943469fffe70c1b6"],"manufacturer":"Popp","model":"Smart thermostat based on Danfoss Ally (014G2461) (701721)","name":"0x943469fffe70c1b6","sw_version":"01.12.0008 01.12"},"entity_category":"config","icon":"mdi:lock","name":"0x943469fffe70c1b6 keypad lockout","options":["unlock","lock1","lock2"],"state_topic":"zigbee2mqtt/0x943469fffe70c1b6","unique_id":"0x943469fffe70c1b6_keypad_lockout_zigbee2mqtt","value_template":"{{ value_json.keypad_lockout }}"}'
Debug 2022-12-05 16:14:11Received MQTT message on 'homeassistant/sensor/0x943469fffe70c1b6/programming_operation_mode/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x943469fffe70c1b6"],"manufacturer":"Popp","model":"Smart thermostat based on Danfoss Ally (014G2461) (701721)","name":"0x943469fffe70c1b6","sw_version":"01.12.0008 01.12"},"enabled_by_default":false,"name":"0x943469fffe70c1b6 programming operation mode","state_topic":"zigbee2mqtt/0x943469fffe70c1b6","unique_id":"0x943469fffe70c1b6_programming_operation_mode_zigbee2mqtt","value_template":"{{ value_json.programming_operation_mode }}"}'
Debug 2022-12-05 16:14:11Received MQTT message on 'homeassistant/select/0x943469fffe70c1b6/programming_operation_mode/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x943469fffe70c1b6/set/programming_operation_mode","device":{"identifiers":["zigbee2mqtt_0x943469fffe70c1b6"],"manufacturer":"Popp","model":"Smart thermostat based on Danfoss Ally (014G2461) (701721)","name":"0x943469fffe70c1b6","sw_version":"01.12.0008 01.12"},"name":"0x943469fffe70c1b6 programming operation mode","options":["setpoint","schedule","eco"],"state_topic":"zigbee2mqtt/0x943469fffe70c1b6","unique_id":"0x943469fffe70c1b6_programming_operation_mode_zigbee2mqtt","value_template":"{{ value_json.programming_operation_mode }}"}'
Debug 2022-12-05 16:14:11Received MQTT message on 'homeassistant/binary_sensor/0x943469fffe70c1b6/mounted_mode_active/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x943469fffe70c1b6"],"manufacturer":"Popp","model":"Smart thermostat based on Danfoss Ally (014G2461) (701721)","name":"0x943469fffe70c1b6","sw_version":"01.12.0008 01.12"},"name":"0x943469fffe70c1b6 mounted mode active","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x943469fffe70c1b6","unique_id":"0x943469fffe70c1b6_mounted_mode_active_zigbee2mqtt","value_template":"{{ value_json.mounted_mode_active }}"}'
Debug 2022-12-05 16:14:11Received MQTT message on 'homeassistant/switch/0x943469fffe70c1b6/switch_mounted_mode_control/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x943469fffe70c1b6/set/mounted_mode_control","device":{"identifiers":["zigbee2mqtt_0x943469fffe70c1b6"],"manufacturer":"Popp","model":"Smart thermostat based on Danfoss Ally (014G2461) (701721)","name":"0x943469fffe70c1b6","sw_version":"01.12.0008 01.12"},"name":"0x943469fffe70c1b6 mounted_mode_control","payload_off":"false","payload_on":"true","state_topic":"zigbee2mqtt/0x943469fffe70c1b6","unique_id":"0x943469fffe70c1b6_switch_mounted_mode_control_zigbee2mqtt","value_template":"{% if value_json.mounted_mode_control %} true {% else %} false {% endif %}"}'
Debug 2022-12-05 16:14:11Received MQTT message on 'homeassistant/switch/0x943469fffe70c1b6/switch_thermostat_vertical_orientation/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x943469fffe70c1b6/set/thermostat_vertical_orientation","device":{"identifiers":["zigbee2mqtt_0x943469fffe70c1b6"],"manufacturer":"Popp","model":"Smart thermostat based on Danfoss Ally (014G2461) (701721)","name":"0x943469fffe70c1b6","sw_version":"01.12.0008 01.12"},"name":"0x943469fffe70c1b6 thermostat_vertical_orientation","payload_off":"false","payload_on":"true","state_topic":"zigbee2mqtt/0x943469fffe70c1b6","unique_id":"0x943469fffe70c1b6_switch_thermostat_vertical_orientation_zigbee2mqtt","value_template":"{% if value_json.thermostat_vertical_orientation %} true {% else %} false {% endif %}"}'
Debug 2022-12-05 16:14:11Received MQTT message on 'homeassistant/switch/0x943469fffe70c1b6/switch_viewing_direction/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x943469fffe70c1b6/set/viewing_direction","device":{"identifiers":["zigbee2mqtt_0x943469fffe70c1b6"],"manufacturer":"Popp","model":"Smart thermostat based on Danfoss Ally (014G2461) (701721)","name":"0x943469fffe70c1b6","sw_version":"01.12.0008 01.12"},"name":"0x943469fffe70c1b6 viewing_direction","payload_off":"false","payload_on":"true","state_topic":"zigbee2mqtt/0x943469fffe70c1b6","unique_id":"0x943469fffe70c1b6_switch_viewing_direction_zigbee2mqtt","value_template":"{% if value_json.viewing_direction %} true {% else %} false {% endif %}"}'
Debug 2022-12-05 16:14:11Received MQTT message on 'homeassistant/switch/0x943469fffe70c1b6/switch_heat_available/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x943469fffe70c1b6/set/heat_available","device":{"identifiers":["zigbee2mqtt_0x943469fffe70c1b6"],"manufacturer":"Popp","mode [... cut ...]
Info 2022-12-05 16:14:26MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x943469fffe70c1b6","ieee_address":"0x943469fffe70c1b6"},"type":"device_announce"}'
Error 2022-12-05 16:14:31Failed to interview '0x943469fffe70c1b6', device has not successfully been paired
Info 2022-12-05 16:14:31MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x943469fffe70c1b6","ieee_address":"0x943469fffe70c1b6","status":"failed"},"type":"device_interview"}'
```

mjaneczek, 2023-01-06:
I have changed nothing and it pair successfully after 20th try, no idea what changed, anyway sine no activity I can close till someone else will have similar issue.

closed 2023-01-06 by mjaneczek

hacker, 2023-11-07:
For the record: had the same problem with the same resolution.