Koenkk/zigbee2mqtt #15371: Latest dev release doesn't working

Ivano62, 2022-12-05:
### What happened?

After upgrade to latest dev release zigbee2mqtt doesn't start. I get get this error:

Using '/app/data' as data directory
Zigbee2MQTT:error 2022-12-05 17:11:32: Failed to call 'OnEvent' 'stop' (TypeError: Cannot read properties of undefined (reading 'getDevices')
    at Zigbee.devices (/app/lib/zigbee.ts:275:30)
    at OnEvent.stop (/app/lib/extension/onEvent.ts:37:42)
    at Controller.callExtensions (/app/lib/controller.ts:316:40)
    at Controller.stop (/app/lib/controller.ts:193:9)
    at stop (/app/index.js:111:5)
    at process.handleQuit (/app/index.js:117:9))
Zigbee2MQTT:error 2022-12-05 17:11:32: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of null (reading 'clients')
    at Frontend.stop (/app/lib/extension/frontend.ts:83:39)
    at Controller.callExtensions (/app/lib/controller.ts:316:40)
    at Controller.stop (/app/lib/controller.ts:193:9)
    at stop (/app/index.js:111:5)
    at process.handleQuit (/app/index.js:117:9))
Zigbee2MQTT:error 2022-12-05 17:11:32: Not connected to MQTT server!
Zigbee2MQTT:error 2022-12-05 17:11:32: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
TypeError: Cannot read properties of undefined (reading 'end')
    at MQTT.disconnect (/app/lib/mqtt.ts:129:21)
    at Controller.stop (/app/lib/controller.ts:198:9)
    at stop (/app/index.js:111:5)
    at process.handleQuit (/app/index.js:117:9)

### What did you expect to happen?

It start without any errors

### Zigbee2MQTT version

dev ba2d9ae

### Adapter firmware version

0x26720700

### Adapter

RaspBee2

### Debug log

Using '/app/data' as data directory
Zigbee2MQTT:error 2022-12-05 17:11:32: Failed to call 'OnEvent' 'stop' (TypeError: Cannot read properties of undefined (reading 'getDevices')
    at Zigbee.devices (/app/lib/zigbee.ts:275:30)
    at OnEvent.stop (/app/lib/extension/onEvent.ts:37:42)
    at Controller.callExtensions (/app/lib/controller.ts:316:40)
    at Controller.stop (/app/lib/controller.ts:193:9)
    at stop (/app/index.js:111:5)
    at process.handleQuit (/app/index.js:117:9))
Zigbee2MQTT:error 2022-12-05 17:11:32: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of null (reading 'clients')
    at Frontend.stop (/app/lib/extension/frontend.ts:83:39)
    at Controller.callExtensions (/app/lib/controller.ts:316:40)
    at Controller.stop (/app/lib/controller.ts:193:9)
    at stop (/app/index.js:111:5)
    at process.handleQuit (/app/index.js:117:9))
Zigbee2MQTT:error 2022-12-05 17:11:32: Not connected to MQTT server!
Zigbee2MQTT:error 2022-12-05 17:11:32: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
TypeError: Cannot read properties of undefined (reading 'end')
    at MQTT.disconnect (/app/lib/mqtt.ts:129:21)
    at Controller.stop (/app/lib/controller.ts:198:9)
    at stop (/app/index.js:111:5)
    at process.handleQuit (/app/index.js:117:9)

danieledwardgeorgehitchcock, 2022-12-05:
Could you just check and make sure your MQTT server is functioning correctly?

Ivano62, 2022-12-05:
My MQTT server is functioning correctly, in fact, as soon as I downgrade to the stable version 1.28.4 zigbee2mqtt starts regularly without any problem.
The problem has been introduced since the last two dev releases.
Thanks

Ivano62, 2022-12-07:
I've installed latest dev version (d075f07), it fails see below, so I revert to latest stable  version. I hope the problem will be solved. Thanks

Zigbee2MQTT:info  2022-12-07 10:27:28: Logging to console and directory: '/app/data/log/2022-12-07.10-27-27' filename: log.txt
Zigbee2MQTT:error 2022-12-07 10:27:28: Failed to call 'OnEvent' 'stop' (TypeError: Cannot read properties of undefined (reading 'getDevices')
    at Zigbee.devices (/app/lib/zigbee.ts:275:30)
    at OnEvent.stop (/app/lib/extension/onEvent.ts:37:42)
    at Controller.callExtensions (/app/lib/controller.ts:316:40)
    at Controller.stop (/app/lib/controller.ts:193:9)
    at stop (/app/index.js:111:5)
    at process.handleQuit (/app/index.js:117:9))
Zigbee2MQTT:error 2022-12-07 10:27:28: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of null (reading 'clients')
    at Frontend.stop (/app/lib/extension/frontend.ts:83:39)
    at Controller.callExtensions (/app/lib/controller.ts:316:40)
    at Controller.stop (/app/lib/controller.ts:193:9)
    at stop (/app/index.js:111:5)
    at process.handleQuit (/app/index.js:117:9))
Zigbee2MQTT:error 2022-12-07 10:27:28: Not connected to MQTT server!
Zigbee2MQTT:error 2022-12-07 10:27:28: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
Zigbee2MQTT:info  2022-12-07 10:27:28: Disconnecting from MQTT server
TypeError: Cannot read properties of undefined (reading 'end')
    at MQTT.disconnect (/app/lib/mqtt.ts:129:21)
    at Controller.stop (/app/lib/controller.ts:198:9)
    at stop (/app/index.js:111:5)
    at process.handleQuit (/app/index.js:117:9)