Koenkk/zigbee2mqtt #15372: [New device support]: TS110E  _TZ3210_k1msuvg6

JeroenNL85, 2022-12-05:
### Link

https://nl.aliexpress.com/item/1005003693644741.html?spm=a2g0o.order_list.order_list_main.10.6d3279d2eG0S2i&gatewayAdapt=glo2nld

### Database entry

{"id":30,"type":"Router","ieeeAddr":"0xa4c1384c99c113da","nwkAddr":16074,"manufId":4417,"manufName":"_TZ3210_k1msuvg6","powerSource":"Mains (single phase)","modelId":"TS110E","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[4,5,3,6,8,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"61440":0,"onOff":1,"onTime":0,"moesStartUpOnOff":2}},"genLevelCtrl":{"attributes":{"61440":1000,"64514":0,"64515":11,"64516":1000,"currentLevel":255}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1670254526924,"defaultSendRequestWhen":"immediate"}

### Comments

Currently shows unsupported

JeroenNL85, 2022-12-05:
Maybe similar to https://github.com/Koenkk/zigbee2mqtt/issues/11853.
New to Zigbee2MQTT, will try if i can add that external converter

JeroenNL85, 2022-12-10:
Tried to add converter but failed:

[14:24:56] INFO: Starting Zigbee2MQTT...
/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')

Koenkk, 2022-12-10:
Can you try with: https://gist.github.com/Koenkk/34134c29cd219e16b512092750a5e3b9

JeroenNL85, 2022-12-10:
That works,  on/off and brightness works too.
Halogen light start to turn on at 20, will try with a led too.

Power on behaviour gives a error, but i dont need that
Error: Write 0xa4c1384c99c113da/1 genOnOff({"startUpOnOff":2}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:105:13) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)

Koenkk, 2022-12-10:
> Power on behaviour gives a error, but i dont need that

Doesn't seem to be supported, removed it.

KoKolaj, 2022-12-14:
GIRIER Tuya ZigBee 3.0 Smart Dimmer Switch Module 10A - 1 gang

Link
https://www.aliexpress.com/item/1005003693644741.html?spm=a2g0o.order_list.order_list_main.32.2e931802dWk8Ui

Database entry
{"id":62,"type":"Router","ieeeAddr":"0xa4c1388cdbe64b2a","nwkAddr":36722,"manufId":4417,"manufName":"_TZ3210_k1msuvg6","powerSource":"Mains (single phase)","modelId":"TS110E","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[4,5,3,6,8,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"61440":0,"onOff":1,"onTime":0,"moesStartUpOnOff":2}},"genLevelCtrl":{"attributes":{"61440":1000,"64514":0,"64515":11,"64516":1000,"currentLevel":255}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1670996881995,"defaultSendRequestWhen":"immediate"}

Comments
Currently shows unsupported

External converter
No response

Supported color modes
No response

Color temperature range
No response

KoKolaj, 2022-12-14:
OK Koenkk) - ready, thanks ...

Koenkk, 2022-12-15:
Does it work?

KoKolaj, 2022-12-15:
> 
Yes everything is alright. Works

Koenkk, 2022-12-17:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-17 by Koenkk

Koenkk, 2023-05-26:
The device is support out-of-the-box now, you can remove the external converter.

reyko01, 2023-05-26:
Thanks, works, but don't see  "power_on_behavior": ""  and  "switch_type": ""

reyko01, 2023-05-26:
just made a quick check, seems that two elements are missing in exposes
            e.power_on_behavior(),
            tuya.exposes.switchType(),

https://github.com/Koenkk/zigbee-herdsman-converters/blob/7d2499b6667bda7962bfe070056082e009b1835f/src/devices/tuya.js#L4691