Koenkk/zigbee2mqtt #15373: ThirdReality Devices keep reporting OTA update available even though they are up-to-date

aatifk, 2022-12-05:
### What happened?

With the new OTA support added for a bunch of ThirdReality sensors and switches in the recent version, the sensors and switches update fine, but even after the update, a "Check for New Updates" results in the device reporting update available. Updating just installs the same firmware version again. In the attached image, the switches have 1.00.12 as the latest firmware, but you can see that one of them that I triggered the update check for for this example, shows an update available. 

[image]

### What did you expect to happen?

When firmware is updated, rechecking for updates should not show update available again.

### How to reproduce it (minimal and precise)

Update the ThirdReality Device via OTA update. Once update is finished, do the "Check for New Updates" again and it will show an update is available even though it will just install the same firmware again.

### Zigbee2MQTT version

 1.28.4 commit: 52e545f

### Adapter firmware version

20221102

### Adapter

Sonoff USB Zigbee Dongle Type-P

highincolorado, 2022-12-17:
I now have 6 Third Reality devices that show needing updates but are apparently at the latest version, also the version for the update just says “newer”.

deviantintegral, 2023-01-06:
The "newer" bug is fixed in 1.29.0 (and 1.29.1) but I still see the firmware reporting as available even after applying them.

stncttr908, 2023-01-26:
I am still having this issue as well on 1.29.2. Not a huge deal, just OCD because I like to keep my devices up to date. 💯

jpoeppelman1, 2023-01-27:
Yeah, this is super confusing.  Hope they can get this resolved.

NickM-27, 2023-02-01:
In my case it is showing a newer version (.18 instead of .12) however it goes through the update process and then the update is still not applied

deviantintegral, 2023-03-04:
Still an issue.

dronenb, 2023-03-14:
> In my case it is showing a newer version (.18 instead of .12) however it goes through the update process and then the update is still not applied

I was having this exact same issue, then I manually tried doing a "Check All" in the UI and it went away. The HA notifications about the updates went away also.

Koenkk, 2023-03-18:
Can someone provide the debug log when checking for an update?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

deviantintegral, 2023-03-18:
Looks like an assertion fails:

```
Mar 18 19:18:59 zigbee2mqtt npm[100723]: Zigbee2MQTT:info  2023-03-18 19:18:59: Updating 'Fireplace' to latest firmware
Mar 18 19:19:01 zigbee2mqtt npm[100723]: Zigbee2MQTT:debug 2023-03-18 19:19:01: Received Zigbee message from 'Fireplace', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"v1.00.12"}' from endpoint 1 with groupID 0
Mar 18 19:19:01 zigbee2mqtt npm[100723]: Zigbee2MQTT:info  2023-03-18 19:19:01: MQTT publish: topic 'zigbee2mqtt/Fireplace', payload '{"battery":null,"last_seen":"2023-03-18T23:19:01.687Z","linkquality":112,"state":"ON","update":{"installed_version":12,"latest_version":18,"state":"available"},"update_available":null,"voltage":2800}'
Mar 18 19:19:01 zigbee2mqtt npm[100723]: Zigbee2MQTT:debug 2023-03-18 19:19:01: Updating to latest '0x282c02bfffe91752' (3RSS009Z)
Mar 18 19:19:01 zigbee2mqtt npm[100723]: Zigbee2MQTT:debug 2023-03-18 19:19:01: Using endpoint '1'
Mar 18 19:19:03 zigbee2mqtt npm[100723]: Zigbee2MQTT:debug 2023-03-18 19:19:03: Received Zigbee message from 'Fireplace', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":12,"imageType":54181,"manufacturerCode":4659}' from endpoint 1 with groupID 0
Mar 18 19:19:03 zigbee2mqtt npm[100723]: Zigbee2MQTT:info  2023-03-18 19:19:03: MQTT publish: topic 'zigbee2mqtt/Fireplace', payload '{"battery":null,"last_seen":"2023-03-18T23:19:03.678Z","linkquality":112,"state":"ON","update":{"installed_version":12,"latest_version":18,"state":"available"},"update_available":null,"voltage":2800}'
Mar 18 19:19:03 zigbee2mqtt npm[100723]: Zigbee2MQTT:debug 2023-03-18 19:19:03: Got OTA request '{"fieldControl":0,"manufacturerCode":4659,"imageType":54181,"fileVersion":12}'
Mar 18 19:19:03 zigbee2mqtt npm[100723]: Zigbee2MQTT:debug 2023-03-18 19:19:03: ZigbeeOTA: downloaded main index
Mar 18 19:19:03 zigbee2mqtt npm[100723]: Zigbee2MQTT:debug 2023-03-18 19:19:03: getNewImage for '0x282c02bfffe91752', meta {"fileVersion":12,"fileSize":146322,"url":"https://tr-zha.s3.amazonaws.com/firmwares/SmartSwitchGen3_Zigbee_PROD_V12_v1.00.12.ota","sha512":"8149d94b41784a822cc412d867bc6d7f27893989715c57924460ee039e2bb3024b8edf124c9d52325dbe7fa0b230d5dff62de4ca87d46120ca6dc42ce6396194"}
Mar 18 19:19:04 zigbee2mqtt npm[100723]: Zigbee2MQTT:debug 2023-03-18 19:19:04: Update of 'Fireplace' failed (AssertionError [ERR_ASSERTION]: No new image available)
Mar 18 19:19:04 zigbee2mqtt npm[100723]: Zigbee2MQTT:info  2023-03-18 19:19:04: MQTT publish: topic 'zigbee2mqtt/Fireplace', payload '{"battery":null,"last_seen":"2023-03-18T23:19:03.678Z","linkquality":112,"state":"ON","update":{"installed_version":12,"latest_version":18,"state":"available"},"update_available":null,"voltage":2800}'
Mar 18 19:19:04 zigbee2mqtt npm[100723]: Zigbee2MQTT:info  2023-03-18 19:19:04: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Fireplace"},"error":"Update of 'Fireplace' failed (No new image available)","status":"error","transaction":"sj398-1"}'
Mar 18 19:19:04 zigbee2mqtt npm[100723]: Zigbee2MQTT:error 2023-03-18 19:19:04: Update of 'Fireplace' failed (No new image available)
Mar 18 19:19:04 zigbee2mqtt npm[100723]: Zigbee2MQTT:debug 2023-03-18 19:19:04: AssertionError [ERR_ASSERTION]: No new image available
Mar 18 19:19:04 zigbee2mqtt npm[100723]:     at getNewImage (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:422:5)
Mar 18 19:19:04 zigbee2mqtt npm[100723]:     at runMicrotasks (<anonymous>)
Mar 18 19:19:04 zigbee2mqtt npm[100723]:     at processTicksAndRejections (node:internal/process/task_queues:96:5)
Mar 18 19:19:04 zigbee2mqtt npm[100723]:     at Object.updateToLatest (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:282:19)
Mar 18 19:19:04 zigbee2mqtt npm[100723]:     at OTAUpdate.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/otaUpdate.ts:267:41
```

Here's what the OTA in the UI shows. Strangely, one shows no update while the other does:

[image]

Koenkk, 2023-03-19:
Can you try to:
- stop z2m
- remove the `update` state from the device in `data/state.json`
- start z2m, and see if the issue re-appears again.

If it is fixed now, it was caused by an old z2m bug.

jpoeppelman1, 2023-03-19:
> Can you try to:
> 
> * stop z2m
> * remove the `update` state from the device in `data/state.json`
> * start z2m, and see if the issue re-appears again.
> 
> If it is fixed now, it was caused by an old z2m bug.

I tried removing the update data from the state file and restart the docker container.  The app just added the update info back into the state json file.  It did not work.

deviantintegral, 2023-03-19:
I also tried removing the update state, but it worked for me.

closed 2023-04-27 by github-actions[bot]

coolspot18, 2024-01-29:
Is there a fix for this? Third Reality devices don't seem to update their firmware state after an update?

mininja2131, 2024-02-07:
> Is there a fix for this? Third Reality devices don't seem to update their firmware state after an update?

same issue here.

gcs8, 2024-08-27:
Same issue where the update is failing on the below.

`Firmware version
v1.00.72
Manufacturer
[Third Reality](https://www.zigbee2mqtt.io/supported-devices/#v=Third%20Reality)
Model
[3RSB015BZ](https://www.zigbee2mqtt.io/devices/3RSB015BZ.html#third%2520reality-3rsb015bz)`

highincolorado, 2024-08-27:
Firmware 1.00.72 got installed on all my shades and ever since they drop off the network constantly. No help from Third Reality and no way I could find to go back to the previous version. Very frustrated. Well today they dropped 1.00.79, installing it now. Fingers are crossed.

gcs8, 2024-08-27:
Let me know if you figure out the Konami code to get it to take the FW.

highincolorado, 2024-08-27:
My updates both to 1.00.72 and now to 1.00.76 work. I’m sorry that I can no longer recall what I did last year to make it work but the comment about removing update might have been it. Sorry I don’t have more for you.

gcs8, 2024-08-27:
Hmm, my Z2M is only about a month old right now. This is what I am seeing atm.

`info 2024-08-27 13:30:29z2m: Updating 'Front Door Blinds' to latest firmware`

`info 2024-08-27 13:30:35z2m: Update of 'Front Door Blinds' at 0.00%`

`info 2024-08-27 13:30:35z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Front Door Blinds', payload '{"battery":100,"linkquality":134,"position":0,"state":"CLOSE","update":{"installed_version":72,"latest_version":76,"progress":0,"state":"updating"},"update_available":null,"voltage":6200}'`

`info 2024-08-27 13:31:05z2m: Update of 'Front Door Blinds' at 0.11%, ≈ 540 minutes remaining`

`info 2024-08-27 13:31:05z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Front Door Blinds', payload '{"battery":100,"linkquality":145,"position":0,"state":"CLOSE","update":{"installed_version":72,"latest_version":76,"progress":0.11,"remaining":32380,"state":"updating"},"update_available":null,"voltage":6200}'`

`info 2024-08-27 13:31:35z2m: Update of 'Front Door Blinds' at 0.11%, ≈ 1014 minutes remaining`

`info 2024-08-27 13:31:35z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Front Door Blinds', payload '{"battery":100,"linkquality":145,"position":0,"state":"CLOSE","update":{"installed_version":72,"latest_version":76,"progress":0.11,"remaining":60856,"state":"updating"},"update_available":null,"voltage":6200}'`

`info 2024-08-27 13:31:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Front Door Blinds', payload '{"battery":100,"linkquality":142,"position":0,"state":"CLOSE","update":{"installed_version":72,"latest_version":76,"state":"available"},"update_available":null,"voltage":6200}'`

`info 2024-08-27 13:31:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Front Door Blinds"},"error":"Update of 'Front Door Blinds' failed (Update failed with reason: 'ABORT')","status":"error","transaction":"zhnjc-1"}'`

`error 2024-08-27 13:31:40z2m: Update of 'Front Door Blinds' failed (Update failed with reason: 'ABORT')`

gcs8, 2024-08-28:
Oh nice, randomly hit update this am and now I am on v1.00.76 for the blinds, I did see in the update notification in HA that it said it was from 2 hours ago? So, maybe someone touched something.