Koenkk/zigbee2mqtt #15383: Problem with creating new support

kepak1, 2022-12-05:
### What happened?

I tried to add new tuya device with manual (https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html)
I copied this converter and I cannot boot zigbee2mqtt. I get this error message 

`pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.28.2 start
> node index.js

/opt/zigbee2mqtt/dist/util/externally-loaded.js:25
    fromZigbee: [tuya.fz.datapoints],
                         ^

TypeError: Cannot read properties of undefined (reading 'datapoints')
    at /opt/zigbee2mqtt/dist/util/externally-loaded.js:25:26
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/opt/zigbee2mqtt/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/opt/zigbee2mqtt/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/opt/zigbee2mqtt/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/opt/zigbee2mqtt/lib/extension/externalConverters.ts:12:20)
    at new Controller (/opt/zigbee2mqtt/lib/controller.ts:84:58)`

### Zigbee2MQTT version

1.28.2 commit: e66fe85

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

danieledwardgeorgehitchcock, 2022-12-06:
Please can you post your external converter up so we can have a look?

It looks to me like you might be missing a definition..

kepak1, 2022-12-06:
It's copy-paste of converter example in section 2 in https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#_1-standard-part-of-the-setup

kepak1, 2022-12-06:
Woks ! thank you! This should be updated on website

danieledwardgeorgehitchcock, 2022-12-06:
It should! I'll put a PR in later for this

Koenkk, 2022-12-06:
`tuya.fz.datapoints` should work for Zigbee2MQTT 1.28.4

cedrichuchette, 2023-01-08:
Hello, 
I had the same initial issue. Tried what solved yours but still not working.
Note that I don't have same version of zigbee2mqtt but can't see how to update to higher level
```
> zigbee2mqtt@1.27.2 start
> node index.js

/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/opt/zigbee2mqtt/lib/extension/externalConverters.ts:15:17)
    at new Controller (/opt/zigbee2mqtt/lib/controller.ts:84:58)
    at start (/opt/zigbee2mqtt/index.js:108:18)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
```