Koenkk/zigbee2mqtt #15386: Tuya TB25 Zemismart Switch 2 gangs with outlet not working at version 1.28.4

alvarocastrojr, 2022-12-06:
### What happened?

Tuya TB25 Zemismart Switch 2 gangs with outlet was working fine until version 1.28.2, but when upgrading to 1.28.4 it stops working the endpoint switches, and I have to revert to 1.28.2 to get it working again.

### What did you expect to happen?

Expect TB25 to work with the upgrade

### How to reproduce it (minimal and precise)

Just need to upgrade

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

0x26580700

### Adapter

Conbee II

Koenkk, 2022-12-06:
I can't see any changes regarding the TB25.

- Can you provide the data/database.db entry of this device?
- What is logged in the debug log when controlling the switch?
- 
See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

rodrigogbs, 2022-12-10:
same here:

`error 2022-12-10 07:31:24: Device 'Interruptores Suíte - Cabeceira ESQ' has no endpoint 'left'
error 2022-12-10 07:31:25: Device 'Interruptores Suíte - Cabeceira DIR' has no endpoint 'center'
error 2022-12-10 07:31:25: Device 'Interruptores Suíte - Cabeceira ESQ' has no endpoint 'center'
error 2022-12-10 07:31:25: Device 'Interruptores Suíte - Cabeceira DIR' has no endpoint 'left'
error 2022-12-10 07:31:25: Device 'Interruptores Banheiro Social' has no endpoint 'left'
error 2022-12-10 07:31:25: Device 'Interruptores Cozinha' has no endpoint 'left'
error 2022-12-10 07:31:25: Device 'Interruptores Cozinha' has no endpoint 'center'
error 2022-12-10 07:31:48: Device 'Interruptores Banheiro Social' has no endpoint 'left'
error 2022-12-10 07:32:15: Device 'Interruptores Banheiro Social' has no endpoint 'left'
error 2022-12-10 07:32:15: Device 'Interruptores Banheiro Social' has no endpoint 'left'
error 2022-12-10 07:33:03: Device 'Interruptores Banheiro Social' has no endpoint 'left'
error 2022-12-10 07:33:49: Device 'Interruptores Banheiro Social' has no endpoint 'left'
error 2022-12-10 07:34:22: Device 'Interruptores Cozinha' has no endpoint 'left'
error 2022-12-10 07:34:22: Device 'Interruptores Cozinha' has no endpoint 'center'
error 2022-12-10 07:35:00: Device 'Interruptores Banheiro Social' has no endpoint 'right'
error 2022-12-10 07:35:00: Device 'Interruptores Suíte - Cabeceira ESQ' has no endpoint 'right'
error 2022-12-10 07:35:00: Device 'Interruptores Suíte - Cabeceira DIR' has no endpoint 'right'
error 2022-12-10 07:35:00: Device 'Interruptores Cozinha' has no endpoint 'right'
error 2022-12-10 07:35:00: Device 'Interruptores Churrasqueira' has no endpoint 'right'
error 2022-12-10 07:35:43: Device 'Interruptores Banheiro Social' has no endpoint 'left'
error 2022-12-10 07:36:02: Device 'Interruptores Banheiro Social' has no endpoint 'left'
error 2022-12-10 07:36:04: Device 'Interruptores Banheiro Social' has no endpoint 'left'
error 2022-12-10 07:36:49: Device 'Interruptores Banheiro Social' has no endpoint 'left'
error 2022-12-10 07:36:51: Device 'Interruptores Banheiro Social' has no endpoint 'left'
error 2022-12-10 07:37:25: Device 'Interruptores Banheiro Social' has no endpoint 'left'
error 2022-12-10 07:37:56: Device 'Interruptores Banheiro Social' has no endpoint 'left'
error 2022-12-10 07:38:11: Device 'Interruptores Banheiro Social' has no endpoint 'left'
`

alvarocastrojr, 2022-12-10:
Well, I've reverted back the version without getting the logs, sorry. 
But it is the same as fellow rodrigo above.

it happens on all my TB25, I have 5 of them.

if the logs are still needed I can upgrade and get it.

Thanks

Koenkk, 2022-12-11:
Found the issue, fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-11 by Koenkk