Koenkk/zigbee2mqtt #15387: Ikea E1743: failed to unbind default_bind_group

holantomas, 2022-12-06:
### What happened?

Same issue as #2954. Cannot unbind switch from `default_bind_group`. Not even using ieee address.

I spent, last two days trying to somehow bind it with bulbs group (FW 2.2.010 - so need to bind it with group not directly with devices as docs says). Factory reset, repairing, delete and repair whole zigbee network(needed anyway - migration to conbee II), tons of server restarting etc.

It was working for several years. Then came little blackout which, somehow damage my network. I tried to upgrade FW of my CC2531. It not help, so I migrated to Conbee II. Nothing helped, it behave same way.

I'm using only one adapter at time.

### What did you expect to happen?

Just unbind from `default_bind_group` and bind with own bulbs group.

### How to reproduce it (minimal and precise)

Just try to bind fresh switch.

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

0x26580700(Conbee II) & 20211115(CC2531)

### Adapter

Conbee II

### Debug log

```
Debug 2022-12-06 09:51:55Received Zigbee message from 'bedroom_lightswitch', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Info 2022-12-06 09:51:55MQTT publish: topic 'rataje/138/patro_1/bedroom_lightswitch', payload '{"action":"on","battery":60,"click":"on","linkquality":167,"update":{"state":null},"update_available":null}'
Info 2022-12-06 09:51:55MQTT publish: topic 'rataje/138/patro_1/bedroom_lightswitch', payload '{"action":"","battery":60,"click":null,"linkquality":167,"update":{"state":null},"update_available":null}'
Info 2022-12-06 09:51:55MQTT publish: topic 'rataje/138/patro_1/bedroom_lightswitch', payload '{"action":null,"battery":60,"click":"","linkquality":167,"update":{"state":null},"update_available":null}'
Info 2022-12-06 09:51:55MQTT publish: topic 'rataje/138/patro_1/bedroom_lightswitch/action', payload 'on'
Info 2022-12-06 09:51:55MQTT publish: topic 'rataje/138/patro_1/bedroom_lightswitch/click', payload 'on'
Debug 2022-12-06 09:51:57Received MQTT message on 'rataje/138/patro_1/bridge/request/device/unbind' with data '{"from": "0xccccccfffe708894", "to": "default_bind_group"}'
Debug 2022-12-06 09:51:57unbinding cluster 'genOnOff' from 'bedroom_lightswitch' to 'default_bind_group'
Error 2022-12-06 09:51:57Failed to unbind cluster 'genOnOff' from 'bedroom_lightswitch' to 'default_bind_group' (Error: Unbind 0xccccccfffe708894/1 genOnOff from '901' failed (Error: status: 136))
Info 2022-12-06 09:51:57MQTT publish: topic 'rataje/138/patro_1/bridge/log', payload '{"message":{"cluster":"genOnOff","from":"bedroom_lightswitch","to":"default_bind_group"},"type":"device_unbind_failed"}'
Debug 2022-12-06 09:51:57unbinding cluster 'genLevelCtrl' from 'bedroom_lightswitch' to 'default_bind_group'
Error 2022-12-06 09:52:00Failed to unbind cluster 'genLevelCtrl' from 'bedroom_lightswitch' to 'default_bind_group' (Error: Unbind 0xccccccfffe708894/1 genLevelCtrl from '901' failed (Error: status: 136))
Info 2022-12-06 09:52:00MQTT publish: topic 'rataje/138/patro_1/bridge/log', payload '{"message":{"cluster":"genLevelCtrl","from":"bedroom_lightswitch","to":"default_bind_group"},"type":"device_unbind_failed"}'
Debug 2022-12-06 09:52:00unbinding cluster 'closuresWindowCovering' from 'bedroom_lightswitch' to 'default_bind_group'
Error 2022-12-06 09:52:01Failed to unbind cluster 'closuresWindowCovering' from 'bedroom_lightswitch' to 'default_bind_group' (Error: Unbind 0xccccccfffe708894/1 closuresWindowCovering from '901' failed (Error: status: 136))
Info 2022-12-06 09:52:01MQTT publish: topic 'rataje/138/patro_1/bridge/log', payload '{"message":{"cluster":"closuresWindowCovering","from":"bedroom_lightswitch","to":"default_bind_group"},"type":"device_unbind_failed"}'
Info 2022-12-06 09:52:02MQTT publish: topic 'rataje/138/patro_1/bridge/response/device/unbind', payload '{"data":{"clusters":[],"failed":["genOnOff","genLevelCtrl","closuresWindowCovering"],"from":"0xccccccfffe708894","to":"default_bind_group"},"error":"Failed to unbind","status":"error"}'
Error 2022-12-06 09:52:02Failed to unbind
```

danieledwardgeorgehitchcock, 2022-12-06:
What firmware version is your device on?

Ikea changed the behaviour of their bindings between firmwares a while ago - [see here](https://www.zigbee2mqtt.io/devices/E1743.html#binding)

danieledwardgeorgehitchcock, 2022-12-06:
Have you tried adding your bulb to a group and then binding the E1743 to that group?

holantomas, 2022-12-06:
@danieledwardgeorgehitchcock I wrote that above. Yes I already have group with 3 bulbs(tried with only one too). But I need it first unbind from `default_bind_group` and ti not working.

danieledwardgeorgehitchcock, 2022-12-06:
Apologies, so you did! Could you post the logs for what happens when you try and bind the E1743 to the group?

holantomas, 2022-12-06:
@danieledwardgeorgehitchcock wait a minute. After 1000 tries, it somehow start working now, even the fails in log and even it's still binded to `default_bind_group`. I just wanted to get log for you. Don't understand this. I guess we can close this. Sorry for your time.

closed 2023-01-14 by github-actions[bot]