Koenkk/zigbee2mqtt #15388: [Feature request]: Split Aqara from Xiaomi

ZZJHONS, 2022-12-06:
### Is your feature request related to a problem? Please describe

Aqara is an independent company from Xiaomi, not a white label like Roborock or MiJia

[image]

https://www.reddit.com/r/homeautomation/comments/wi0jc0/new_aqara_s1_scene_plates_installed/is9vpgb/?context=3

### Describe the solution you'd like

Create a new Vendor and move the Aqara devices.

### Describe alternatives you've considered

Aqara is gaining marketshare and it's releasing new products each years so I think it's better to split ASAP as official Xiaomi brands will grow too.

### Additional context

Will a Issue in the fronted also be needed?

ZZJHONS, 2023-01-07:
Headsup

ZZJHONS, 2023-01-07:
More comments about this:

https://www.reddit.com/r/HomeKit/comments/103zy5a/comment/j3bp269/?context=3

[image]

ZZJHONS, 2023-02-07:
headsup!

ZZJHONS, 2023-03-13:
:(

Koenkk, 2023-03-13:
Could you make a pr to do the split?

closed 2023-04-22 by github-actions[bot]

ZZJHONS, 2023-04-22:
Sorry, I don't have the time or expertise to do this now