Koenkk/zigbee2mqtt #15391: [New device support]: Tuya Garage Door Opener Controller

pasitiff, 2022-12-06:
### Link

https://aliexpress.ru/item/1005003822106367.html?spm=a2g2w.orderdetails.0.0.74384aa6GhZqQE&sku_id=12000031355782218&_ga=2.162941820.1720801430.1670323560-525979554.1670323560

### Database entry

{"id":32,"type":"Router","ieeeAddr":"0xa4c13835fac0bf43","nwkAddr":30968,"manufId":4417,"manufName":"_TZE200_nklqjk62","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�\u0000\u0000\u0000i�\u0000\u0000\u0000f�\u0000\u0000\u0000\u0012","65506":54,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_nklqjk62","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1670321108495}

### Comments

this device looks the same like in https://github.com/Koenkk/zigbee2mqtt/issues/10358 but with another Device model.
could you please add support or at least provide with converter

### External converter

```shell
-
```

uowis, 2022-12-06:
could we see your Zigbee2mqtt logs ?

pasitiff, 2022-12-06:
sure!

Zigbee2MQTT:info  2022-12-06 15:19:10: Device '0xa4c13835fac0bf43' joined
Zigbee2MQTT:info  2022-12-06 15:19:10: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13835fac0bf43","ieee_address":"0xa4c13835fac0bf43"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-12-06 15:19:10: Starting interview of '0xa4c13835fac0bf43'
Zigbee2MQTT:info  2022-12-06 15:19:10: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13835fac0bf43","ieee_address":"0xa4c13835fac0bf43","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-12-06 15:19:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c13835fac0bf43"},"type":"device_connected"}'
Zigbee2MQTT:info  2022-12-06 15:19:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c13835fac0bf43"},"type":"pairing"}'
Zigbee2MQTT:info  2022-12-06 15:19:10: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13835fac0bf43","ieee_address":"0xa4c13835fac0bf43"},"type":"device_announce"}'
Zigbee2MQTT:info  2022-12-06 15:19:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xa4c13835fac0bf43"},"type":"device_announced"}'
Zigbee2MQTT:info  2022-12-06 15:19:24: Successfully interviewed '0xa4c13835fac0bf43', device has successfully been paired
Zigbee2MQTT:warn  2022-12-06 15:19:24: Device '0xa4c13835fac0bf43' with Zigbee model 'TS0601' and manufacturer name '_TZE200_nklqjk62' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Zigbee2MQTT:info  2022-12-06 15:19:24: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xa4c13835fac0bf43","ieee_address":"0xa4c13835fac0bf43","status":"successful","supported":false},"type":"device_interview"}'
Zigbee2MQTT:info  2022-12-06 15:19:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0xa4c13835fac0bf43","supported":false},"type":"pairing"}'
Zigbee2MQTT:info  2022-12-06 15:19:37: MQTT publish: topic 'zigbee2mqtt/NewVeranda', payload '{"state_center":"OFF","state_left":"OFF","state_right":"OFF"}'
Zigbee2MQTT:warn  2022-12-06 15:19:51: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_nklqjk62'
Zigbee2MQTT:warn  2022-12-06 15:19:51: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2022-12-06 15:19:51: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_nklqjk62'
Zigbee2MQTT:warn  2022-12-06 15:19:51: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2022-12-06 15:19:51: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_nklqjk62'
Zigbee2MQTT:warn  2022-12-06 15:19:51: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

pasitiff, 2023-01-06:
.

closed 2023-02-13 by github-actions[bot]

pasitiff, 2023-02-14:
not supported:(