Koenkk/zigbee2mqtt #15402: [New device support]: Add missing feature to Xiaomi WS-USC02 WS-USC03 and WS-USC04 like the WS-USC01 have.

marc3k18, 2022-12-06:
### Link

https://www.zigbee2mqtt.io/supported-devices/#v=Xiaomi

### Database entry

{"id":43,"type":"Router","ieeeAddr":"0x54ef4410003c83b3","nwkAddr":35499,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.switch.b1naus01","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2,3,4,5,6,9,1794,2820],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.switch.b1naus01","manufacturerName":"LUMI","powerSource":1,"zclVersion":3,"appVersion":31,"stackVersion":2,"hwVersion":1,"dateCode":"12-28-2021"}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b69f97","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":31,"stackVersion":2,"hwVersion":1,"dateCode":"12-28-2021","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1670355837628,"defaultSendRequestWhen":"immediate"}

### Comments

The USC01 have all those feature: switch, action, flip_indicator_light, operation_mode, mode_switch. The USC02 should have the same.

USC03 and USC04 should also have the energy option since it's available in the Aqara apps.

Thanks!

CarlosGomez-dev, 2022-12-13:
I updated WS-USC01/02 as I recently bought them and was able to test them. I do not own the USC03/04 but if you have them you can create an external converter copying USC01 and USC02 and adding `e.power(), e.energy(), e.voltage()` to the `exposes` array (not sure if they support all 3)

visualage, 2023-01-01:
I have WS-USC04. Here is the PR to bring the feature list up to USC02.
https://github.com/Koenkk/zigbee-herdsman-converters/pull/5242

visualage, 2023-01-01:
Updated the PR to include power, energy and voltage for WS-USC04. It works fine, though the power is not updated immediately switching on/off.

CrazyCoder, 2023-01-03:
Getting this with WS-USC04 when trying to Refresh the power in Exposes settings tab:

```
Debug 2023-01-03 08:31:49 Received MQTT message on 'zigbee2mqtt/Patio Switch/get' with data '{"power":""}'
Debug 2023-01-03 08:31:49 Publishing get 'get' 'power' to 'Patio Switch'
Error 2023-01-03 08:31:49 Publish 'get' 'power' to 'Patio Switch' failed: 'TypeError: Cannot read properties of undefined (reading 'read')'
```

closed 2023-02-10 by github-actions[bot]

eobando00, 2023-04-10:
Hello, I have the USC03 and I really need the de-attach relay mode. I am happy to test if someone helps me a bit.

tmkhoa, 2023-05-21:
> Hello, I have the USC03 and I really need the de-attach relay mode. I am happy to test if someone helps me a bit.

Hi eobando00, I created a PR for WS-USC03 to expose the power, energy, etc. including the operational mode: https://github.com/Koenkk/zigbee-herdsman-converters/pull/5770

eobando00, 2023-06-02:
All set ! Thank you for your help! AWESOME!