Koenkk/zigbee2mqtt #15408: coordinator_backup.json is not created

moritzbeck01, 2022-12-07:
### What happened?

I noticed that the `coordinator_backup.json` file was not present under the `/app/data` path on my Docker install.
The Docker logs do mention that this directory is used for backups. All the other files are also present there.
I run a fairly newly migrated (by copying the `database.db` and `configuration.yaml` files) install with maybe a dozen devices.

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

0x26580700

### Adapter

ConBee 2

moritzbeck01, 2022-12-07:
> You can create it probably via /settings/tools -> request z2m backup in UI.

That doesn't actually create anything, it just zips the mentioned directory and makes it available for download.

samip5, 2022-12-07:
Huh, for me it does create that file although not using ConBee. :)

moritzbeck01, 2022-12-07:
> 

So if you delete the file and then request a backup, it gets recreated?

moritzbeck01, 2022-12-07:
> Conbee II doesn't support backups, so no `coordinator_backup.json` will be created when using this adapter. Only Texas Instruments based adapters support this.

Ah, makes sense then. The docs are really ambiguous about this, even with hindsight.
So how would I go about migrating to e.g. ZHA/zigpy then?

Koenkk, 2022-12-07:
> The docs are really ambiguous about this, even with hindsight.

You are right, I updated the docs.

> So how would I go about migrating to e.g. ZHA/zigpy then?

This will require repairing all devices. Even when you would have a `coordinator_backup.json` migrating without repairing is not possible since ZHA is not compatible with the Z2M `database.db` (and vice-versa).

moritzbeck01, 2022-12-07:
> You are right, I updated the docs.

Thanks!

> This will require repairing all devices. Even when you would have a `coordinator_backup.json` migrating without repairing is not possible since ZHA is not compatible with the Z2M `database.db` (and vice-versa).

I thought both support the open coordinator backup?

Koenkk, 2022-12-07:
> I thought both support the open coordinator backup?

Yes we do, but this does contain the information stored in `database.db` which zigbee2mqtt requires (I have no clue if ZHA has such file). The `coordinator_backup.json` file only contains the data stored on the coordinator side.

closed 2022-12-07 by moritzbeck01