Koenkk/zigbee2mqtt #15412: JY-GZ-01AQ cannot be added to groups

ciB89, 2022-12-07:
### What happened?

I recently purchased some of these devices and wanted to enable their linkage_alarm. I thought easiest was to listen to the topic with a wildcard (which works), but then I need to publish to enable the alarm on all devices on a broadcast - hence a group.

It seems that the device is already in a group with id 0?
Also, I do not understand why the response is with an empty groupname.

### What did you expect to happen?

Add the device to a group

### How to reproduce it (minimal and precise)

Create a group and add the device.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

Slaesh's CC2652RB stick

### Debug log

Received Zigbee message from 'Feuermelder Büro', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.sensor_smoke.acn03"}' from endpoint 1 with groupID 0
Debug 2022-12-07 09:36:32Received Zigbee message from 'Feuermelder Büro', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,174,12,3,40,25,4,33,168,19,5,33,13,0,6,36,2,0,0,0,0,8,33,17,1,10,33,156,142,12,32,1,102,32,3,103,32,1,104,33,168,0,160,33,0,0,161,32,0,162,32,0,163,32,0,164,32,1,165,32,1],"type":"Buffer"}}' from endpoint 1 with groupID 0
Debug 2022-12-07 09:35:17Error: Command 0x54ef4410005a7a37/1 genGroups.add({"groupid":3,"groupname":""}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 431 - 1 - 203 - 4 - 0 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7)
Info 2022-12-07 09:35:17MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/add', payload '{"data":{"device":"0x54ef4410005a7a37/1","group":"Feuermelder"},"error":"Failed to add from group (Command 0x54ef4410005a7a37/1 genGroups.add({\"groupid\":3,\"groupname\":\"\"}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 431 - 1 - 203 - 4 - 0 after 10000ms))","status":"error","transaction":"9x35m-1"}'
Error 2022-12-07 09:35:17Failed to add from group (Command 0x54ef4410005a7a37/1 genGroups.add({"groupid":3,"groupname":""}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 431 - 1 - 203 - 4 - 0 after 10000ms))

Koenkk, 2022-12-07:
End devices usually don't work in groups since they are sleeping while the group commands are send.

ciB89, 2022-12-07:
Well, the smoke detecters can be "linked" together, so when one alarm goes off the other ones also go off. That is only possible when the device does not sleep the entire time, correct? At least the Aqara hub manages to do that.

Koenkk, 2022-12-07:
I'm not sure how the Aqara hub does this but looks like some proprietary commands. To support this the Zigbee traffic has to be sniffed when setting this up through the Aqara hub: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

ciB89, 2022-12-07:
> I'm not sure how the Aqara hub does this but looks like some proprietary commands. To support this the Zigbee traffic has to be sniffed when setting this up through the Aqara hub: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

I don't own an aqara hub. You can select which other smoke detectors to "link" with in the aqara home app. I'm pretty sure it's just some automation it runs on the hub since I can manually trigger the alarm (also from z2m) - it might have a short delay, but it's working. I guess I have to setup an automation/blueprint for linking the device (also for muting them again). I was hoping groups would be working, since that would make it a lot easier to setup.

ciB89, 2022-12-07:
Although after looking at it again, the device behaves differently when it is triggered by a linked device vs when it detects fire itself. That is currently not possible with z2m, so there must be something missing.

closed 2023-01-14 by github-actions[bot]