Koenkk/zigbee2mqtt #15426: [New device support]: Phillips Hue Lightstrip Plus 2m (LCL006)

olibuijr, 2022-12-07:
### Link

https://www.philips-hue.com/en-gb/p/hue-white-and-colour-ambiance-lightstrip-plus-base-v4-2-metre/8718699703424

### Database entry

Zigbee2MQTT:info  2022-12-07 15:30:02: Device '0x001788010d0b59b3' joined Zigbee2MQTT:info  2022-12-07 15:30:02: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010d0b59b3","ieee_address":"0x001788010d0b59b3"},"type":"device_joined"}' Zigbee2MQTT:info  2022-12-07 15:30:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x001788010d0b59b3"},"type":"device_connected"}' Zigbee2MQTT:info  2022-12-07 15:30:02: Starting interview of '0x001788010d0b59b3' Zigbee2MQTT:info  2022-12-07 15:30:02: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010d0b59b3","ieee_address":"0x001788010d0b59b3","status":"started"},"type":"device_interview"}' Zigbee2MQTT:info  2022-12-07 15:30:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x001788010d0b59b3"},"type":"pairing"}' Zigbee2MQTT:info  2022-12-07 15:30:03: Successfully interviewed '0x001788010d0b59b3', device has successfully been paired Zigbee2MQTT:warn  2022-12-07 15:30:03: Device '0x001788010d0b59b3' with Zigbee model 'LCL006' and manufacturer name 'Signify Netherlands B.V.' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html Zigbee2MQTT:info  2022-12-07 15:30:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x001788010d0b59b3","ieee_address":"0x001788010d0b59b3","status":"successful","supported":false},"type":"device_interview"}' Zigbee2MQTT:info  2022-12-07 15:30:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x001788010d0b59b3","supported":false},"type":"pairing"}'

### Comments

Zigbee2MQTT:info  2022-12-07 15:30:02: Device '0x001788010d0b59b3' joined
Zigbee2MQTT:info  2022-12-07 15:30:02: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010d0b59b3","ieee_address":"0x001788010d0b59b3"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-12-07 15:30:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x001788010d0b59b3"},"type":"device_connected"}'
Zigbee2MQTT:info  2022-12-07 15:30:02: Starting interview of '0x001788010d0b59b3'
Zigbee2MQTT:info  2022-12-07 15:30:02: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010d0b59b3","ieee_address":"0x001788010d0b59b3","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-12-07 15:30:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x001788010d0b59b3"},"type":"pairing"}'
Zigbee2MQTT:info  2022-12-07 15:30:03: Successfully interviewed '0x001788010d0b59b3', device has successfully been paired
Zigbee2MQTT:warn  2022-12-07 15:30:03: Device '0x001788010d0b59b3' with Zigbee model 'LCL006' and manufacturer name 'Signify Netherlands B.V.' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:info  2022-12-07 15:30:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x001788010d0b59b3","ieee_address":"0x001788010d0b59b3","status":"successful","supported":false},"type":"device_interview"}'
Zigbee2MQTT:info  2022-12-07 15:30:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x001788010d0b59b3","supported":false},"type":"pairing"}'

danieledwardgeorgehitchcock, 2022-12-07:
You should be able to get the color temperature for this device following the guide on the main site.

Once done, we can add support

Koenkk, 2022-12-07:
`LCL006` is already supported indeed.

closed 2022-12-07 by Koenkk