Koenkk/zigbee2mqtt #15433: [New device support]: VA4220ZB / VA4221ZB Sinopé Smart Water Valve

alray31, 2022-12-07:
### Link

https://support.sinopetech.com/en/1.5.7.3/

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0x500b91400003e680","nwkAddr":973,"manufId":4508,"manufName":"Sinope Technologies\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":"VA4220ZB\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":3,"inClusterList":[0,1,3,4,5,6,8,1026,1280,1794,2821,65281],"outClusterList":[3,6,25],"clusters":{"genBasic":{"attributes":{"modelId":"VA4220ZB\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","manufacturerName":"Sinope Technologies\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":129,"zclVersion":2,"appVersion":12,"stackVersion":3,"hwVersion":0,"dateCode":"20201104\u0000\u0000\u0000\u0000\u0000\u0000\u0000","swBuildId":"3405\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0026b6a0f8","zoneState":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":12,"stackVersion":3,"hwVersion":0,"dateCode":"20201104\u0000\u0000\u0000\u0000\u0000\u0000\u0000","swBuildId":"3405\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1670446481691,"defaultSendRequestWhen":"immediate"}

### Comments

Hi,

Trying to bring support to Sinopé Smart Water Valve Gen 2 ZIgbee. Both models VA4220ZB and VA4221ZB are the same valve with the difference of the 4220 being for 3/4'' pipes and the 4221 for 1'' pipes. I own the VA4220ZB variant. I followed the guides to add new device support and made an external converter. I used info from an already supported water valve from Sinopé (model VA4200WZ) from zigbee-herdsman-converters/devices/sinope.js. The already supported model VA4200WZ looks to be a very similar model of VA4220ZB, the main differences appears to only be that the VA4200WZ is both WiFi and Zigbee, the VA4220ZB is Zigbee only (and 2nd gen from the manufacturer). 

So I copied the code from sinope.js under VA4200WZ and pasted it inside my external converted that I made for my valve (VA4220ZB.js). At that point, I was able to operate the valve from zigbee2mqtt: send close and open commands, the valve would report it's status correctly (Open, Close, position in % from 0 to 100 and Battery Low: False). However, Battery % exposed as ''unknown'' %.

Looking at the debug log, I saw that I still had a message because of a missing converter for SSiasZone:
```
Zigbee2MQTT:debug 2022-12-07 17:09:42: Received Zigbee message from 'Main Water Valve', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":50}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-12-07 17:09:42: No converter available for 'VA4220ZB' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":50}'
```

I tried to add the missing converter by adding an existing one ( fz.ias_water_leak_alarm_1) from zigbee-herdsman-converters/converters/fromZigbee.js  in the definition

Now, I do not have the debug message for the missing converter anymore but Battery level in % is still unknown and the Info log is showing this:
`info  2022-12-07 00:10:30: MQTT publish: topic 'zigbee2mqtt/Main Water Valve', payload '{"**battery":null**,"battery_low":null,"linkquality":165,"position":null,"state":null}'`

Which make me beleive i'm simply not using the correct converter for SSiasZone. I dont really know what to do at this point since I'm a new user. Any help would be appreciated.

Regards,

alray
[image]

@MattL0 I think you might be able to help me since you look to be active in the sinope.js file?

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['VA4220ZB'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'VA4220ZB', // Vendor model number, look on the device for a model number
    vendor: 'Sinopé', // Vendor of the device (only used for documentation and startup logging)
    description: 'Sedna Smart water valve 2nd gen – Zigbee', // Description 
    fromZigbee: [fz.ias_water_leak_alarm_1, fz.cover_position_via_brightness, fz.cover_state_via_onoff, fz.battery],
    toZigbee: [tz.cover_via_brightness],
    exposes: [e.valve_switch(), e.valve_position(), e.battery_low(), e.battery()],
};

module.exports = definition;
```

MattL0, 2022-12-07:
On the "console dev" tab  you can try to poll each attribute . First select the cluster , after the attribute (there is one for battery percentage), then press read .

Sorry I do not have a computer with me right now.

alray31, 2022-12-08:
@MattL0 ,

Thank you for taking time to answer. I did as mentioned and polled the batteryPercentageRemaining and also batteryVoltage attributes from the PowerCfg cluster and it returned a value of 0% and a battery voltage value of 65
`Read result of 'genPowerCfg': {"batteryVoltage":65,"batteryPercentageRemaining":0}`

The battery % value in the expose tab changed from ''null'' % to ''0'' % after polling.

I'm positive that the batteries are good, these are 4 new AAA lithium batteries. The valve operates flawlessly with the AC adapter unplugged using only battery power. In fact, the valve make a higher pitched noise when operating from the batteries compared to AC power adapter which makes me believe the batteries are providing even a bit more power than what the AC power adapter can deliver.

Here are all the attributes I was able to poll from genPowerCfg:
`Read result of 'genPowerCfg': {"batterySize":4,"batteryAlarmState":0,"batteryVoltage":65,"batteryPercentageRemaining":0,"batteryQuantity":4,"batteryAlarmMask":15,"batteryVoltMinThres":54}`

Any ideas?

MattL0, 2022-12-08:
I am not a bettery expert in zigbee2mqtt, but seems like battery voltage would be a good way to monitor the battery. 

you have 4 aaa 1.5volts battery so this means 6 volts in serial. Not sure if i'm right, but 65 / 10 seems to be near that number (6.5 volt). This would make your teory about the voltage of the battery higher right!

MattL0, 2022-12-08:
It seems like you also have the [seMetering](https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/cluster.ts#L3267) cluster. Could you try to poll the [remainingBattLife ](https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/cluster.ts#L3334) attribute with the same method on the dev tool tab?

MattL0, 2022-12-08:
I would try with this first :

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const constants = require('zigbee-herdsman-converters/lib/constants');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['VA4220ZB'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'VA4220ZB', // Vendor model number, look on the device for a model number
    vendor: 'Sinopé', // Vendor of the device (only used for documentation and startup logging)
    description: 'Sedna smart water valve', // Description 
    fromZigbee: [fz.ignore_iaszone_statuschange, fz.cover_position_via_brightness, fz.cover_state_via_onoff, fz.battery, fz.metering],
    toZigbee: [tz.cover_via_brightness],
    exposes: [e.valve_switch(), e.valve_position(), e.battery_low(), e.battery()],
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        const binds = ['genBasic', 'genGroups', 'genOnOff', 'ssIasZone' 'genLevelCtrl', 'genPowerCfg', 'seMetering',
'manuSpecificSinope'];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.batteryPercentageRemaining(endpoint);
        await reporting.onOff(endpoint);
        await reporting.brightness(endpoint); // valve position
              try {
                await reporting.batteryVoltage(endpoint);
             } catch (error) {/* Do Nothing */}
            try {
                await reportingbatteryAlarmState(endpoint);
             } catch (error) {/* Do Nothing */}
    },
};

module.exports = definition;
```

alray31, 2022-12-08:
I replaced the external converter with the one you provided and restarted Z2M. Tried to poll seMetering remainingBattlife and the log reported an unsupported attribute error:
`Error 2022-12-08 13:16:32Publish 'set' 'read' to 'Main Water Valve' failed: 'Error: Read 0x500b91400003e680/1 seMetering(["remainingBattLife"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

I also polled seMetering UnitofMesure and it reported the value ''7''. I checked in the [zigbee cluster librabry](https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf) and according to  section 10.4.2.2.4.1, value 7 is for ''Liters & l/h (Liters per Hour) in pure binary format''. Then polled MeteringDeviceType and it returned ''2' for ''Water Metering''.  Sounds to me that the presence of the seMetering cluster is solely for the optional FS4220/FS4221 flow meters that can be physically connected to the valve.

[image]
[image]

Another remark about battery voltage, I own a few of Sinope's water leak sensors, each are using the same AAA batteries but only 2 of them. When polling battery voltage of these sensors, they are all reporting between 2400 et 2600. I find it strange that the Sinope valve reports only a 2 digit values of 65 (well 64 now), shouldn't it be reporting a 4 digits values like the water sensors, i.e: 6500 instead of 65? I wish I knew how to multiply this value by by 100 so Z2M sees it as 6500 and then see what battery % is.

[image]
[image]
[image]
[image]
[image]

alray31, 2022-12-10:
With some research, it looks like the valve can't natively report battery percentage, even on the official Sinopé's app, Neviweb. batteryPercentageRemaining attribute from genPwrCfg always report 0% and remainingBattLife attribute from seMeterting is unsupported.

For now, I removed e.battery from the exposed attributes list since it was useless and replaced it with batteryVoltage. I updated my first post with the external converter I'm using and updated screenshot.  Everything works fine and no error in the logs. I can't test all the seMetering attribute since I do not own the external flow sensor.

Would still be nice to use batteryVoltage to convert it to battery percentage but I dont know how to add this piece to my external converter.

I know that:

- BatteryVoltage when batteries are new is 6800mV. 
- According to the manufacturer, the valve will stop working at 5400mV
- So 6800mV = 100%, 5400mV = 0%

We could use something like:
If batteryVoltage >= 5600mV then battery = 100%
if batteryVoltage < 5600mV then battery = ((batteryVoltage - 5400mv)/(5600mV - 5400mV) x 100)

But I don't know if possible and how to integrate this logic and to inject the result in the payload. I saw some example in libs/utils.js so it looks like its somehow possible.

MattL0, 2022-12-13:
```js

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const constants = require('zigbee-herdsman-converters/lib/constants');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['VA4220ZB'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'VA4220ZB', // Vendor model number, look on the device for a model number
    vendor: 'Sinopé', // Vendor of the device (only used for documentation and startup logging)
    description: 'Sedna smart water valve', // Description 
    
    fromZigbee: [fz.ignore_iaszone_statuschange, fz.cover_position_via_brightness, fz.cover_state_via_onoff, fz.battery, fz.metering],
    toZigbee: [tz.cover_via_brightness],
    meta: {battery: {voltageToPercentage: {min: 54, max: 68}}},
    exposes: [e.valve_switch(), e.valve_position(), e.battery_low(), e.battery()],
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        const binds = ['genBasic', 'genGroups', 'genOnOff', 'ssIasZone', 'genLevelCtrl', 'genPowerCfg', 'seMetering',
                       'manuSpecificSinope'];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.batteryPercentageRemaining(endpoint);
        await reporting.onOff(endpoint);
        await reporting.brightness(endpoint); // valve position
        try {
            await reporting.batteryVoltage(endpoint);
        } catch (error) {/* Do Nothing */}
        try {
            await reportingbatteryAlarmState(endpoint);
        } catch (error) {/* Do Nothing */}
    },
};

module.exports = definition;
```

closed 2023-01-13 by alray31