Koenkk/zigbee2mqtt #15436: Consumption measurement (power) but no energy field

maxime1992, 2022-12-08:
### What happened?

I have bought this device to control the heaters at my place: https://www.zigbee2mqtt.io/devices/064882.html#legrand-064882

It also has access to the power as stated per the doc above, which is great.

I was hoping though that I could use this device in the energy dashboard of Home Assistant but unlike connected plugs that expose the energy field in kwh, this one doesn't.

### What did you expect to happen?

Have an energy field in kwh

### Zigbee2MQTT version

1.28.4-dev commit: 2c53bc2

### Adapter firmware version

20211217

### Adapter

sonoff

pierrepinon, 2022-12-26:
Hi @maxime1992,

Legrand device only expose instantaneous power, but you can use "Riemann sum Integral" available in HA to know power consumption of your heaters

maxime1992, 2022-12-28:
Heyyy that is fantastic, I just got it setup perfectly! Thanks @pierrepinon ! :smiley: 

I love getting a real use case to have an helper I've never used before :tada:

closed 2022-12-28 by maxime1992