Koenkk/zigbee2mqtt #15439: MOES UFO-R11 **TypeError: Cannot read properties of undefined (reading 'seq')

remlei, 2022-12-08:
### What happened?

IR Code Sent to UFO-R11 fails to send the IR Code randomly.

### What did you expect to happen?

I expect it to work reliably.

### How to reproduce it (minimal and precise)

Its random actually, if I set some sort of qued IR send command with mqtt publish. the IR code just fail to work.

Or you can just spam any IR codes at very short interval to trigger this easily. though my automation script when it comes to sending the IR via MQTT publish actually had 1 second delay per input to at least minimize the occurrence of the issue but even with that it still trigger the issue randomly.

### Zigbee2MQTT version

1.28.4 commit: 52e545f

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 Dongle-P

### Debug log

2022-12-08 19:38:33Received MQTT message on 'zigbee2mqtt/UFO-R11/set' with data '{"ir_code_to_send": "B5cjRRFwAh0C4AsDAx0CbgaAF0AH4AUDwBsDHQJuBoAbAR0CQAMBHQLgBQ9AE4AD4AcXC24GcAJuBh0CbgZwAg=="}'
2022-12-08 19:38:33Publishing 'set' 'ir_code_to_send' to 'UFO-R11'
2022-12-08 19:38:33Sending IR code: "B5cjRRFwAh0C4AsDAx0CbgaAF0AH4AUDwBsDHQJuBoAbAR0CQAMBHQLgBQ9AE4AD4AcXC24GcAJuBh0CbgZwAg=="
2022-12-08 19:38:33Sent IRCode part: 0AH4AUDwBsDHQJuBoAbAR0CQAMBHQLgBQ9AE4AD4AcXC24GcAJ (sum: 111, seq:0)
2022-12-08 19:38:33Sending IR code initiated.
2022-12-08 19:38:33Received Zigbee message from 'UFO-R11', type 'commandZosungSendIRCode02', cluster 'zosungIRTransmit', data '{"maxlen":64,"position":150,"seq":0}' from endpoint 1 with groupID 0
2022-12-08 19:38:33"IR-Message-Code02" received (msg:{"seq":0,"position":150,"maxlen":64})
2022-12-08 19:38:33Sent IRCode part: uBh0CbgZwAg=="}} (sum: 106, seq:0)
2022-12-08 19:38:33Received MQTT message on 'zigbee2mqtt/UFO-R11/set' with data '{"ir_code_to_send": "B5cjRRFwAh0C4AsDAx0CbgaAF0AH4AUDwBsDHQJuBoAbAR0CQAMBHQLgBQ9AE4AD4AcXC24GcAJuBh0CbgZwAg=="}'
2022-12-08 19:38:33Publishing 'set' 'ir_code_to_send' to 'UFO-R11'
2022-12-08 19:38:33Sending IR code: "B5cjRRFwAh0C4AsDAx0CbgaAF0AH4AUDwBsDHQJuBoAbAR0CQAMBHQLgBQ9AE4AD4AcXC24GcAJuBh0CbgZwAg=="
2022-12-08 19:38:33Received Zigbee message from 'UFO-R11', type 'commandZosungSendIRCode04', cluster 'zosungIRTransmit', data '{"seq":0,"zero0":0,"zero1":0}' from endpoint 1 with groupID 0
2022-12-08 19:38:33"IR-Message-Code04" received (msg:{"zero0":0,"seq":0,"zero1":0})
2022-12-08 19:38:33Sending IR code initiated.
2022-12-08 19:38:33IRCode has been successfuly sent. (seq:0)
2022-12-08 19:38:34Received Zigbee message from 'UFO-R11', type 'commandZosungSendIRCode01', cluster 'zosungIRTransmit', data '{"cmd":2,"length":166,"seq":0,"unk1":0,"unk2":57348,"unk3":1,"unk4":0,"zero":0}' from endpoint 1 with groupID 0
2022-12-08 19:38:34"IR-Message-Code01" received (msg:{"zero":0,"seq":0,"length":166,"unk1":0,"unk2":57348,"unk3":1,"cmd":2,"unk4":0})
2022-12-08 19:38:34TypeError: Cannot read properties of undefined (reading 'seq') at messagesGet (/app/node_modules/zigbee-herdsman-converters/lib/zosung.js:13:14) at Object.convert (/app/node_modules/zigbee-herdsman-converters/lib/zosung.js:46:27) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:143:51) at EventEmitter.emit (node:events:525:35) at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22) at Controller.<anonymous> (/app/lib/zigbee.ts:106:27) at Controller.emit (node:events:513:28) at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14) at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18) at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)
2022-12-08 19:38:34Received Zigbee message from 'UFO-R11', type 'commandZosungSendIRCode02', cluster 'zosungIRTransmit', data '{"maxlen":64,"position":0,"seq":0}' from endpoint 1 with groupID 0
2022-12-08 19:38:34"IR-Message-Code02" received (msg:{"seq":0,"position":0,"maxlen":64})
2022-12-08 19:38:34TypeError: Cannot read properties of undefined (reading 'seq') at messagesGet (/app/node_modules/zigbee-herdsman-converters/lib/zosung.js:13:14) at Object.convert (/app/node_modules/zigbee-herdsman-converters/lib/zosung.js:57:27) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:143:51) at EventEmitter.emit (node:events:525:35) at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22) at Controller.<anonymous> (/app/lib/zigbee.ts:106:27) at Controller.emit (node:events:513:28) at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14) at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18) at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)
2022-12-08 19:38:35Received Zigbee message from 'UFO-R11', type 'commandZosungSendIRCode02', cluster 'zosungIRTransmit', data '{"maxlen":64,"position":0,"seq":0}' from endpoint 1 with groupID 0
2022-12-08 19:38:35"IR-Message-Code02" received (msg:{"seq":0,"position":0,"maxlen":64})
2022-12-08 19:38:35TypeError: Cannot read properties of undefined (reading 'seq') at messagesGet (/app/node_modules/zigbee-herdsman-converters/lib/zosung.js:13:14) at Object.convert (/app/node_modules/zigbee-herdsman-converters/lib/zosung.js:57:27) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:143:51) at EventEmitter.emit (node:events:525:35) at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22) at Controller.<anonymous> (/app/lib/zigbee.ts:106:27) at Controller.emit (node:events:513:28) at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14) at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18) at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)

closed 2023-01-15 by github-actions[bot]

DKanarsky, 2023-09-24:
I can confirm the same behavior of my own device. Did you solve the issue?

omershtricker, 2024-08-04:
I am experiencing this exact issue, my current best solution is to check if the commend has been sent and resend it of not,