Koenkk/zigbee2mqtt #15440: [Feature request]: Enable set/get for "window_open" and "local_temperature" of Aqara Smart Radiator Thermostat E1

programmdesign, 2022-12-08:
### Is your feature request related to a problem? Please describe

I am trying to use automations, to combine the TRV with window and temperature sensors.

a) I would like to set "window_open" to true or false, depending on the state of a windows sensor. 
b) I would like to set "local_temperature" based on the temperature of a temperature sensor.

Currently using get / set for these exposed values is not possible.

### Describe the solution you'd like

Be able to get/set the "window_open" and "local_temperature" attribute of the TRV.

### Describe alternatives you've considered

- For "local_temperature", I'm able to publish a payload to "zigbee2mqtt/0x00000..../set/sensor_temp" to update "local_temperature".
- For "window_open": Turn the system_mode to "Off" if the window is open.

### Additional context

[image]

danieledwardgeorgehitchcock, 2022-12-08:
I'm not sure that this would be possible - I think the `window_open` attribute that is something that is only reported by the device, not a cluster attribute that could be get / set

programmdesign, 2022-12-08:
I see. Thanks for your response.
For "local_temperature" it should be possible though, right? At least I'm able to set it via "sensor_temp" but not via the UI.

buhralex, 2022-12-10:
> 

Maybe take a look at https://github.com/KartoffelToby/better_thermostat.
Although this is not directly mqtt based you can assign a window contact to the TRV and define a temperature sensor that should control the TRV.

closed 2023-01-18 by github-actions[bot]

shrx, 2024-01-14:
Does anyone know what the `window_open` value is based on?