Koenkk/zigbee2mqtt #15441: Aqara Bulb Timeout

Bastiencc, 2022-12-08:
### What happened?

Since some weeks (maybe related with the last update ?) I get sometimes Timeout error with my aqara bulb. It seems to happen only when I try to turn off the bulb. I also happen 

### What did you expect to happen?

Turn off the bulb without error

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20211217

### Adapter

zzh! zStack3x0

### Debug log

`info  2022-12-08 15:31:50: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur', payload '{"action":"on-press","action_duration":null,"battery":100,"brightness":255,"counter":1,"duration":0,"linkquality":32,"update":{"installed_version":"6.1.1.28573","state":"idle"},"update_available":false}'
info  2022-12-08 15:31:50: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur', payload '{"action":"","action_duration":null,"battery":100,"brightness":255,"counter":1,"linkquality":32,"update":{"installed_version":"6.1.1.28573","state":"idle"},"update_available":false}'
info  2022-12-08 15:31:50: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur/action', payload 'on-press'
info  2022-12-08 15:31:50: MQTT publish: topic 'zigbee2mqtt/chambre_1_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":69,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":"1.34\u0000","state":"idle"},"update_available":false}'
info  2022-12-08 15:31:51: MQTT publish: topic 'zigbee2mqtt/chambre_1_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":58,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":"1.34\u0000","state":"idle"},"update_available":false}'
info  2022-12-08 15:31:51: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur', payload '{"action":"on-press","action_duration":null,"battery":100,"brightness":255,"counter":1,"duration":0,"linkquality":36,"update":{"installed_version":"6.1.1.28573","state":"idle"},"update_available":false}'
info  2022-12-08 15:31:51: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur', payload '{"action":"","action_duration":null,"battery":100,"brightness":255,"counter":1,"linkquality":36,"update":{"installed_version":"6.1.1.28573","state":"idle"},"update_available":false}'
info  2022-12-08 15:31:51: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur/action', payload 'on-press'
info  2022-12-08 15:31:54: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur', payload '{"action":"on-press","action_duration":null,"battery":100,"brightness":255,"counter":1,"duration":0,"linkquality":40,"update":{"installed_version":"6.1.1.28573","state":"idle"},"update_available":false}'
info  2022-12-08 15:31:54: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur', payload '{"action":"","action_duration":null,"battery":100,"brightness":255,"counter":1,"linkquality":40,"update":{"installed_version":"6.1.1.28573","state":"idle"},"update_available":false}'
info  2022-12-08 15:31:54: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur/action', payload 'on-press'
info  2022-12-08 15:31:56: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur', payload '{"action":"on-press","action_duration":null,"battery":100,"brightness":255,"counter":1,"duration":0,"linkquality":40,"update":{"installed_version":"6.1.1.28573","state":"idle"},"update_available":false}'
info  2022-12-08 15:31:56: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur', payload '{"action":"","action_duration":null,"battery":100,"brightness":255,"counter":1,"linkquality":40,"update":{"installed_version":"6.1.1.28573","state":"idle"},"update_available":false}'
info  2022-12-08 15:31:56: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur/action', payload 'on-press'
info  2022-12-08 15:31:57: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur', payload '{"action":"on-press","action_duration":null,"battery":100,"brightness":255,"counter":1,"duration":0,"linkquality":40,"update":{"installed_version":"6.1.1.28573","state":"idle"},"update_available":false}'
info  2022-12-08 15:31:57: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur', payload '{"action":"","action_duration":null,"battery":100,"brightness":255,"counter":1,"linkquality":40,"update":{"installed_version":"6.1.1.28573","state":"idle"},"update_available":false}'
info  2022-12-08 15:31:57: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur/action', payload 'on-press'
info  2022-12-08 15:32:03: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur', payload '{"action":"on-press","action_duration":null,"battery":100,"brightness":255,"counter":1,"duration":0,"linkquality":43,"update":{"installed_version":"6.1.1.28573","state":"idle"},"update_available":false}'
info  2022-12-08 15:32:03: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur', payload '{"action":"","action_duration":null,"battery":100,"brightness":255,"counter":1,"linkquality":43,"update":{"installed_version":"6.1.1.28573","state":"idle"},"update_available":false}'
info  2022-12-08 15:32:03: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur/action', payload 'on-press'
info  2022-12-08 15:32:10: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur', payload '{"action":"on-press","action_duration":null,"battery":100,"brightness":255,"counter":1,"duration":0,"linkquality":43,"update":{"installed_version":"6.1.1.28573","state":"idle"},"update_available":false}'
info  2022-12-08 15:32:10: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur', payload '{"action":"","action_duration":null,"battery":100,"brightness":255,"counter":1,"linkquality":43,"update":{"installed_version":"6.1.1.28573","state":"idle"},"update_available":false}'
info  2022-12-08 15:32:10: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur/action', payload 'on-press'
error 2022-12-08 15:32:13: Publish 'set' 'brightness' to 'chambre_1_plafond' failed: 'Error: Command 0x00158d0002d39c01/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 3432 - 1 - 21 - 8 - 11 after 10000ms)'
info  2022-12-08 15:32:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to 'chambre_1_plafond' failed: 'Error: Command 0x00158d0002d39c01/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":254,\"transtime\":10}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 3432 - 1 - 21 - 8 - 11 after 10000ms)'","meta":{"friendly_name":"chambre_1_plafond"},"type":"zigbee_publish_error"}'
info  2022-12-08 15:32:13: MQTT publish: topic 'zigbee2mqtt/chambre_1_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":58,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":"1.34\u0000","state":"idle"},"update_available":false}'
info  2022-12-08 15:32:13: MQTT publish: topic 'zigbee2mqtt/chambre_1_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":58,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":"1.34\u0000","state":"idle"},"update_available":false}'
info  2022-12-08 15:32:13: MQTT publish: topic 'zigbee2mqtt/chambre_1_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":61,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":"1.34\u0000","state":"idle"},"update_available":false}'
error 2022-12-08 15:32:36: Publish 'set' 'brightness' to 'chambre_1_plafond' failed: 'Error: Command 0x00158d0002d39c01/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 3432 - 1 - 23 - 8 - 11 after 10000ms)'
[... cut ...]
info  2022-12-08 15:38:09: MQTT publish: topic 'zigbee2mqtt/chambre_1_plafond', payload '{"brightness":0,"color_mode":"color_temp","color_temp":338,"linkquality":25,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":"1.34\u0000","state":"idle"},"update_available":false}'
info  2022-12-08 15:38:27: MQTT publish: topic 'zigbee2mqtt/chambre_1_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":29,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":"1.34\u0000","state":"idle"},"update_available":false}'
`

Bastiencc, 2022-12-08:
I'm not sur but seems that the issue came after updating aqara bulb from 1.32 to 1.34. 
Is it possible to downgrade?

Bastiencc, 2022-12-13:
After downgrade all my bulb firmware to 1.32, seems to be ok now...

closed 2022-12-13 by Bastiencc

fama-lama, 2023-07-31:
i have the same problem with all my aqara ZNLDP12LM bulbs on bulb firmware 1.34. 

> Error 2023-07-30 22:42:20Publish 'set' 'brightness' to 'Living standing lamp' failed: 'Error: Command 0x00158d0007fff57e/1 genLevelCtrl.moveToLevelWithOnOff({"level":58,"transtime":450}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 56523 - 1 - 149 - 8 - 11 after 10000ms)'

>Error 2023-07-30 22:46:51Publish 'set' 'brightness' to 'Kitchen lamp' failed: 'Error: Command 0x00158d0007ebbbf3/1 genLevelCtrl.moveToLevelWithOnOff({"level":57,"transtime":450}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 44469 - 1 - 38 - 8 - 11 after 10000ms)'

>Error 2023-07-30 22:51:23Publish 'set' 'brightness' to 'Kitchen lamp' failed: 'Error: Command 0x00158d0007ebbbf3/1 genLevelCtrl.moveToLevelWithOnOff({"level":55,"transtime":450}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 44469 - 1 - 205 - 8 - 11 after 10000ms)'

>Error 2023-07-30 22:51:23Publish 'set' 'brightness' to 'Living standing lamp' failed: 'Error: Command 0x00158d0007fff57e/1 genLevelCtrl.moveToLevelWithOnOff({"level":55,"transtime":450}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 56523 - 1 - 206 - 8 - 11 after 10000ms)'

>Error 2023-07-30 23:17:01Publish 'set' 'brightness' to 'Kitchen lamp' failed: 'Error: Command 0x00158d0007ebbbf3/1 genLevelCtrl.moveToLevelWithOnOff({"level":47,"transtime":450}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 44469 - 1 - 143 - 8 - 11 after 10000ms)'

>Error 2023-07-30 23:17:02Publish 'set' 'brightness' to 'Living standing lamp' failed: 'Error: Command 0x00158d0007fff57e/1 genLevelCtrl.moveToLevelWithOnOff({"level":47,"transtime":450}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 56523 - 1 - 144 - 8 - 11 after 10000ms)'

>Error 2023-07-30 23:29:06Publish 'set' 'brightness' to 'Living standing lamp' failed: 'Error: Command 0x00158d0007fff57e/1 genLevelCtrl.moveToLevelWithOnOff({"level":45,"transtime":450}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 56523 - 1 - 222 - 8 - 11 after 10000ms)'

>Error 2023-07-30 23:59:52Exception while calling fromZigbee converter: Expected one of: 1, 2, 4, 8, 16, 31, 32, 42, 64, 85, 96, 127, got: 'undefined'}

>Error 2023-07-31 00:00:23Publish 'set' 'state' to 'workshop lamp' failed: 'Error: Command 0x00158d000800481f/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 38029 - 1 - 19 - 6 - 11 after 10000ms)'

>Error 2023-07-31 00:00:23Publish 'set' 'brightness' to 'Kitchen lamp' failed: 'Error: Command 0x00158d0007ebbbf3/1 genLevelCtrl.moveToLevelWithOnOff({"level":3,"transtime":10}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 44469 - 1 - 26 - 8 - 11 after 10000ms)'

>Error 2023-07-31 00:00:23Publish 'set' 'brightness' to 'Living standing lamp' failed: 'Error: Command 0x00158d0007fff57e/1 genLevelCtrl.moveToLevelWithOnOff({"level":3,"transtime":10}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 56523 - 1 - 27 - 8 - 11 after 10000ms)'

>Error 2023-07-31 00:11:19Publish 'set' 'brightness' to 'Kitchen lamp' failed: 'Error: Command 0x00158d0007ebbbf3/1 genLevelCtrl.moveToLevelWithOnOff({"level":3,"transtime":450}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeU [... cut ...] r 10000ms)'

>Error 2023-07-31 07:21:13Publish 'set' 'brightness' to 'workshop lamp' failed: 'Error: Command 0x00158d000800481f/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":450}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 38029 - 1 - 196 - 8 - 11 after 10000ms)'