Koenkk/zigbee2mqtt #15446: Problem with Aqara T1 MCCGQ12LM & RTCGQ12LM

ildmgithub, 2022-12-08:
### What happened?

   I haven't updated often, but every time I've tried to update over the past few months, those updates have caused problems to operation of Aqara`s motion sensors and door/window contact sensors.
All 10 sensors stopped sending states immediately after updates, so I returned  to Zigbee2MQTT 1.22.2-1 every time. Both Aqara`s motion sensors and the door/window contact sensors work with 1.22.2-1 without problems. 
   Last update I have tried was 1.28.4-1 and it had the same problem. Window sensors just kept silence but I have captured the error from motion sensors after last update.
By the way I have noticed that they have proper operation in Edge 1.28.4-1, but I can't use it due to big delays in the operation of Gledopter dimmers GL-SD-001 after switching to Edge.

### What did you expect to happen?

Proper operation after update

### How to reproduce it (minimal and precise)

Update Zigbee2MQTT ver

### Zigbee2MQTT version

all later than 1.22.2-1 and up to 1.28.4-1

### Adapter firmware version

 20210319

### Adapter

 zStack3x0

### Debug log

Error 2022-12-07 18:41:38 Failed to configure 'bedroom_motion., attempt 1 (Error: Read 0x54ef4410003a38b6/1 genPowerCfg (["batteryVoltage"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reserveflits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 32733 - 1 - 119 - 1 - 1 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7)) 

Error 2022-12-07 19:07:34 Failed to configure 'kidsroom_motion', attempt 2 (Error: Read 0x54ef44100038b95e/1 genPowerCfg (["batteryVoltage"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reserveflits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 31289 - 1 - 111 - 1 - 1 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))

Otnow, 2022-12-12:
In order for these devices to work properly on the latest version of _z2m_, you need to update the coordinator firmware and re-pair _Aqara_ devices: https://www.zigbee2mqtt.io/devices/RTCGQ12LM.html#adapter-firmware

ildmgithub, 2022-12-14:
Thanks. 
After the update everything is OK

closed 2022-12-14 by ildmgithub