Koenkk/zigbee2mqtt #15447: [New device support]: lumi.switch.n3acn1

Deuteronomy93, 2022-12-08:
### Link

https://www.aqara.com/cn/d27_overview

### Database entry

{"id":87,"type":"Router","ieeeAddr":"0x54ef441000644d0b","nwkAddr":8591,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.switch.n3acn1","epList":[1,2,3,21,41,42,43,51,52,61,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2,3,4,5,6,9,1794,2820],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.switch.n3acn1"}},"aqaraOpple":{"attributes":{"15":{"type":"Buffer","data":[1,10,0,0,0,0,16,16,51,0,0,0,2,10,1,240,3,0,16,240,21,0,0,0]},"223":{"type":"Buffer","data":[22,33,0,0,23,33,0,0,12,33,0,0,24,33,0,0,30,33,0,0,31,33,26,0,32,33,248,0,34,33,98,0,35,33,0,0,36,32,0,37,35,65,110,81,5,254,35,128,0,4,0]},"229":{"type":"Buffer","data":[8,0,8,0,0,4,96,170,232,37,52,90,186,37,56,197,158,37,66,149,25,37,3,129,212,37,114,103,97,37,98,227,95,37,96]},"238":5427,"247":{"type":"Buffer","data":[100,16,1,101,16,1,102,16,1,3,40,37,152,57,200,50,254,64,149,57,34,98,195,59,150,57,254,72,14,69,151,57,77,118,56,66,5,33,1,0,154,32,0,9,33,0,10,13,35,51,21,0,0,14,35,0,0,0,0,10,33,0,0,12,32,1,17,35,1,0,0,0]},"512":0,"670":0}},"genOnOff":{"attributes":{"245":117440544,"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"245":117440545,"onOff":1}},"aqaraOpple":{"attributes":{"512":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[0,3,4,5,6,18,64704],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"245":52530944,"onOff":1}},"aqaraOpple":{"attributes":{"512":0}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"epId":21,"inClusterList":[],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":7.989096164703369}}},"binds":[],"configuredReportings":[],"meta":{}},"41":{"epId":41,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"42":{"epId":42,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"43":{"epId":43,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"51":{"epId":51,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"52":{"epId":52,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"61":{"epId":61,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":51,"stackVersion":2,"hwVersion":1,"dateCode":"02-17-2022","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1511975526},"lastSeen":1670521448223,"defaultSendRequestWhen":"immediate"}

### Comments

Hi

I've copied the full converter file from here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/xiaomi.js

The device I'm adding has effectively already been added but as the 2 button version: lumi.switch.n2acn1

I've edited the entry to replace the model numbers, include a center button, change "both" presses to "all" etc. The library of zigbee herdsman converters have been added to my Z2M folder.

When using this converter and restarting Z2M, it hangs with the below info in the log. These exposes came from the 2 button's approved converter:
[01:50:40] INFO: Preparing to start...
[01:50:40] INFO: Socat not enabled
[01:50:40] INFO: Zigbee Herdsman debug logging enabled
[01:50:40] INFO: Starting Zigbee2MQTT...
/app/dist/util/externally-loaded.js:756
            exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                   ^
SyntaxError: Unexpected token '.'
    at new Script (node:vm:100:7)
    at createScript (node:vm:265:10)
    at Object.runInNewContext (node:vm:306:10)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)

When using a different converter (which can't differentiate button presses properly) I've found that rather than referencing endpoints 1, 2 and 3, for left, center and right, it uses 41 for left, 42 for center, 43 for right (log info showing this below).
The clusters in Z2M show info for endpoints 1, 2 and 3, but not for the ones in the 40s.

0x54ef441000644d0b
-Endpoint 1
--Output clusters
---genTime
---genOta
--Input clusters
---genBasic
---genDeviceTempCfg
---genIdentify
---genGroups
---genScenes
---genOnOff
---genAlarms
---seMetering
---haElectricalMeasurement
-Endpoint 2
--Input clusters
---genBasic
---genIdentify
---genGroups
---genScenes
---genOnOff
-Endpoint 3
--Input clusters
---genBasic
---genIdentify
---genGroups
---genScenes
---genOnOff
[... 803 lines cut ...]
        vendor: 'Xiaomi',
        description: 'Aqara smart wall switch H1 Pro (with neutral, triple rocker)',
        meta: {multiEndpoint: true},
        endpoint: (device) => {
            return {'left': 1, 'center': 2, 'right': 3};
        },
        fromZigbee: [fz.on_off, fz.xiaomi_power, fz.aqara_opple, fz.xiaomi_multistate_action],
        toZigbee: [tz.on_off, tz.xiaomi_switch_operation_mode_opple, tz.xiaomi_switch_power_outage_memory,
            tz.xiaomi_led_disabled_night, tz.xiaomi_flip_indicator_light],
        exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('center'), e.switch().withEndpoint('right'), e.power(), e.energy(), e.voltage(),
            e.device_temperature(), e.power_outage_memory(), e.led_disabled_night(), e.flip_indicator_light(),
            e.action([
                'single_left', 'single_center', 'single_right', 'single_all',
                'double_left', 'double_center', 'double_right', 'double_all']),
            exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for left button').withEndpoint('left'),
            exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for center button').withEndpoint('center')],
            exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for right button').withEndpoint('right')],
        onEvent: preventReset,
        ota: ota.zigbeeOTA,
    },
];
```

Deuteronomy93, 2022-12-12:
After removing the converter, restarting Z2M to get the network up again, and then adding the converter and restarting, Z2M works. Not sure what the issue was but no edits were made to the converter.

I have played around with the converter more and come to these theories:

- The 2 button switch's converter does not work correctly.
- I say this because the only way to use the single, double presses etc is when they're controlling the relay rather than decoupled. This makes them pretty much useless as whenever you would double press for example, it will turn the lights off then on again.
- What I'm trying to achieve is to decouple all of them and be able to use the single and double presses. Singles for lights, doubles for curtains/blinds etc.

The current converter I'm using doesn't allow the center button to be turned on/off or decoupled (I'll look at this later as I had a previous converter where it worked).

Please can someone review and let me know how I can achieve the above?

Many thanks!

``
    {
        zigbeeModel: ['lumi.switch.n3acn1'],
        model: 'QBKG32LM',
        vendor: 'Xiaomi',
        description: 'Aqara smart wall switch H1 Pro (with neutral, three rocker)',
        meta: {multiEndpoint: true},
        endpoint: (device) => {
            return {'left': 1, 'centre': 2, 'right': 3};
            return {'left': 41, 'center': 42, 'right': 43}; 
        },
        fromZigbee: [fz.on_off, fz.xiaomi_power, fz.aqara_opple, fz.xiaomi_multistate_action],
        toZigbee: [tz.on_off, tz.xiaomi_switch_operation_mode_opple, tz.xiaomi_switch_power_outage_memory,
            tz.xiaomi_led_disabled_night, tz.xiaomi_flip_indicator_light],
        exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('center'), e.switch().withEndpoint('right'), e.power(), e.energy(), e.voltage(),
            e.device_temperature(), e.power_outage_memory(), e.led_disabled_night(), e.flip_indicator_light(),
            e.action([
                'single_left', 'single_center', 'single_right', 'single_all',
				'double_left', 'double_center', 'double_right', 'double_all']),
            exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for left button').withEndpoint('left'),
            exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for center button').withEndpoint('center'),
            exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for right button').withEndpoint('right')],
        onEvent: preventReset,
        ota: ota.zigbeeOTA,
    },
``

CarlosGomez-dev, 2022-12-13:
Try this: 

```js 

const h1_pro_actions = {
  cluster: 'genMultistateInput',
  type: ['attributeReport', 'readResponse'],
  convert: (model, msg, publish, options, meta) => {
    if ([1, 2].includes(msg.data.presentValue)) {
      const mapping = { 41: 'left', 42: 'center', 43: 'right' };
      const times = { 1: 'single', 2: 'double' };
      const button = mapping[msg.endpoint.ID];
      return { action: `${times[msg.data.presentValue]}_${button}` };
    }
  },
}

const definition = {
        zigbeeModel: ['lumi.switch.n3acn1'],
        model: 'QBKG32LM',
        vendor: 'Xiaomi',
        description: 'Aqara smart wall switch H1 Pro (with neutral, three rocker)',
        meta: {multiEndpoint: true},
        endpoint: (device) => {
            return {'left': 1, 'centre': 2, 'right': 3};
            return {'left': 41, 'center': 42, 'right': 43}; 
        },
        fromZigbee: [fz.on_off, fz.xiaomi_power, fz.aqara_opple, fz.xiaomi_multistate_action, h1_pro_actions],
        toZigbee: [tz.on_off, tz.xiaomi_switch_operation_mode_opple, tz.xiaomi_switch_power_outage_memory,
            tz.xiaomi_led_disabled_night, tz.xiaomi_flip_indicator_light],
        exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('center'), e.switch().withEndpoint('right'), e.power(), e.energy(), e.voltage(),
            e.device_temperature(), e.power_outage_memory(), e.led_disabled_night(), e.flip_indicator_light(),
            e.action([
                'single_left', 'single_center', 'single_right', 'single_all',
				'double_left', 'double_center', 'double_right', 'double_all']),
            exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for left button').withEndpoint('left'),
            exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for center button').withEndpoint('center'),
            exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for right button').withEndpoint('right')],
        onEvent: preventReset,
        ota: ota.zigbeeOTA,
    }
```

I removed the second return in the endpoint key, it wasn't doing anything, and added the h1_pro_actions object, hopefully that's enough to report the actions.

I don't see a debug log for "all" button press, if you find the endpoint just add it to the h1_pro_actions mapping.

Deuteronomy93, 2022-12-13:
@CarlosGomez-dev 

Many thanks for your feedback! I'm trying to use the converter however it's causing Z2M to hang when restarting.

Firstly I get the below, this string has always caused issues in my previous converters:
```
[19:27:09] INFO: Preparing to start...
[19:27:09] INFO: Socat not enabled
[19:27:09] INFO: Zigbee Herdsman debug logging enabled
[19:27:09] INFO: Starting Zigbee2MQTT...
/app/dist/util/externally-loaded.js:45
        onEvent: preventReset,
                 ^
ReferenceError: preventReset is not defined
    at /app/dist/util/externally-loaded.js:45:18
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
```
If I comment out "onEvent: preventReset," I get the following when trying to start Z2M:
```
[19:30:27] INFO: Preparing to start...
[19:30:27] INFO: Socat not enabled
[19:30:27] INFO: Zigbee Herdsman debug logging enabled
[19:30:27] INFO: Starting Zigbee2MQTT...
/app/node_modules/zigbee-herdsman-converters/index.js:78
    definition.toZigbee.push(tz.scene_store, tz.scene_recall, tz.scene_add, tz.scene_remove, tz.scene_remove_all, tz.read, tz.write,
                        ^
TypeError: Cannot read properties of undefined (reading 'push')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:78:25)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
```
I've downloaded all of the converters from https://github.com/Koenkk/zigbee-herdsman-converters)/lib/ (as well as to/from zigbee) and edited the paths accordingly. It looks like it's referencing files I cannot find in Github?

Any thoughts on how to proceed?

Many thanks again!

CarlosGomez-dev, 2022-12-13:
Are you using an external converter as described in https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html? You shouldn't need to download all converters, you should have one file with your external converter.

According to the error you're missing the `preventReset` function which is which is in the [xiaomi](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/xiaomi.js#L24) devices file, just copy/paste this function to wherever your converter is:

```js 
const preventReset = async (type, data, device) => {
    if (
        // options.allow_reset ||
        type !== 'message' ||
        data.type !== 'attributeReport' ||
        data.cluster !== 'genBasic' ||
        !data.data[0xfff0] ||
        // eg: [0xaa, 0x10, 0x05, 0x41, 0x87, 0x01, 0x01, 0x10, 0x00]
        !data.data[0xFFF0].slice(0, 5).equals(Buffer.from([0xaa, 0x10, 0x05, 0x41, 0x87]))
    ) {
        return;
    }
    const options = {manufacturerCode: 0x115f};
    const payload = {[0xfff0]: {
        value: [0xaa, 0x10, 0x05, 0x41, 0x47, 0x01, 0x01, 0x10, 0x01],
        type: 0x41,
    }};
    await device.getEndpoint(1).write('genBasic', payload, options);
};
```

Not sure if that'll fix the 2nd error, if it doesn't I would suggest setting up an external converter following the [z2m guide](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html).

closed 2023-01-21 by github-actions[bot]

nvp760, 2023-04-25:
Hi.. got any progress on this device? Still unsupported by Z2M.

karrui, 2023-11-29:
This just started happening to me after I unpaired and re-paired the device recently. 
The decoupled mode no longer does anything, and the power/energy values are no longer being communicated:

[image]
[image]

anyone else seeing this and know how to fix it? 

seems similar to https://github.com/Koenkk/zigbee2mqtt/issues/17305. I'm already using SkyConnect with the latest firmware, and it WAS working before I tried to re-pair it (due to a stuck firmware OTA update that was not actually available). Could it be due to a failed firmware update?

karrui, 2023-12-09:
Seems to have been fixed with the following external converter retrieved from https://github.com/Koenkk/zigbee2mqtt/issues/12029

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const herdsman = require('zigbee-herdsman');
const {
    precisionRound, mapNumberRange, isLegacyEnabled, toLocalISOString, numberWithinRange, hasAlreadyProcessedMessage,
    calibrateAndPrecisionRoundOptions, addActionGroup, postfixWithEndpointName, getKey,
    batteryVoltageToPercentage, getMetaValue,
} = require('zigbee-herdsman-converters/lib/utils');
const manufacturerOptions = {
    xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},
};
const tzLocal = {
    xiaomi_switch_power_outage_memory: {
        key: ['power_outage_memory'],
        convertSet: async (entity, key, value, meta) => {
//add 'QBKG32LM' to line 2123 in toZigbee.js
            if (['SP-EUC01', 'ZNCZ04LM', 'ZNCZ15LM', 'QBCZ14LM', 'QBCZ15LM', 'SSM-U01', 'SSM-U02', 'DLKZMK11LM', 'DLKZMK12LM',
                'WS-EUK01', 'WS-EUK02', 'WS-EUK03', 'WS-EUK04', 'QBKG19LM', 'QBKG20LM', 'QBKG25LM', 'QBKG26LM',
                'QBKG31LM', 'QBKG32LM', 'QBKG34LM', 'QBKG38LM', 'QBKG39LM', 'QBKG40LM', 'QBKG41LM', 'ZNDDMK11LM'].includes(meta.mapped.model)) {
                await entity.write('aqaraOpple', {0x0201: {value: value ? 1 : 0, type: 0x10}}, manufacturerOptions.xiaomi);
            } else if (['ZNCZ02LM', 'QBCZ11LM', 'LLKZMK11LM'].includes(meta.mapped.model)) {
                const payload = value ?
                    [[0xaa, 0x80, 0x05, 0xd1, 0x47, 0x07, 0x01, 0x10, 0x01], [0xaa, 0x80, 0x03, 0xd3, 0x07, 0x08, 0x01]] :
                    [[0xaa, 0x80, 0x05, 0xd1, 0x47, 0x09, 0x01, 0x10, 0x00], [0xaa, 0x80, 0x03, 0xd3, 0x07, 0x0a, 0x01]];

                await entity.write('genBasic', {0xFFF0: {value: payload[0], type: 0x41}}, manufacturerOptions.xiaomi);
                await entity.write('genBasic', {0xFFF0: {value: payload[1], type: 0x41}}, manufacturerOptions.xiaomi);
            } else if (['ZNCZ11LM'].includes(meta.mapped.model)) {
                const payload = value ?
                    [0xaa, 0x80, 0x05, 0xd1, 0x47, 0x00, 0x01, 0x10, 0x01] :
                    [0xaa, 0x80, 0x05, 0xd1, 0x47, 0x01, 0x01, 0x10, 0x00];

                await entity.write('genBasic', {0xFFF0: {value: payload, type: 0x41}}, manufacturerOptions.xiaomi);
            } else {
                throw new Error('Not supported');
            }
            return {state: {power_outage_memory: value}};
        },
        convertGet: async (entity, key, meta) => {
//add 'QBKG32LM' to line 2146 in toZigbee.js
            if (['SP-EUC01', 'ZNCZ04LM', 'ZNCZ15LM', 'QBCZ14LM', 'QBCZ15LM', 'SSM-U01', 'SSM-U02', 'DLKZMK11LM', 'DLKZMK12LM',
                'WS-EUK01', 'WS-EUK02', 'WS-EUK03', 'WS-EUK04', 'QBKG19LM', 'QBKG20LM', 'QBKG25LM', 'QBKG26LM',
                'QBKG31LM', 'QBKG32LM', 'QBKG34LM', 'QBKG38LM', 'QBKG39LM', 'QBKG40LM', 'QBKG41LM', 'ZNDDMK11LM'].includes(meta.mapped.model)) {
                await entity.read('aqaraOpple', [0x0201]);
            } else if (['ZNCZ02LM', 'QBCZ11LM', 'ZNCZ11LM'].includes(meta.mapped.model)) {
                await entity.read('aqaraOpple', [0xFFF0]);
            } else {
                throw new Error('Not supported');
            }
        },
    },
    xiaomi_led_disabled_night: {
      key: ['led_disabled_night'],
//add 'QBKG32LM' to line 2256 in toZigbee.js
      convertSet: async (entity, key, value, meta) => {
          if (['ZNCZ04LM', 'ZNCZ15LM', 'QBCZ14LM', 'QBCZ15LM', 'QBKG19LM', 'QBKG20LM', 'QBKG25LM', 'QBKG26LM',
              'QBKG31LM', 'QBKG32LM', 'QBKG34LM', 'DLKZMK11LM', 'SSM-U01'].includes(meta.mapped.model)) {
              await entity.write('aqaraOpple', {0x0203: {value: value ? 1 : 0, type: 0x10}}, manufacturerOptions.xiaomi);
          } else if (['ZNCZ11LM'].includes(meta.mapped.model)) {
              const payload = value ?
                  [0xaa, 0x80, 0x05, 0xd1, 0x47, 0x00, 0x03, 0x10, 0x00] :
                  [0xaa, 0x80, 0x05, 0xd1, 0x47, 0x01, 0x03, 0x10, 0x01];
[... 61 lines cut ...]
    model: 'QBKG32LM',
    vendor: 'Xiaomi',
    description: 'Aqara smart wall switch H1 Pro (with neutral, triple rocker)',
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {'left': 1, 'center': 2, 'right': 3};
    },
    fromZigbee: [fz.on_off, fz.xiaomi_power, fz.aqara_opple, fz.xiaomi_multistate_action],
    toZigbee: [tz.on_off, tz.xiaomi_switch_operation_mode_opple, tz.xiaomi_switch_power_outage_memory,
        tz.xiaomi_led_disabled_night, tz.xiaomi_flip_indicator_light],
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('center'), e.switch().withEndpoint('right'), e.power(), e.energy(), e.voltage(),
        e.device_temperature(), e.power_outage_memory(), e.led_disabled_night(), e.flip_indicator_light(),
        e.action([
            'single_left', 'single_center', 'single_right', 'single_all',
            'double_left', 'double_center', 'double_right', 'double_all']),
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
            .withDescription('Decoupled mode for left button').withEndpoint('left'),
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
            .withDescription('Decoupled mode for center button').withEndpoint('center'),
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
            .withDescription('Decoupled mode for right button').withEndpoint('right')],
}

module.exports = definition;
```