Koenkk/zigbee2mqtt #15450: [New device support]: LSC (Action) Radiator Valve TRV _TZE200_sur6q7ko

zz04303, 2022-12-08:
### Link

https://shop.action.com/nl-nl/p/8712879156871/lsc-smart-connect-heating-set-radiatorknop

### Database entry

{"id":4,"type":"EndDevice","ieeeAddr":"0x847127fffe72fecc","nwkAddr":61124,"manufId":4098,"manufName":"_TZE200_sur6q7ko","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"ï¿½ï¿½$+\u0011","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_sur6q7ko","powerSource":3,"zclVersion":3,"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{}},"genDeviceTempCfg":{"attributes":{}},"genScenes":{"attributes":{"currentScene":0}},"genTime":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1670527795838,"defaultSendRequestWhen":"immediate"}

### Comments

This device seems to be equivalent to:

       fingerprint: tuya.fingerprint('TS0601', [
            '_TZE200_hue3yfsn', /* model: 'TV02-Zigbee', vendor: 'TuYa' */

I've copied '_TZE200_hue3yfsn' its properties to an external converter, adjusted it to _TZE200_sur6q7ko and is now operational.

Zigbee2MQTT 1.28.4-1 
Home Assistant 2022.9.7
Supervisor 2022.11.2
Operating System 9.0
Frontend 20220907.2 - latest

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_sur6q7ko',
        },
    ],
    model: 'TS0601_sur6q7ko',
    vendor: 'TuYa',
    description: 'Thermostat radiator valve',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.battery_low(), e.child_lock(), e.open_window(), e.open_window_temperature().withValueMin(5).withValueMax(30),
        e.comfort_temperature().withValueMin(5).withValueMax(30), e.eco_temperature().withValueMin(5).withValueMax(30),
        tuya.exposes.errorStatus(), tuya.exposes.frostProtection('The device display "AF", press the pair button to cancel.'),
        exposes.climate()
            .withSystemMode(['off', 'heat'], ea.STATE_SET, 'When switched to the "off" mode, the device will display ' +
                '"HS" and the valve will be fully closed. Press the pair button to cancel or switch back to "heat" mode. Battery life' +
                ' can be prolonged by switching the heating off. After switching to `heat` mode, `preset` will be reset to `auto` and' +
                ' after changing `preset` to `manual` temperature setpoint will be 20 degrees.')
            .withPreset(['auto', 'manual', 'holiday'],
                '`auto` uses schedule properties, check them. `manual` allows you to control the device, `holiday` uses ' +
                '`holiday_start_stop` and `holiday_temperature` properties.')
            .withLocalTemperatureCalibration(-5, 5, 0.1, ea.STATE_SET)
            .withLocalTemperature(ea.STATE)
            .withSetpoint('current_heating_setpoint', 5, 30, 0.5, ea.STATE_SET),
        exposes.numeric('boost_timeset_countdown', ea.STATE_SET).withUnit('second').withDescription('Setting '+
[... 43 lines cut ...]
            [40, 'child_lock', tuya.valueConverter.lockUnlock],
            [45, 'error_status', tuya.valueConverter.raw],
            [46, 'holiday_start_stop', tuya.valueConverter.thermostatHolidayStartStop],
            [101, 'boost_timeset_countdown', tuya.valueConverter.raw],
            [102, 'open_window_temperature', tuya.valueConverter.divideBy10],
            [104, 'comfort_temperature', tuya.valueConverter.divideBy10],
            [105, 'eco_temperature', tuya.valueConverter.divideBy10],
            [106, 'schedule', tuya.valueConverter.thermostatScheduleDaySingleDP],
            [107, null, tuya.valueConverter.TV02SystemMode],
            [107, 'system_mode', tuya.valueConverterBasic.lookup({'heat': false, 'off': true})],
            [107, 'heating_stop', tuya.valueConverter.onOff],
            [115, 'online', tuya.valueConverter.onOffNotStrict],
            [108, 'schedule_monday', tuya.valueConverter.thermostatScheduleDaySingleDP],
            [112, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDaySingleDP],
            [109, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDaySingleDP],
            [113, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDaySingleDP],
            [110, 'schedule_friday', tuya.valueConverter.thermostatScheduleDaySingleDP],
            [114, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDaySingleDP],
            [111, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDaySingleDP],            
        ],
    },
};

module.exports = definition;
```

zz04303, 2022-12-08:
When still in original Tuya app:
ZigBee Module: V1.05
MCU Module: V2.0.5
No updates available

Koenkk, 2022-12-09:
Added, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-09 by Koenkk