Koenkk/zigbee2mqtt #15458: Z2M crash every morning

xm4rcell0x, 2022-12-09:
### What happened?

It's 4 or 5 days that i have installed z2m and every morning my entities on homeassistant aren't avaiable, that's because z2m stop to work properly.
If a try to open the frontend, i can't do it
If i restart the docker container manually it works without problem and i can access to the frontend.

### What did you expect to happen?

i expect that i don't have to manually restart z2m every morning.

### How to reproduce it (minimal and precise)

I can't reproduce this problem manually.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

Sonoff Dongle-E

### Debug log

**Additional context**
[z2m issue logs](https://drive.google.com/drive/folders/1jdS9ZyjI79l3Y9ZEzMUek9V1IQRpiI3w?usp=share_link)

Damianod27, 2022-12-09:
@xm4rcell0x Try to download the software from this page.  Of course in the same version :  ncp-uart-sw_v6.10.3_115200.gbl

I have no problems with it in Sonoff Dongle-E

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP

xm4rcell0x, 2022-12-10:
Uh, ok.
That's a modded firmware?

Damianod27, 2022-12-11:
@xm4rcell0x 

I don't know for sure but I suppose so.  After downloading this software, all my problems stopped.  You can always go back to what you are using if you have other or worse problems.

xm4rcell0x, 2022-12-12:
Hi @Damianod27 
I have not updated yet my Dongle E because with new Z2M's dev version seem to fix my issue. 

Now I want to try to update the firmware and to roll-back to the latest stable version of Z2M as you suggested me. 
I'll keep you updated in the next days.
Thank you so much for your help.

xm4rcell0x, 2022-12-12:
@Damianod27 i have just updated the sonoff firmware with the xps1989's one, but on the z2m frontend i see the same coordinator revision and the same build. Can you confirm that it's correct to have the 297 build ?

Damianod27, 2022-12-12:
@xm4rcell0x I'm not home until Friday, so I can't confirm it for you.  I uploaded the exact NCP-UART-SW_v6.10.3_115200.gbl file.  As you can see on the server there is the last update 3 months ago.  So it must be the same file because there was no update.  I won it for Dongle-E three weeks ago.  I haven't had any problems since then.  Send me the contents of the configuration file.

xm4rcell0x, 2022-12-12:
Ok, i uploaded this exact firmware. 
I'll keep you updated in the next days to see if the error will occur again.
the configuration file hasn't changed, and it's [here](https://drive.google.com/drive/folders/1jdS9ZyjI79l3Y9ZEzMUek9V1IQRpiI3w?usp=share_link)

xm4rcell0x, 2022-12-17:
I can confirm that the dev's versions fix my problem.

xm4rcell0x, 2023-01-07:
This issue has not been resolved. I keep having problems since that day, and the only way to fix it is to manually restart the container. 
There's another user with my exact problem #16000 

I'have updated the logs with the newer ones [Z2M Logs](https://drive.google.com/drive/folders/1jdS9ZyjI79l3Y9ZEzMUek9V1IQRpiI3w?usp=share_link)  It starts every couple of days with this:
`debug 2023-01-07 05:59:17: Active device 'Striscia Pedonale' was last seen '29940.87' minutes ago.
warn  2023-01-07 05:59:29: Failed to ping 'Striscia Cappa Cucina' (attempt 1/1, Read 0xa4c138f77ad5bd03/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 2101 - 1 - 151 - 0 - 1 after 10000ms))
debug 2023-01-07 05:59:32: Active device 'Striscia Cappa Cucina' was last seen '312.85' minutes ago.
debug 2023-01-07 05:59:43: Received Zigbee message from 'Lavatrice', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
info  2023-01-07 05:59:43: MQTT publish: topic 'zigbee2mqtt/Lavatrice', payload '{"child_lock":"UNLOCK","current":0,"energy":31.63,"indicator_mode":"off/on","last_seen":"2023-01-07T05:59:43+01:00","linkquality":136,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":235}'
warn  2023-01-07 05:59:44: Failed to ping '0xa4c138eae1e3d5cc' (attempt 1/1, Read 0xa4c138eae1e3d5cc/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 44581 - 1 - 152 - 0 - 1 after 10000ms))
debug 2023-01-07 05:59:47: Active device '0xa4c138eae1e3d5cc' was last seen '29685.41' minutes ago.
debug 2023-01-07 05:59:57: Received Zigbee message from 'Lampada Salone', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2023-01-07 05:59:57: MQTT publish: topic 'zigbee2mqtt/Lampada Salone', payload '{"last_seen":"2023-01-07T05:59:57+01:00","linkquality":120,"state":"OFF"}'
warn  2023-01-07 05:59:59: Failed to ping '0xa4c13804a90c773a' (attempt 1/1, Read 0xa4c13804a90c773a/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 57278 - 1 - 153 - 0 - 1 after 10000ms))
debug 2023-01-07 06:00:02: Active device '0xa4c13804a90c773a' was last seen '25543.29' minutes ago.
debug 2023-01-07 06:00:09: Received Zigbee message from 'Macchinetta del caffÃ¨', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":234}' from endpoint 1 with groupID 0
info  2023-01-07 06:00:09: MQTT publish: topic 'zigbee2mqtt/Macchinetta del caffÃ¨', payload '{"child_lock":"UNLOCK","current":0,"energy":3.97,"indicator_mode":"off/on","last_seen":"2023-01-07T06:00:09+01:00","linkquality":132,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":234}'
debug 2023-01-07 06:00:09: Received Zigbee message from 'Macchinetta del caffÃ¨', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,397]}' from endpoint 1 with groupID 0
info  2023-01-07 06:00:09: MQTT publish: topic 'zigbee2mqtt/Macchinetta del caffÃ¨', payload '{"child_lock":"UNLOCK","current":0,"energy":3.97,"indicator_mode":"off/on","last_seen":"2023-01-07T06:00:09+01:00","linkquality":136,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":234}'
debug 2023-01-07 06:00:10: Received MQTT message on 'homeassistant/status' with data 'offline'
debug 2023-01-07 06:00:11: Received Zigbee message from 'Intrattenimento Salone', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":8,"rmsCurrent":93,"rmsVoltage":233}' from endpoint 1 with groupID 0
error 2023-01-07 06:00:11: Not connected to MQTT server!
error 2023-01-07 06:00:11: Cannot send message: topic: 'zigbee2mqtt/Intrattenimento Salone', payload: '{"child_lock":"UNLOCK","current":0.09,"energy":18.8,"indicator_mode":"off/on","last_seen":"2023-01-07T06:00:11+01:00","linkquality":168,"power":8,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":233}
debug 2023-01-07 06:00:11: Received Zigbee message from 'Intrattenimento Salone', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1880]}' from endpoint 1 with groupID 0
error 2023-01-07 06:00:11: Not connected to MQTT server!
error 2023-01-07 06:00:11: Cannot send message: topic: 'zigbee2mqtt/Intrattenimento Salone', payload: '{"child_lock":"UNLOCK","current":0.09,"energy":18.8,"indicator_mode":"off/on","last_seen":"2023-01-07T06:00:11+01:00","linkquality":168,"power":8,"power_outage_memory": [... cut ...]
debug 2023-01-07 09:21:10: Saving state to file /app/data/state.json
error 2023-01-07 09:24:13: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of null (reading 'clients')
    at Frontend.stop (/app/lib/extension/frontend.ts:83:39)
    at Controller.callExtensions (/app/lib/controller.ts:316:40)
    at Controller.stop (/app/lib/controller.ts:193:9)
    at stop (/app/index.js:111:5)
    at process.handleQuit (/app/index.js:117:9))
`
Container keep running but devices and frontend are unresponsive.

kizio1000, 2023-01-07:
Hi, in my case there is no indication what is going on. I do not have in logs any problem before z2m is getting stuck.

Last entires are ending with:

`debug 2023-01-07 00:06:25: Received Zigbee message from 'Zmywarka', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
debug 2023-01-07 00:06:25: Received Zigbee message from 'Termometr strych', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":1131}' from endpoint 1 with groupID 0
info  2023-01-07 00:06:25: MQTT publish: topic 'zigbee2mqtt/Termometr strych', payload '{"battery":80,"humidity":60.93,"linkquality":196,"temperature":11.31,"voltage":2900}'`

Then in next log file:
`debug 2023-01-07 00:06:46: Loaded state from file /config/zigbee2mqtt/state.json
info  2023-01-07 00:06:46: Logging to console and directory: '/config/zigbee2mqtt/log/2023-01-07.00-06-46' filename: log.txt
debug 2023-01-07 00:06:46: Removing old log directory '/config/zigbee2mqtt/log/2023-01-04.00-08-26'
info  2023-01-07 00:06:46: Starting Zigbee2MQTT version 1.29.1 (commit #unknown)
info  2023-01-07 00:06:46: Starting zigbee-herdsman (0.14.83)
debug 2023-01-07 00:06:46: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220719140040-if00"}}'
debug 2023-01-07 00:11:46: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 00:16:46: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 00:21:46: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 00:26:46: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 00:31:46: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 00:36:46: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 00:41:46: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 00:46:46: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 00:51:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 00:56:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:01:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:06:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:11:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:16:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:21:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:26:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:31:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:36:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:41:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:46:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:51:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 01:56:47: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:01:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:06:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:11:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:16:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:21:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:26:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:31:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:36:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:41:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:46:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:51:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 02:56:48: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:01:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:06:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:11:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:16:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:21:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:26:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:31:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:36:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:41:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:46:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:51:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 03:56:49: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 04:01:50: Saving state to file /config/zigbee2mqtt/state.json
[... cut ...]
debug 2023-01-07 06:06:51: Saving state to file /config/zigbee2mqtt/state.json
debug 2023-01-07 06:11:52: Saving state to file /config/zigbee2mqtt/state.json
error 2023-01-07 06:12:06: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of null (reading 'clients')
    at Frontend.stop (/app/lib/extension/frontend.ts:83:39)
    at Controller.callExtensions (/app/lib/controller.ts:316:40)
    at Controller.stop (/app/lib/controller.ts:193:9)
    at stop (/app/index.js:111:5)
    at process.handleQuit (/app/index.js:117:9))`

xm4rcell0x, 2023-01-07:
I have updated the permission on GDrive. 
Anyway, before 1.28 so in November, do you had any problem?
I started with 1.28 so I cannot say how it works with previous versions

kizio1000, 2023-01-07:
Before 1.28 system was ok, it started after installing 1.28. and continued on 1.29 and 1.29.1

xm4rcell0x, 2023-01-07:
Thanks for your informations
We can continue on your issue, maybe it's more visible than this.

closed 2023-02-24 by github-actions[bot]