Koenkk/zigbee2mqtt #15460: Eco-Dim.07 timeouting

attiolli, 2022-12-09:
### What happened?

Takes very long time to send a light toggle / dim command to the Eco-Dim.07 device.

### What did you expect to happen?

The command should activate light / dimming almost immediately as usual.

### How to reproduce it (minimal and precise)

Pair the Eco-Dim.07 to the z2m. Start using the light and try for eg. to toggle ON/OFF .

### Zigbee2MQTT version

1.28.4 

### Adapter firmware version

zStack3x0 v.20220219

### Adapter

SMLIGHT SLZB-05 Zigbee ethernet

### Debug log

# Toggled the ecodim to go OFF. Send command through z2m web interface or from HomeAssistant and after waiting several seconds (sometimes even maby 10-20s) this appears to the log:
Zigbee2MQTT:error 2022-12-09 13:44:16: Publish 'set' 'state' to 'makuuhuoneen-kattovalot' failed: 'Error: Command 0x385b44fffe1b3a1d/1 genOnOff.on({},
{"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
failed (Timeout - 41005 - 1 - 9 - 6 - 11 after 10000ms)'

# Then the ecodim device actually gets commanded and does its thing:
Zigbee2MQTT:info  2022-12-09 13:44:16: MQTT publish: topic 'zigbee2mqtt/makuuhuoneen-kattovalot', payload '{"brightness":246,"last_seen":"2022-12-09T13:44:16+02:00","linkquality":83,"state":"OFF"}'
Zigbee2MQTT:info  2022-12-09 13:44:17: MQTT publish: topic 'zigbee2mqtt/makuuhuoneen-kattovalot', payload '{"brightness":0,"last_seen":"2022-12-09T13:44:17+02:00","linkquality":83,"state":"OFF"}'

Other devices (about 32 devices) are working in the network just fine.

attiolli, 2022-12-11:
Investigated a bit more this case... Actually the first command for eg. toggle works fine and the light obeys it immediately, but after that, if I try to toggle it **again** lets say within 20-30sec the timeout problem happens (log prints out the timeout error which I posted).

attiolli, 2022-12-12:
Tried also with the coordinator on the same room next to the dimmer. Same timeout issue happens. Also tried to force the pairing straight to the coordinator to avoid any possibly misbehaving middle-man routers but no luck. Still timeouting.

closed 2023-01-20 by github-actions[bot]

ajin, 2024-10-14:
I’m having the same issue. @Dixet @attiolli, did you manage to fix it?

attiolli, 2024-10-14:
I don't have any news about this problem. It still exists in my environment
but not that big of a deal because the initial light toggle does not take
that long.  Still would appreciate it if someone finds a solution for
this issue. I remember that the EcoDim device was failing to answer within
the expected time window so chances are that this is some kind of
end-device failure / bug.

ajin, 2024-10-15:
After testing various options, I was able to solve this issue by setting the transition time to `1` in the `Settings (specific)` tab. Now the device is responding quickly.

attiolli, 2024-10-16:
After testing the same transition=1 parameter under settings(specific), I can confirm that also my devices are now working. The dimmer does not timeout anymore. Thank you for this discovery @ajin.

mmosquera91, 2024-11-04:
> After testing various options, I was able to solve this issue by setting the transition time to `1` in the `Settings (specific)` tab. Now the device is responding quickly.

You're a life saver :). I was about to send back my ecodim and start to look for another solution! works like a charm! thanks!