Koenkk/zigbee2mqtt #15469: Problem with functioning LED ZigBee + RF RGBW+CCT SD HUE TUYA LC031 (probably same as WZ5 5 in 1 LED Controller) recognized as TuYa TS0505B

MrLasiu, 2022-12-09:
### What happened?

As title states 
OXT LC031 RGBW+CCT controller was recognized as TuYa TS0505B 
Causes errors with color, they are not pure as well as brightness control when changing from color to white.
In Log i can see some missing data. 

found it only in PL store ;P
https://www.houseiq.pl/pl/p/OXT-Kontroler-LED-ZigBee-RF-RGBW-CCT-SD-HUE-TUYA/1831

i think this is similar to 
https://pl.aliexpress.com/item/1005003383051911.html?gatewayAdapt=glo2pol

### What did you expect to happen?

Maybe someone can change configuration or show me how to to that ;)

### Zigbee2MQTT version

1.28.4-1

### Adapter firmware version

20220219

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Debug log

2022-12-09 21:18:50Device '0xa4c1380f494d0186' announced itself
2022-12-09 21:18:51Received Zigbee message from '0xa4c1380f494d0186', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
2022-12-09 21:18:51Received Zigbee message from '0xa4c1380f494d0186', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
2022-12-09 21:18:51Received Zigbee message from '0xa4c1380f494d0186', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
2022-12-09 21:18:51Received Zigbee message from '0xa4c1380f494d0186', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorCapabilities":27}' from endpoint 1 with groupID 0
2022-12-09 21:18:51Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0xa4c1380f494d0186!
2022-12-09 21:18:51Received Zigbee message from '0xa4c1380f494d0186', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorTempPhysicalMax":500,"colorTempPhysicalMin":153}' from endpoint 1 with groupID 0

krzysztof-cislo, 2023-01-07:
Problem exists also in the 1.29.1 version although in this version the device is badly recognized as Xiaomi ZNCZ04LM.

**Image**

[image]

**Logs**

```console

Info 2023-01-07 09:46:57Zigbee: allowing new devices to join.
Info 2023-01-07 09:46:57MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"nqxh3-1"}'
Info 2023-01-07 09:46:58Device '0x04cf8cdf3c8a28e4' joined
Info 2023-01-07 09:46:58MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c8a28e4","ieee_address":"0x04cf8cdf3c8a28e4"},"type":"device_joined"}'
Info 2023-01-07 09:46:58MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x04cf8cdf3c8a28e4"},"type":"device_connected"}'
Info 2023-01-07 09:46:58MQTT publish: topic 'homeassistant/switch/0x04cf8cdf3c8a28e4/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x04cf8cdf3c8a28e4/set","device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c8a28e4"],"manufacturer":"Xiaomi","model":"Mi power plug ZigBee EU (ZNCZ04LM)","name":"0x04cf8cdf3c8a28e4"},"json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c8a28e4","name":"0x04cf8cdf3c8a28e4","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x04cf8cdf3c8a28e4","unique_id":"0x04cf8cdf3c8a28e4_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Info 2023-01-07 09:46:58MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c8a28e4/power/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c8a28e4"],"manufacturer":"Xiaomi","model":"Mi power plug ZigBee EU (ZNCZ04LM)","name":"0x04cf8cdf3c8a28e4"},"device_class":"power","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c8a28e4","name":"0x04cf8cdf3c8a28e4 power","state_class":"measurement","state_topic":"zigbee2mqtt/0x04cf8cdf3c8a28e4","unique_id":"0x04cf8cdf3c8a28e4_power_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.power }}"}'
Info 2023-01-07 09:46:58MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c8a28e4/energy/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c8a28e4"],"manufacturer":"Xiaomi","model":"Mi power plug ZigBee EU (ZNCZ04LM)","name":"0x04cf8cdf3c8a28e4"},"device_class":"energy","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c8a28e4","name":"0x04cf8cdf3c8a28e4 energy","state_class":"total_increasing","state_topic":"zigbee2mqtt/0x04cf8cdf3c8a28e4","unique_id":"0x04cf8cdf3c8a28e4_energy_zigbee2mqtt","unit_of_measurement":"kWh","value_template":"{{ value_json.energy }}"}'
Info 2023-01-07 09:46:58MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c8a28e4/device_temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c8a28e4"],"manufacturer":"Xiaomi","model":"Mi power plug ZigBee EU (ZNCZ04LM)","name":"0x04cf8cdf3c8a28e4"},"device_class":"temperature","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c8a28e4","name":"0x04cf8cdf3c8a28e4 device temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0x04cf8cdf3c8a28e4","unique_id":"0x04cf8cdf3c8a28e4_device_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.device_temperature }}"}'
Info 2023-01-07 09:46:58MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c8a28e4/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c8a28e4"],"manufacturer":"Xiaomi","model":"Mi power plug ZigBee EU (ZNCZ04LM)","name":"0x04cf8cdf3c8a28e4"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c8a28e4","name":"0x04cf8cdf3c8a28e4 voltage","state_class":"measurement","state_topic":"zigbee2mqtt/0x04cf8cdf3c8a28e4","unique_id":"0x04cf8cdf3c8a28e4_voltage_zigbee2mqtt","unit_of_measurement":"V","value_template":"{{ value_json.voltage }}"}'
Info 2023-01-07 09:46:58MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c8a28e4/current/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c8a28e4"],"manufacturer":"Xiaomi","model":"Mi power plug ZigBee EU (ZNCZ04LM)","name":"0x04cf8cdf3c8a28e4"},"device_class":"current","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c8a28e4","name":"0x04cf8cdf3c8a28e4 current","state_class":"measurement","state_topic":"zigbee2mqtt/0x04cf8cdf3c8a28e4","unique_id":"0x04cf8cdf3c8a28e4_current_zigbee2mqtt","unit_of_measurement":"A","value_template":"{{ value_json.current }}"}'
Info 2023-01-07 09:46:58MQTT publish: topic 'homeassistant/binary_sensor/0x04cf8cdf3c8a28e4/consumer_connected/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c8a28e4"],"manufacturer":"Xiaomi","model":"Mi power plug ZigBee EU (ZNCZ04LM)","name":"0x04cf8cdf3c8a28e4"},"device_class":"connectivity","entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c8a28e4","name":"0x04cf8cdf3c8a28e4 consumer connected","payload_off":false,"payload_on":true,"state_topic":" [... cut ...] device.","name":"current_calibration","property":"current_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for current, takes into effect on next report of device.","name":"current_precision","property":"current_precision","type":"numeric","value_max":3,"value_min":0}],"supports_ota":true,"vendor":"Xiaomi"},"friendly_name":"0x04cf8cdf3c8a28e4","ieee_address":"0x04cf8cdf3c8a28e4","status":"successful","supported":true},"type":"device_interview"}'

```

durkdeboer, 2023-01-24:
same problem here with the wz5 controller in rgbw mode was recognized as TuYa TS0504B (not as TS0505B

doggogithub, 2023-02-05:
Hello

I have similar issue with 

OXT LC031 RGBW+CCT controller was recognized as TuYa TS0505B

I can not change colours, they are slighty off, like red is orange, blue is not blue, green is, well hard to describe.
There are no pure colours.

```
Info 2023-02-04 00:14:46Device '0xa3c2530f632d0883' joined
Info 2023-02-04 00:14:46MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa3c2530f632d0883","ieee_address":"0xa3c2530f632d0883"},"type":"device_joined"}'
Info 2023-02-04 00:14:46Starting interview of '0xa3c2530f632d0883'
Info 2023-02-04 00:14:46MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa3c2530f632d0883","ieee_address":"0xa3c2530f632d0883","status":"started"},"type":"device_interview"}'
Info 2023-02-04 00:14:48Successfully interviewed '0xa3c2530f632d0883', device has successfully been paired
Info 2023-02-04 00:14:48Device '0xa3c2530f632d0883' is supported, identified as: TuYa Zigbee RGB+CCT light (TS0505B)
Info 2023-02-04 00:14:48MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Zigbee RGB+CCT light","exposes":[{"features":[{"access":7,"description":"On/off state of this light","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0},{"access":7,"description":"Color temperature of this light","name":"color_temp","presets":[{"description":"Coolest temperature supported","name":"coolest","value":153},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500}],"property":"color_temp","type":"numeric","unit":"mired","value_max":500,"value_min":153},{"description":"Color of this light in the CIE 1931 color space (x/y)","features":[{"access":7,"name":"x","property":"x","type":"numeric"},{"access":7,"name":"y","property":"y","type":"numeric"}],"name":"color_xy","property":"color","type":"composite"}],"type":"light"},{"access":2,"description":"Triggers an effect on the light (e.g. make light blink for a few seconds)","name":"effect","property":"effect","type":"enum","values":["blink","breathe","okay","channel_change","finish_effect","stop_effect"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0505B","options":[{"access":2,"description":"Controls the transition time (in seconds) of on/off, brightness, color temperature (if applicable) and color (if applicable) changes. Defaults to 0 (no transition).","name":"transition","property":"transition","type":"numeric","value_min":0},{"access":2,"description":"When enabled colors will be synced, e.g. if the light supports both color x/y and color temperature a conversion from color x/y to color temperature will be done when setting the x/y color (default true).","name":"color_sync","property":"color_sync","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"TuYa"},"friendly_name":"0xa3c2530f632d0883","ieee_address":"0xa3c2530f632d0883","status":"successful","supported":true},"type":"device_interview"}'
Info 2023-02-04 00:14:48Configuring '0xa3c2530f632d0883'
Info 2023-02-04 00:14:48MQTT publish: topic 'zigbee2mqtt/0xa3c2530f632d0883', payload '{"brightness":86,"color":{"h":32,"hue":32,"s":82,"saturation":82,"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"linkquality":105,"state":"OFF"}'
Info 2023-02-04 00:14:48MQTT publish: topic 'zigbee2mqtt/0xa3c2530f632d0883', payload '{"brightness":86,"color":{"h":32,"hue":32,"s":82,"saturation":82,"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"linkquality":109,"state":"OFF"}'
Info 2023-02-04 00:14:48Successfully configured '0xa3c2530f632d0883'

Error 2023-02-04 00:34:27Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'hasOwnProperty')}
```

closed 2023-03-16 by github-actions[bot]

jonwilliams84, 2023-03-21:
I have the same issue, namely colours being "off"...e.g. red is more orange, blue is closer to cyan, purple is pink etc

Sangoku, 2023-06-09:
I thing there is a general problem here.. i have THE described https://www.aliexpress.com/item/1005003469344159.html and have the same problem..

szymonzdziabek, 2023-07-12:
The same here. The OXT LC031 LED RGB+CCT controller bought in Poland, currently recognized in Zigbee2MQTT as Skydance WZ5_dim_2, however I'm pretty sure that was shown as another one in the past. Anyway, now I see Skydance WZ5_dim_2 and the Zigbee model TS0505B (_TZB210_3zfp8mki).
1) The problem is about color accuracy. All of them are not so bright, not so pure as I can set up using RF remote controller attached to the LC031. Orange is violet, red is pink-orange:
[image]
[image]

2) Additionally I'm not sure if switching between white (CCT) and RGB mode is functioning properly. Using the brightness slider in HA sometimes causes switching to CCT mode and sometimes not.
3) Effects (blink, breathe and so on) are not working anymore, however I think I play with them around 9 months ago. I can run them using RF.