Koenkk/zigbee2mqtt #15470: [New device support]:  CK-BL702-AL-01(7009_Z102LG03-1)

sango74, 2022-12-09:
### Link

https://fr.aliexpress.com/item/1005003241784242.html

### Database entry

{"id":17,"type":"Router","ieeeAddr":"0xb4e84286708e0000","nwkAddr":53643,"manufId":4742,"manufName":"eWeLink","powerSource":"Unknown","modelId":"CK-BL702-AL-01(7009_Z102LG03-1)","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClu sterList":[0,3,4,5,6,8,768,4096,61184,64529,64599],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":0,"hwVersion":0,"dateCode ":"20221025","swBuildId":"1.2.0","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1670619695963,"defaultSendRequestWhen":"immediate"}

### Comments

Hello,
impossible to control this bulb.
Thanks in advance for the support

### Supported color modes

rgb

Koenkk, 2022-12-10:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-10 by Koenkk

sango74, 2022-12-11:
Hello, perfect !

RezzZ, 2024-01-25:
hello, anyone else getting offline devices for this model CK-BL702-AL-01(7009_Z102LG03-1) even though the lights still respond to every command send over mqtt (homeassistant)?

Br3Ak3R, 2024-06-09:
> hello, anyone else getting offline devices for this model CK-BL702-AL-01(7009_Z102LG03-1) even though the lights still respond to every command send over mqtt (homeassistant)?
> 
> edit: hmm now that I look more closely, even though the models are the same my light is actually different.. its a ewelink gu10 zigbee led bulb.. I'll try and ask for a new device support. Maybe that helps.

Can confirm same issue with my 5 bulbs i got this week from aliexpress.

https://de.aliexpress.com/item/1005006492947103.html

Sometimes they respond with delay even if it says offline. Now i have issue pairing one of them

ankokdog, 2024-09-30:
Same here. 6 χ E14 CK-BL702-AL-01(7009_Z102LG03-1) ewelink

ouiouiblog, 2024-10-07:
I have the same models, plugins says that devices are note optimized but all is working fine (except perhaps an important charge on statistics graphic)