Koenkk/zigbee2mqtt #15472: ti.router transmit poser off by 1

jerrm, 2022-12-10:
### What happened?

Set transmit power via frontend exposes panel to a value (ex: "7"), wait a few seconds, click the frontend "refresh" button, the value will be incremented by 1 (ex: "8").

### What did you expect to happen?

Values to stay as set in the frontend.

### How to reproduce it (minimal and precise)

See above,

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

Router 20221102, Coordinator 20220103

### Adapter

Sonoff ZBDongle-P

### Debug log

Info 2022-12-09 23:06:39MQTT publish: topic 'zigbee2mqtt/0x00124b0026b8822f', payload '{"last_seen":"2022-12-09T23:06:38-05:00","linkquality":240,"transmit_power":7}'
Debug 2022-12-09 23:07:29Received MQTT message on 'zigbee2mqtt/0x00124b0026b8822f/get' with data '{"transmit_power":""}'
Debug 2022-12-09 23:07:29Publishing get 'get' 'transmit_power' to '0x00124b0026b8822f'
Debug 2022-12-09 23:07:29Received Zigbee message from '0x00124b0026b8822f', type 'readResponse', cluster 'genBasic', data '{"4919":7}' from endpoint 8 with groupID 0
Info 2022-12-09 23:07:29MQTT publish: topic 'zigbee2mqtt/0x00124b0026b8822f', payload '{"last_seen":"2022-12-09T23:07:29-05:00","linkquality":244,"transmit_power":8}'

jerrm, 2023-01-10:
@Koenkk, can you comment?  

Should the refreshed value be off by one?  

To match the 20220125 tx power, should we set to 8 (reads back as 9) or 9 (reads back as 10).

Koenkk, 2023-01-11:
The set value should be +1, can you make a pr?

jerrm, 2023-01-12:
@Koenkk:

Looked at the converter and am confused where/what to increment.

If I'm reading correctly, fromZigbee `fzLocal.tirouter` is already increasing by +1.  Won't adding +1 at the set in `tzLocal.tirouter` make things worse?  A GUI setting of "9" be stored as "10" and then read back as "11"?

Do we need to +1 on one side and then -1 on the other?

Do I need to get more sleep?

```
const tzLocal = {
    tirouter: {
        key: ['transmit_power'],
        convertSet: async (entity, key, value, meta) => {
            await entity.write('genBasic', {0x1337: {value, type: 0x28}});
            return {state: {[key]: value}};
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('genBasic', [0x1337]);
        },
    },
```

```
const fzLocal = {
    tirouter: {
        cluster: 'genBasic',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {linkquality: msg.linkquality};
            if (msg.data['4919']) result['transmit_power'] = msg.data['4919'] + 1;
            return result;
        },
    },
```

```
module.exports = [
    {
        zigbeeModel: ['ti.router'],
        model: 'ti.router',
        vendor: 'Custom devices (DiY)',
        description: 'Texas Instruments router',
        fromZigbee: [fzLocal.tirouter],
        toZigbee: [tzLocal.tirouter],
        exposes: [exposes.numeric('transmit_power', ea.ALL).withValueMin(-20).withValueMax(20).withValueStep(1).withUnit('dBm')
            .withDescription('Transmit power, supported from firmware 20221102. The max for CC1352 is 20 dBm and 5 dBm for CC2652' +
                            ' (any higher value is converted to 5dBm)')],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(8);
            const payload = [{attribute: 'zclVersion', minimumReportInterval: 0, maximumReportInterval: 3600, reportableChange: 0}];
            await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
            await endpoint.configureReporting('genBasic', payload);
        },
    },
```

Koenkk, 2023-01-12:
Sorry  I was not clear, `if (msg.data['4919']) result['transmit_power'] = msg.data['4919'] + 1;` has to be changed to `if (msg.data['4919']) result['transmit_power'] = msg.data['4919'] ;`. Can you try if that works?

jerrm, 2023-01-12:
Yes.  It may not be until this weekend, but I will test and submit.

That makes more sense.  Pretty much what I suspected after actually looking at the converter, but I didn't know if the stick itself needed an altered by one value.  Didn't look like it did based on the firmware patch, but I had not had time to pull down the SDK to see the code behind the patch.

jerrm, 2023-01-15:
@Koenkk, PR tested and submitted.

closed 2023-01-15 by Koenkk