Koenkk/zigbee2mqtt #15473: OTA update for ecodim dimmerswitch 07/10 fails

Kilill, 2022-12-10:
### What happened?

Dev Branch, newly added OTA support for ecodim fails:
```
Info 2022-12-07 11:10:42MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Dimmer Up Livingroom Ceiling"},"error":"Failed to check if update available for 'Dimmer Up Livingroom Ceiling' (No image available for imageType '44')","status":"error","transaction":"esjdk-2"}'
Error 2022-12-07 11:10:42Failed to check if update available for 'Dimmer Up Livingroom Ceiling' (No image available for imageType '44')
```

### What did you expect to happen?

The OTA update should retreive the new firmware and execute an OTA update

### How to reproduce it (minimal and precise)

Have an ecodim 07 V3.0 dimmer switch and try to run an OTA update

### Zigbee2MQTT version

1.28.4-dev commit: d075f07

### Adapter firmware version

20220219

### Adapter

Sonof zigbee 3.0  USB dongle (zStack3x0)

Koenkk, 2022-12-10:
`No image available for imageType '44'` this means we don't have an OTA file available.

You can try requesting it from the manufacturer and add it to the [Zigbee-OTA repo](https://github.com/koenkk/zigbee-ota#adding-new-and-updating-existing-ota-files)

Kilill, 2022-12-10:
Eh.... 
it is there: https://github.com/Koenkk/zigbee-OTA/blob/master/images/Tuya/ZB21S3_HZC_Dimmer1_ECO_1.01_20221021.ota
i'm guessing that z2mqtt ota is confused somehow as to what file to look for....
not sure how to get a log of what its actually looking for ....

Koenkk, 2022-12-11:
Can you provide the debug log when checking for an update? The `imageType` (`44`) matches, not sure about the `manufacturerCode`

Kilill, 2022-12-13:
Hmm , the comment with the log seems to have been deleted ?
well here it is again

[log.zip](https://github.com/Koenkk/zigbee2mqtt/files/10216932/log.zip)

Koenkk, 2022-12-13:
The OTA in the repo is for `manufacturerCode` `4098`: https://github.com/Koenkk/zigbee-OTA/blob/31264c563907c9f0a045ed5fa1cf06f30466923c/index.json#L2215

The device requests `manufacturerCode` `4714`:

```
debug 2022-12-07 16:27:38: Got OTA request '{"fieldControl":0,"manufacturerCode":4714,"imageType":44,"fileVersion":50947376}'
```

So the OTA seems to be for a different device.

Kilill, 2022-12-15:
ahh.
Back to ECO Dim i go then :-)

erlendk, 2023-01-01:
@Kilill Not sure if it is possible to force an OTA-file upon a device it doesn't match with, but did you try updating the device locally before adding the OTA-file to the repo?https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files

Kilill, 2023-01-02:
ah forgot about that one  will try that one laters

Also, new file received from EcoDim today suposedly with corrected manufacturer code,  pull request on [zigbee-OTA](https://github.com/Koenkk/zigbee-OTA)

Kilill, 2023-01-04:
Hmm, right so after delving into the intricacies of header information of a zigbee-ota file (and learning more than I really wanted to know :-D )
The new file as an incorrectly formatted version value , at least compared to the original.
I have poked EcoDim again to see if they can come up with a correct file for the 3:d time.... in the mean time i'l try to force load it and see what happens, worst case I get a nice new brick.

closed 2023-02-12 by github-actions[bot]