Koenkk/zigbee2mqtt #15474: Data request failed with error: No network route' 205

Anjerlaan, 2022-12-10:
### What happened?

I have problems with my current Z2M installation on a Raspberry Pip 3b+ with Buster.
On my current SD card I wanted to update from 1.21 to 1.28 about 2 months ago, but I only got errors after the update.
I first updated the ZZH stick firmware to **zStack3x0 20220219**, which worked, but when I wanted to update Z2M to 1.28 I got errors which caused Z2M not to start.

Then I found out that I first should have update the Pi to Node-js to 16.2, which I did afterwards with success
But it was too late...I only got errors after trying to re-update to version 1.28.

Then I remembered that I had a Backup of my SD-Card from Januari 2022, which has version 1.17.2 installed.
I first updated the Pi Buster and upgraded Node-js to 16.2, which went well
Then I updated also to Z2M 1.28, also without any errors.
And I  copied a Z2M back-up of the other SD-Card back to /opt/zigbee2mqtt/data

[image]

And restarted Z2M.
But when I start the zigbee service, I see the devices in the Web Front End but they are not recognized because of the following message:

`failed (Data request failed with error: 'No network route' (205))'`

Is there a solution to save my Z2M network?

### Zigbee2MQTT version

v.1.28.4

### Adapter firmware version

zStack3x0 20220219

### Adapter

ZZH

a8ree, 2023-02-18:
@Anjerlaan Did you find a solution to this? I am running the ZZH stick and this morning all my devices are uncontactable with the no network route 205 issue.

Incidentally, I have never changed the firmware since I bought this stick in 2021 - do you know if there are any improvements in it?

Anjerlaan, 2023-02-18:
I first did a re-flash (I already had the latest) of the zzh firmware, which i read somewhere. 
But later I found out that it also happens when some of the lights in a group are offline (during christmas time my wife replaced 2 of 5 lights for christmas light, not zigbee), the other 3 lights in the group also reported 205 no network route errors. When all 5 lights were active in the group, the errors were also gone.