Koenkk/zigbee2mqtt #15481: Neo NAS-PD07 without values

flayy73, 2022-12-10:
### What happened?

I bought a pair of Neo NAS-PD07. Z2M pair correctly the devices but all the values are empty.
any advice on how to resolve 
TKS

### Zigbee2MQTT version

 1.28.4

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10200552/log.txt)

lucianoberger, 2022-12-18:
I am also having the same problem using the same hardware

SuperDobbo, 2022-12-18:
I have the same issue, Ive removed and readded it a bunch of times.

Koenkk, 2022-12-19:
Can you provide the data/database.db entry of this device?

SuperDobbo, 2022-12-19:
@Koenkk If you can tell me how to find the data/database.db entry I will provide it

SuperDobbo, 2022-12-19:
I think Ive found it, here is the device detail from the db:

{"id":26,"type":"EndDevice","ieeeAddr":"0xa4c1385c94bc732c","nwkAddr":51621,"manufId":4417,"manufName":"_TZ3000_kmh5qpmb","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_kmh5qpmb","zclVersion":3,"appVersion":67,"modelId":"TS0202","powerSource":3}},"ssIasZone":{"attributes":{"61441":1,"iasCieAddr":"0x00124b00257caf72","zoneState":1,"currentZoneSensitivityLevel":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":822316258},"lastSeen":1671482813647,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-12-20:
Does occupancy work  with the following external converter:  https://gist.github.com/Koenkk/e880d476d7267eb6c6b4caeac1c24f0c ?

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

SuperDobbo, 2022-12-20:
Yes presence works with that converter. The image has changed and the battery is not reporting it shows as none% (mine is cabled in so that may be the reason) and the temperature and humidity sensors have vanished.

Koenkk, 2022-12-20:
Updated https://gist.github.com/Koenkk/e880d476d7267eb6c6b4caeac1c24f0c to add humidity and temperature. It may take up to 24 hours beofre battery is reported.

SuperDobbo, 2022-12-20:
Temperature and humidity entities have appeared however they do not update. I will keep an eye on it and see if they start working, I notice it only updates when readings change. The battery seems to have started working and is reporting a value.

jscl32, 2022-12-21:
First of all, thank you for your work!
I followed your procedure. I do have a return of the presence, but no return on the temperature and humidity (N/A).
And this, for the state of the device :
{
    "battery_low": false,
    "occupancy": true,
    "tamper": false,
    "device": {
        "applicationVersion": 67,
        "dateCode": "",
        "friendlyName": "0xa4c138c7863cac6a",
        "hardwareVersion": 1,
        "ieeeAddr": "0xa4c138c7863cac6a",
        "manufacturerID": 4417,
        "manufacturerName": "_TZ3000_kmh5qpmb",
        "model": "TS0202",
        "networkAddress": 21187,
        "powerSource": "Battery",
        "stackVersion": 0,
        "type": "EndDevice",
        "zclVersion": 3
    },
    "linkquality": 255
}

[image]

Koenkk, 2022-12-21:
Can you make sure the configure succeeds? You can force it via the z2m frontend -> device -> yellow refresh icon. If it fails, make sure to wakeup the sensor by clicking a button on it right before configuring it.

jscl32, 2022-12-21:
Yes, i dit it but no change.
I've this message 
022-12-21 17:43:25Failed to configure '0xa4c138c7863cac6a', attempt 1 (Error: Bind 0xa4c138c7863cac6a/1 msRelativeHumidity from '0x00212effff069ab6/1' failed (Error: status: 130) at DeconzAdapter.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:768:19) at Endpoint.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13) at Object.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:9) at Object.configure (/opt/zigbee2mqtt/dist/util/externally-loaded.js:23:9) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:115:13) at Configure.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/configure.ts:55:21))

Koenkk, 2022-12-21:
Configure fails, can you try again with https://gist.github.com/Koenkk/e880d476d7267eb6c6b4caeac1c24f0c and provide me the log?

jscl32, 2022-12-22:
I've the same message
2-12-22 10:49:18Failed to configure '0xa4c138c7863cac6a', attempt 4 (Error: Bind 0xa4c138c7863cac6a/1 msRelativeHumidity from '0x00212effff069ab6/1' failed (Error: status: 130) at DeconzAdapter.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:768:19) at Endpoint.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13) at Object.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:9) at Object.configure (/opt/zigbee2mqtt/dist/util/externally-loaded.js:25:9) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:115:13) at Configure.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/configure.ts:55:21))

This is the log debug :
[log_debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/10285804/log_debug.txt)

Koenkk, 2022-12-22:
- Does the temperature reporting work now? (since it fails at humidity)
- If not, please try again with the updated version of https://gist.github.com/Koenkk/e880d476d7267eb6c6b4caeac1c24f0c

jscl32, 2022-12-22:
No, that doesn't work.
When i try to force the configuration 
2022-12-22 17:39:07Failed to configure '0xa4c138c7863cac6a', attempt 1 (Error: Read 0xa4c138c7863cac6a/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received) at DeconzAdapter.sendZclFrameToEndpoint (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:658:23) at Endpoint.sendRequest (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:299:20) at Endpoint.read (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:472:28) at Object.configureMagicPacket (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/tuya.js:1179:9) at Object.configure (/opt/zigbee2mqtt/dist/util/externally-loaded.js:22:9) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:115:13) at Configure.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/configure.ts:55:21))

Koenkk, 2022-12-22:
Assuming you woke the device right before configuring it, I think we are stuck now.

To continue this someone needs to sniff the traffic when pairing this device to the official TuYa gateway: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html Then we can reverse engineer how the device works.

Secretowo, 2022-12-24:
Hi there, I'm having the same problem but for me the configure seemed successful.
Occupancy and Battery level are working using the custom converter.
I hope these logs are relevant: https://gist.github.com/Secretowo/f5d15bc470db5621507c5c821e25b7dc

Secretowo, 2022-12-27:
After seeing issue https://github.com/Koenkk/zigbee2mqtt/issues/10822 I could confirm that my sensor is a fake and doesnt have a temperature/humidity sensor.  Occupancy does thankfully still work.

Koenkk, 2022-12-28:
@Secretowo I see, I guess the others in this issue are also affected by this. Based on the modelID I also think this is a different device, original uses `TS0601`, this one uses `TS0202`. `_TZ3000_kmh5qpmb` will be detected as a `TS0202` motion sensor now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-02-04 by github-actions[bot]