Koenkk/zigbee2mqtt #15482: [Feature request]: Vimar 14593 switch missing activePower

mattiazeni, 2022-12-10:
### Is your feature request related to a problem? Please describe

I have a Vimar 14593 switch with embedded power sensor. I correctly added it to zigbe2mqtt and to HA and I am able to switch it on and off. However, it seems the active power consumption is not available. The sensor is connected in zigbee mode.

### Describe the solution you'd like

I would like this device to expose also the active power consumed in realtime. 

### Describe alternatives you've considered

I don't think there is an alternative.

### Additional context

I explored the device in the zigbee2mqtt windows in HA and I can see it exposes a cluster haElectricalMeasurement.

[image]

I then went to the Dev console and I am able to retrieve the power consumption value as showed below:
 
[image]

mattiazeni, 2022-12-10:
I am new to zigbee2mqtt and HA but I am a Software Engineer, if you can point me to what I should modify and an example I can probably help.

danieledwardgeorgehitchcock, 2022-12-11:
I have one of these units - will see if I can add the functionality over the next few days

mattiazeni, 2022-12-11:
I made some progress yesterday but being the first time I'm not sure this is the correct way, I started with HA and zigbee2mqtt only two days ago :)

With this:

> zigbeeModel: ['Mains_Power_Outlet_v1.0'],
> model: '14593',
> vendor: 'Vimar',
> description: '16A outlet IoT connected',
> extend: extend.switch(),
> fromZigbee: [fz.electrical_measurement],
> configure: async (device, coordinatorEndpoint, logger) => {
>        const endpoint = device.getEndpoint(10);
>        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
> },
> exposes: [e.switch(), e.power()]

It correctly shows power consumption. 

[image]

[image]

However the value it 10x the actual one. It can be fixed playing with the power_calibration parameter:
[image]

mattiazeni, 2022-12-11:
There is no readEletricalMeasurementConverterAttributes() function in fromZigbee.js even though I see a comment that refers to it:

```
Zigbee2MQTT:error 2022-12-11 10:46:08: Failed to configure 'Presa Tv', attempt 3 (ReferenceError: readEletricalMeasurementConverterAttributes is not defined
```

danieledwardgeorgehitchcock, 2022-12-11:
D'oh!

I saw that and thought that might be the solution! It looks like it should be but yes, when I was looking at it, I couldn't find it either.. Not even an issue about it which is strange..?

Out of curiosity, do you get any response from reading the `multiplier` or `devisor` attributes from `haElectricalMeasurement`?

mattiazeni, 2022-12-11:
Yep I can read them:

```
2022-12-11 11:17:00Read result of 'haElectricalMeasurement': {"acPowerMultiplier":10,"acPowerDivisor":1}
```

danieledwardgeorgehitchcock, 2022-12-11:
Ok,

So I think this may be a quirk of the Vimar device.

The factor calculation for the converter is `acPowerMultiplier` * `acPowerDivisor` which in this case, would equate to 10 which is the factor offset that you are experiencing.

For this to be calculated correctly for the device, I'd have thought both the `acPowerMultiplier` and `acPowerDivisor` should have a value of 1.

We can get around this by creating a local fromZigbee converter:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const fzLocal = {
    vimar_active_power: {
        cluster: 'haElectricalMeasurement',
        type: ['attributeReport', 'readResponse'],
        convert: async (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('activePower')) {
                payload.power = msg.data['activePower'];
            };
        return payload;
        },
    },
};

const definition = {
    zigbeeModel: ['Mains_Power_Outlet_v1.0'],
    model: '14593',
    vendor: 'Vimar',
    description: '16A outlet IoT connected',
    fromZigbee: [fz.on_off, fz.ignore_basic_report, fzLocal.vimar_active_power],
    toZigbee: [tz.on_off],
    exposes: [e.switch(), e.power()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(10);
        await readEletricalMeasurementConverterAttributes(endpoint);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
    },
};

module.exports = definition;
```

mattiazeni, 2022-12-11:
It works! Thanks a lot.

Just a minor fix, we need to declare the payload variable :)

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const fzLocal = {
    vimar_active_power: {
        cluster: 'haElectricalMeasurement',
        type: ['attributeReport', 'readResponse'],
        convert: async (model, msg, publish, options, meta) => {
            var payload = {}
            if (msg.data.hasOwnProperty('activePower')) {
                payload.power = msg.data['activePower'];
            };
            return payload;
        },
    },
};

const definition = {
    zigbeeModel: ['Mains_Power_Outlet_v1.0'],
    model: '14593',
    vendor: 'Vimar',
    description: '16A outlet IoT connected',
    fromZigbee: [fz.on_off, fz.ignore_basic_report, fzLocal.vimar_active_power],
    toZigbee: [tz.on_off],
    exposes: [e.switch(), e.power()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(10);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
    },
};

module.exports = definition;
```

danieledwardgeorgehitchcock, 2022-12-11:
That we do! Good spot!

Do you want to submit a PR or do you want me to?

mattiazeni, 2022-12-11:
You solved it, the honour is yours :)

mattiazeni, 2022-12-14:
I found some time today, I did it [here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5147).

mattiazeni, 2022-12-18:
I made a brand new installation of HA and with the stock config such as below and without multipliers it works correctly.

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const definition = {
    zigbeeModel: ['Mains_Power_Outlet_v1.0'],
    model: '14593',
    vendor: 'Vimar',
    description: '16A outlet IoT connected',
    fromZigbee: [fz.on_off, fz.ignore_basic_report, fz.electrical_measurement],
    toZigbee: [tz.on_off],
    exposes: [e.switch(), e.power()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(10);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
    },
};

module.exports = definition;
```

closed 2023-01-27 by github-actions[bot]