Koenkk/zigbee2mqtt #15483: TS0601 _TZE200_yh7aoahi and _TZE200_vzekyi4c without external converter don't work

jocamane, 2022-12-10:
### What happened?

Hello, I've had an external converter on my z2m for many months, but today I read an answer from @Koenkk to an issue that "external converters are not made for permanent usage, only while testing (since they can break over releases) . The correct solution would be to integrate the external converter for out-of-the-box support."

So I decided to remove the external converter from the z2m and after that my devices stopped working.

**Devices are a smoke detector TS0601 _TZE200_yh7 and _TZE200_vzekyi4c**

**_External converter: ts0601_smoke.js_**

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const fzLocal = {
    tuya_smoke: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp;
                const value = tuya.getDataValue(dpValue);
                switch (dp) {
                case tuya.dataPoints.state:
                    result.smoke = value === 0;
                    break;
                case 14:
                    // battery state, ignore
                    break
                case 15:
                    result.battery = value;
                    break;
                default:
                    meta.logger.warn(`zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #${ dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
    }
}

const definition = {
    zigbeeModel: ['5p1vj8r'],
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_t5p1vj8r'}, {modelID: 'TS0601', manufacturerName: '_TZE200_uebojraa'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_yh7aoahi'}, {modelID: 'TS0601', manufacturerName: '_TZE200_vzekyi4c'}],
    model: 'TS0601_smoke',
    vendor: 'TuYa',
    description: 'Smoke sensor',
    fromZigbee: [fzLocal.tuya_smoke],
    toZigbee: [],
    exposes: [e.smoke(), e.battery()],
};

module.exports = definition;

[1.pdf](https://github.com/Koenkk/zigbee2mqtt/files/10200719/1.pdf)

Can you please help me

thanks

### What did you expect to happen?

that by removing the external converter the equipment would work.

### How to reproduce it (minimal and precise)

I remove the external converter (ts0601_smoke.js)
I restart the z2m
after that the equipment does not work "not supoorted"

### Zigbee2MQTT version

 1.28.2

### Adapter firmware version

0x26780700

### Adapter

ConBee2

Koenkk, 2022-12-11:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-11 by Koenkk

jocamane, 2022-12-11:
OK, thanks @Koenkk .
in the January production release is already included right? 
and then I can remove the external converter ?

Koenkk, 2022-12-11:
Yes and yes 😄

jocamane, 2023-01-22:
Hi @Koenkk 
Yesterday i install the last version (1.29.2) and is ok.

Thanks