Koenkk/zigbee2mqtt #15485: Lidl Livarno HG08010 - TS0505B: 1) Led's never full off & 2) power_on_behavior => error: UNSUPPORTED_ATTRIBUTE

eerr0EERR, 2022-12-11:
### What happened?

1) The LEDs still glow softly when device switched to OFF.
2) Setting "power_on_behavior" generates error: UNSUPPORTED_ATTRIBUTE

### What did you expect to happen?

1) LEDs completely shut off when device switched to OFF
2) Setting "power_on_behavior" being able to select  ... else remove from frontend

### How to reproduce it (minimal and precise)

1) When device switched to OFF via frontend ... state is OFF but LEDs sofly glowing
2) When setting "power_on_behavior" via frontend ... error is generated

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20211217

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

Error 2022-12-11 08:07:36Publish 'set' 'power_on_behavior' to 'Led***' failed: 'Error: Write 0x***/1 genOnOff({"startUpOnOff":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2022-12-11 08:07:36Error: Write 0x***/1 genOnOff({"startUpOnOff":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:105:13) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)

SAOPP, 2022-12-20:
I can confirm the appearance in this month (release), in the exposes of the lamp - power on behavior. But as indicated by the author of the topic, it is impossible to read this attribute and set it accordingly. I know that the native control system of this device allows you to do this - @Koenkk is it possible to implement support for this feature through z2m?

[image]
[image]
[image]

Koenkk, 2022-12-20:
Could you check if the issue is fixed with the following external converter:  https://gist.github.com/Koenkk/bdb57a89e43f3708c8b964ad61c9cf81

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

SAOPP, 2022-12-20:
@Koenkk Ok, I'll check in the morning.

SAOPP, 2022-12-20:
@Koenkk I can't check the behavior of the lamp right now, but I added a converter, and it still fails to read or write the state, and errors appear during these actions. It seems to me that this did not help, but I can only answer about the behavior of the lamp in the morning.

SAOPP, 2022-12-21:
@Koenkk without any changes with this converter homie.

SAOPP, 2022-12-21:
@Koenkk any updates on this or it's as an dimmer is not supported this function?

Koenkk, 2022-12-21:
Can you provide a debug log when trying to set it?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

SAOPP, 2022-12-21:
@Koenkk here is on:

```
Debug 2022-12-21 18:46:26Received MQTT message on 'z2m-zigstar-poe/kidsroom_floorlamp_big/set' with data '{"power_on_behavior":"on"}'
Debug 2022-12-21 18:46:26Publishing 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big'
Info 2022-12-21 18:46:26MQTT publish: topic 'z2m-zigstar-poe/kidsroom_floorlamp_big', payload '{"brightness":254,"color":{"h":212,"hue":212,"s":10,"saturation":10,"x":0.3131,"y":0.3232},"color_mode":"color_temp","color_temp":153,"last_seen":"2022-12-21T18:46:26+02:00","linkquality":43,"state":"OFF"}'
Error 2022-12-21 18:46:26Publish 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big' failed: 'Error: Write 0xa4c1388beb210ac1/1 genOnOff({"startUpOnOff":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2022-12-21 18:46:26Received MQTT message on 'z2m-zigstar-poe/kidsroom_floorlamp_big/set' with data '{"power_on_behavior":"toggle"}'
Debug 2022-12-21 18:46:26Publishing 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big'
Info 2022-12-21 18:46:26MQTT publish: topic 'z2m-zigstar-poe/kidsroom_floorlamp_big', payload '{"brightness":254,"color":{"h":212,"hue":212,"s":10,"saturation":10,"x":0.3131,"y":0.3232},"color_mode":"color_temp","color_temp":153,"last_seen":"2022-12-21T18:46:26+02:00","linkquality":47,"state":"OFF"}'
Error 2022-12-21 18:46:26Publish 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big' failed: 'Error: Write 0xa4c1388beb210ac1/1 genOnOff({"startUpOnOff":2}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2022-12-21 18:46:27Received MQTT message on 'z2m-zigstar-poe/kidsroom_floorlamp_big/set' with data '{"power_on_behavior":"previous"}'
Debug 2022-12-21 18:46:27Publishing 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big'
Info 2022-12-21 18:46:27MQTT publish: topic 'z2m-zigstar-poe/kidsroom_floorlamp_big', payload '{"brightness":254,"color":{"h":212,"hue":212,"s":10,"saturation":10,"x":0.3131,"y":0.3232},"color_mode":"color_temp","color_temp":153,"last_seen":"2022-12-21T18:46:27+02:00","linkquality":43,"state":"OFF"}'
Error 2022-12-21 18:46:27Publish 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big' failed: 'Error: Write 0xa4c1388beb210ac1/1 genOnOff({"startUpOnOff":255}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
```

Koenkk, 2022-12-21:
Seems the wrong converter is being used, can you try again with https://gist.github.com/Koenkk/bdb57a89e43f3708c8b964ad61c9cf81, also confirm that the description in the frontend now ends with `CUSTOM`.

SAOPP, 2022-12-21:
@Koenkk The same:

```
Debug 2022-12-21 22:27:24Received MQTT message on 'z2m-zigstar-poe/kidsroom_floorlamp_big/get' with data '{"power_on_behavior":""}'
Debug 2022-12-21 22:27:24Publishing get 'get' 'power_on_behavior' to 'kidsroom_floorlamp_big'
Debug 2022-12-21 22:27:24Received Zigbee message from 'kidsroom_floorlamp_big', type 'readResponse', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Info 2022-12-21 22:27:24MQTT publish: topic 'z2m-zigstar-poe/kidsroom_floorlamp_big', payload '{"brightness":254,"color":{"h":212,"hue":212,"s":10,"saturation":10,"x":0.3131,"y":0.3232},"color_mode":"color_temp","color_temp":153,"last_seen":"2022-12-21T22:27:24+02:00","linkquality":102,"state":"ON"}'
Error 2022-12-21 22:27:24Publish 'get' 'power_on_behavior' to 'kidsroom_floorlamp_big' failed: 'Error: Read 0xa4c1388beb210ac1/1 genOnOff(["startUpOnOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2022-12-21 22:27:25Received MQTT message on 'z2m-zigstar-poe/kidsroom_floorlamp_big/get' with data '{"power_on_behavior":""}'
Debug 2022-12-21 22:27:25Publishing get 'get' 'power_on_behavior' to 'kidsroom_floorlamp_big'
Debug 2022-12-21 22:27:25Received Zigbee message from 'kidsroom_floorlamp_big', type 'readResponse', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Info 2022-12-21 22:27:25MQTT publish: topic 'z2m-zigstar-poe/kidsroom_floorlamp_big', payload '{"brightness":254,"color":{"h":212,"hue":212,"s":10,"saturation":10,"x":0.3131,"y":0.3232},"color_mode":"color_temp","color_temp":153,"last_seen":"2022-12-21T22:27:25+02:00","linkquality":98,"state":"ON"}'
Error 2022-12-21 22:27:25Publish 'get' 'power_on_behavior' to 'kidsroom_floorlamp_big' failed: 'Error: Read 0xa4c1388beb210ac1/1 genOnOff(["startUpOnOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2022-12-21 22:27:26Received MQTT message on 'z2m-zigstar-poe/kidsroom_floorlamp_big/set' with data '{"power_on_behavior":"off"}'
Debug 2022-12-21 22:27:26Publishing 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big'
Info 2022-12-21 22:27:26MQTT publish: topic 'z2m-zigstar-poe/kidsroom_floorlamp_big', payload '{"brightness":254,"color":{"h":212,"hue":212,"s":10,"saturation":10,"x":0.3131,"y":0.3232},"color_mode":"color_temp","color_temp":153,"last_seen":"2022-12-21T22:27:26+02:00","linkquality":102,"state":"ON"}'
Error 2022-12-21 22:27:26Publish 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big' failed: 'Error: Write 0xa4c1388beb210ac1/1 genOnOff({"startUpOnOff":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2022-12-21 22:27:27Received MQTT message on 'z2m-zigstar-poe/kidsroom_floorlamp_big/set' with data '{"power_on_behavior":"on"}'
Debug 2022-12-21 22:27:27Publishing 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big'
Info 2022-12-21 22:27:27MQTT publish: topic 'z2m-zigstar-poe/kidsroom_floorlamp_big', payload '{"brightness":254,"color":{"h":212,"hue":212,"s":10,"saturation":10,"x":0.3131,"y":0.3232},"color_mode":"color_temp","color_temp":153,"last_seen":"2022-12-21T22:27:27+02:00","linkquality":102,"state":"ON"}'
Error 2022-12-21 22:27:27Publish 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big' failed: 'Error: Write 0xa4c1388beb210ac1/1 genOnOff({"startUpOnOff":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2022-12-21 22:27:27Received MQTT message on 'z2m-zigstar-poe/kidsroom_floorlamp_big/set' with data '{"power_on_behavior":"toggle"}'
Debug 2022-12-21 22:27:27Publishing 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big'
Info 2022-12-21 22:27:27MQTT publish: topic 'z2m-zigstar-poe/kidsroom_floorlamp_big', payload '{"brightness":254,"color":{"h":212,"hue":212,"s":10,"saturation":10,"x":0.3131,"y":0.3232},"color_mode":"color_temp","color_temp":153,"last_seen":"2022-12-21T22:27:27+02:00","linkquality":98,"state":"ON"}'
Error 2022-12-21 22:27:27Publish 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big' failed: 'Error: Write 0xa4c1388beb210ac1/1 genOnOff({"startUpOnOff":2}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2022-12-21 22:27:28Received MQTT message on 'z2m-zigstar-poe/kidsroom_floorlamp_big/set' with data '{"power_on_behavior":"previous"}'
Debug 2022-12-21 22:27:28Publishing 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big'
[... cut ...] 22:27:28+02:00","linkquality":102,"state":"ON"}'
Error 2022-12-21 22:27:28Publish 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big' failed: 'Error: Write 0xa4c1388beb210ac1/1 genOnOff({"startUpOnOff":255}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
```

SAOPP, 2022-12-21:
> also confirm that the description in the frontend now ends with `CUSTOM`

Btw, where is it at frontend?

Koenkk, 2022-12-22:
Click on your device in the frontend, then `Description` like this:

[image]

SAOPP, 2022-12-22:
[image]

@Koenkk Well, seems something wrong at my side? :|

SAOPP, 2022-12-22:
But I have an extrenal converter into the settings and in frontend also.

SAOPP, 2022-12-22:
@Koenkk 

Debug:
```
Debug 2022-12-22 19:23:36Received MQTT message on 'z2m-zigstar-poe/kidsroom_floorlamp_big/get' with data '{"power_on_behavior":""}'
Debug 2022-12-22 19:23:36Publishing get 'get' 'power_on_behavior' to 'kidsroom_floorlamp_big'
Debug 2022-12-22 19:23:36Received Zigbee message from 'kidsroom_floorlamp_big', type 'readResponse', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Info 2022-12-22 19:23:36MQTT publish: topic 'z2m-zigstar-poe/kidsroom_floorlamp_big', payload '{"brightness":254,"color":{"h":212,"hue":212,"s":10,"saturation":10,"x":0.3131,"y":0.3232},"color_mode":"color_temp","color_temp":153,"last_seen":"2022-12-22T19:23:36+02:00","linkquality":160,"state":"ON"}'
Error 2022-12-22 19:23:36Publish 'get' 'power_on_behavior' to 'kidsroom_floorlamp_big' failed: 'Error: Read 0xa4c1388beb210ac1/1 genOnOff(["moesStartUpOnOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2022-12-22 19:23:37Received MQTT message on 'z2m-zigstar-poe/kidsroom_floorlamp_big/set' with data '{"power_on_behavior":"off"}'
Debug 2022-12-22 19:23:37Publishing 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big'
Info 2022-12-22 19:23:37MQTT publish: topic 'z2m-zigstar-poe/kidsroom_floorlamp_big', payload '{"brightness":254,"color":{"h":212,"hue":212,"s":10,"saturation":10,"x":0.3131,"y":0.3232},"color_mode":"color_temp","color_temp":153,"last_seen":"2022-12-22T19:23:37+02:00","linkquality":174,"state":"ON"}'
Error 2022-12-22 19:23:37Publish 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big' failed: 'Error: Write 0xa4c1388beb210ac1/1 genOnOff({"moesStartUpOnOff":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2022-12-22 19:23:37Received MQTT message on 'z2m-zigstar-poe/kidsroom_floorlamp_big/set' with data '{"power_on_behavior":"previous"}'
Debug 2022-12-22 19:23:37Publishing 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big'
Info 2022-12-22 19:23:38MQTT publish: topic 'z2m-zigstar-poe/kidsroom_floorlamp_big', payload '{"brightness":254,"color":{"h":212,"hue":212,"s":10,"saturation":10,"x":0.3131,"y":0.3232},"color_mode":"color_temp","color_temp":153,"last_seen":"2022-12-22T19:23:38+02:00","linkquality":171,"state":"ON"}'
Error 2022-12-22 19:23:38Publish 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big' failed: 'Error: Write 0xa4c1388beb210ac1/1 genOnOff({"moesStartUpOnOff":2}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2022-12-22 19:23:38Received MQTT message on 'z2m-zigstar-poe/kidsroom_floorlamp_big/set' with data '{"power_on_behavior":"on"}'
Debug 2022-12-22 19:23:38Publishing 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big'
Info 2022-12-22 19:23:38MQTT publish: topic 'z2m-zigstar-poe/kidsroom_floorlamp_big', payload '{"brightness":254,"color":{"h":212,"hue":212,"s":10,"saturation":10,"x":0.3131,"y":0.3232},"color_mode":"color_temp","color_temp":153,"last_seen":"2022-12-22T19:23:38+02:00","linkquality":178,"state":"ON"}'
Error 2022-12-22 19:23:38Publish 'set' 'power_on_behavior' to 'kidsroom_floorlamp_big' failed: 'Error: Write 0xa4c1388beb210ac1/1 genOnOff({"moesStartUpOnOff":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
```

Screens:

[image]

[image]

SAOPP, 2022-12-22:
By the way, I apologize, I started a conversation about this Tuya lamp, it seemed to me that there was also a conversation about it, I briefly read the title when I was looking for a problem on it.

Koenkk, 2022-12-23:
The correct converter is used now but it doesn't seem to work for this device.

- Are you sure this device supports it? 
- If yes, can you provide a sniff when setting this via the original tuya gateway? (https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html)

SAOPP, 2022-12-23:
Yes with original Tuya hub this function is present, maybe this functional if saved it in original stuff and repair lamp to the stick... Thank u for ya time homie. I will try with sniffer bit later...

SharkXXL, 2023-01-15:
Hello,

I have the same power_on_behaviour issue with a Lidl Livarno lamp. 
I tried with the external converter file, adjusted the ext_converter.js with the correct vendor, model and zigbee model. But didn't work.

Adapter: CC2652RB USB Dongle
Zigbee2MQTT version: 1.29.2

Zigbee Model: TS0502A
Vendor: Lidl
Model: HG06492C

Debug Log:
Debug 2023-01-15 16:49:49Received MQTT message on 'zigbee2mqtt/lightbulb_01/set' with data '{"power_on_behavior":"previous"}'
Debug 2023-01-15 16:49:49Publishing 'set' 'power_on_behavior' to 'lightbulb_01'
Error 2023-01-15 16:49:49Publish 'set' 'power_on_behavior' to 'lightbulb_01' failed: 'Error: Write 0x84fd27fffeccc448/1 genOnOff({"startUpOnOff":255}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

@Koenkk could you please look into this and let me know if I did everything correct, or if the ext_converter.js would need further adjustments?

Thanks

bruvv, 2023-01-15:
Same issue, please see: https://github.com/Koenkk/zigbee2mqtt/issues/15554#issuecomment-1383149773

Koenkk, 2023-01-16:
@SharkXXL tuya lights don't support power on behavior

eerr0EERR, 2023-02-02:
Solved and close

closed 2023-02-02 by eerr0EERR

RubenKelevra, 2023-07-05:
@eerr0EERR wrote:

> 1. The LEDs still glow softly when device switched to OFF.

This is fixed for you? Mine glow constantly, even when showed off.

wise-rice, 2024-03-24:
> @eerr0EERR wrote:
> 
> > 1. The LEDs still glow softly when device switched to OFF.
> 
> This is fixed for you? Mine glow constantly, even when showed off.

If you remove the ground wire, the glowing will stop. The case will not be grounded anymore, so please do this at you own risk, but the case is coated/painted and as long as nothing is damaged, nothing should happen, IMHO.

RubenKelevra, 2025-11-15:
@bruvv interesting, can you give me a quick overview how to open the housing? 🤔

bruvv, 2025-11-15:
> @bruvv interesting, can you give me a quick overview how to open the housing? 🤔 

Remove the rubber pad from the bottom and then unscrew the 4 screws. There is a 24v to 5v i first removed the ZigBee chip and replaced it with an esp32 but the led strip is from really low quality and is not addressable only per color. So I replaced it with my own dc to dc converter and with an sk6812 144 led strip.