Koenkk/zigbee2mqtt #15488: Problem with Aqara and Xiaomi MiJia body motion sensor

baerni, 2022-12-11:
### What happened?

Hi. I have several Aqara and Xiaomi MiJIa motion sensors.
Some time ago (a few weeks) I noticed that one of them (xiaomi mijia) regularly has the status of motion detection every 55 minutes, even though there is no movement, for example at night.
However, yesterday I noticed that the same thing started happening with another sensor, this time Aqara. 
The other sensors are working properly.
Does anyone know what the problem is?
I attach screenshots showing the behavior of the sensors at night.
[screenshot1](https://ibb.co/Zg7H53c)
[screenshot2](https://ibb.co/6yQNTTV)
[screenshot3](https://ibb.co/pd2byMT)

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

0x26780700

### Adapter

ConBee2

RubenKelevra, 2023-01-14:
I got an Aqara motion sensor and don't see this issue.

Do you use IR emitting devices, like night vision cameras in the area?

baerni, 2023-01-14:
> Do you use IR emitting devices, like night vision cameras in the area?

No

Problem solved. I replaced the battery with a new one in the sensor and the problem disappeared. When I put the old battery back in the sensor, the problem started again

closed 2023-02-21 by github-actions[bot]