Koenkk/zigbee2mqtt #15492: Z2M fail to recognize IKEA LED2004G8

kjd2338, 2022-12-11:
### What happened?

I am using Z2M on HA.

When I trying to reset IKEA LED2004G8 bulb, Z2M recognize it but not correctly.

IKEA LED2004G8 bulb is listed on Z2M's supported device list.

But my Z2M recognize mine as Unsupported device like this.
{"id":19,"type":"Router","ieeeAddr":"0x040d84fffe9b199b","nwkAddr":22272,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRIbulbE26WSglobeclear806lm","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"zclVersion":3}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":16,"stackVersion":103,"hwVersion":1,"dateCode":"20201111","swBuildId":"1.0.012","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1670766400218,"defaultSendRequestWhen":"immediate"}

Mine was manufactured 18th week of 2022, Which means first week of may in 2022.

### Zigbee2MQTT version

1.28.4-11

### Adapter firmware version

20220219

### Adapter

Sonoff zStack3x0

kjd2338, 2022-12-12:
I was able to solve the problem by editing modelID at database.db file

closed 2022-12-12 by kjd2338

Koenkk, 2022-12-12:
Added `TRADFRIbulbE26WSglobeclear806lm`!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)