Koenkk/zigbee2mqtt #15499: Aqara H1 doesn't send events in decoupled mode

pikausp, 2022-12-11:
### What happened?

I have many Aqara H1 devices (with neutral, both single and double rockers). I just bought 3 new ones and none of them seems to send action when they are in decoupled mode. I can control them just fine from the frontend or via HA, however, I never get the event.

https://www.zigbee2mqtt.io/devices/WS-EUK03.html#xiaomi-ws-euk03

I enabled debug logs, but nothing is logged either when I press the buttons. The main difference I noticed between these new buttons and the older ones (which work with no issue) is the version of their firmware

Old functioning ones: Firmware build date 01-29-2021
Newer "broken" ones: Firmware build date 11-10-2021

Any tips on what I should do? I tried factory resetting them as well, but no luck.

### Zigbee2MQTT version

1.28.4 commit: 52e545f

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

pikausp, 2022-12-11:
I will add that I just added a new device H1 device that I haven't paired before (it controls a relay where I don't need any smart functionality so I left it alone) and it's working as the others. Also has the older version of the firmware.

pikausp, 2022-12-12:
I just purchased more H1 switches from another vendor. They are running the old firmware and they work without issue like the rest of the switches with the same firmware.

shiomax, 2022-12-28:
Same issue here. After updating the switch it no longer reports anything. I tried re-applying the firmware update manually but the only one I found was from here

https://github.com/Koenkk/zigbee-OTA/blob/master/images/Xiaomi/20211124154453_OTA_lumi.switch.n1aeu1_0.0.0_1123_20211110.ota

And that is the exact same firmware version 11-10-2021.

Where did this OTA file come from btw? Is that provided by Xiaomi but I just can´t find it, or are people extracting them out of Xiaomi devices running that firmware? Getting access to the OTA file for the older firmware may allow for downgrades so long as Xiaomi did not intentionally block firmware downgrades.

closed 2023-02-04 by github-actions[bot]

j0hann3s, 2023-02-17:
Same issue here. I would also love to get my hands on the previous firmware file to downgrade to get support back for decoupled mode.