Koenkk/zigbee2mqtt #15502: Invalid and problematic version used for HA update entities

frenck, 2022-12-11:
### What happened?

It seems like Zigbee2MQTT feeds Home Assistant false information about the versioning of an update.

For example:

[image]

The latest version is “newer”, but it must contain a valid version number.

As a result of this, the updated entity doesn’t work as expected. A version is required for the update entity.

### What did you expect to happen?

Not “newer”, but an actual version reference.

### How to reproduce it (minimal and precise)

Have an update ready to installed for any device and use the HA discovery.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

Not applicable 

### Adapter

Not applicable

rwojnarowski, 2022-12-12:
i can confirm same issue:

### Zigbee2MQTT version:
1.28.4-1

### Screenshots:
[image]
[image]

Koenkk, 2022-12-12:
Currently the `swBuildId` attribute from the basic cluster is used. Since you cannot determine the new `swBuildId` from an OTA file it shows `newer` once an OTA update available. In Zigbee the `fileVersion` is used to determine if the OTA is newer than the current installed fw. The device requests an OTA with its current `fileVersion`, then z2m checks if the available OTA has a new `fileVersion`. I'll check if we can switch to the `fileVersion` approach, downside is that we only have the `fileVersion` available when the device requests an OTA.

frenck, 2022-12-12:
>  I'll check if we can switch to the fileVersion approach, downside is that we only have the fileVersion available when the device requests an OTA.

That is fine as well, that won't break any logic.

Koenkk, 2022-12-18:
Fixed by https://github.com/Koenkk/zigbee2mqtt/pull/15630

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-18 by Koenkk

frenck, 2022-12-19:
Thanks @Koenkk ❤️ 

Happy Holidays 🎄 

../Frenck

Koenkk, 2022-12-19:
@frenck thanks, you too ❤️