Koenkk/zigbee2mqtt #15509: Switch_type not sent with SSM_U02 state message.

paulbusse, 2022-12-12:
### What happened?

1. When adding an Xiaomi SSM-U02 2 a number entities are being created. For Home Assistant, there are 2 entities creates for `switch_type`

```
homeassistant/sensor/0x54...ca/switch_type/config
homeassistant/select/0x54...ca/switch_type/config
```
Both claiming the same `unique_id`:  `0x54...ca_switch_type_zigbee2mqtt` which is not ok. I think only the `select` should exit.

2. The definition of these 2 devices is a little different but the `select` definition  causes more trouble. The content of the `config` topic is
```json
{
  "availability": [
    {
      "topic": "zigbee2mqtt/bridge/state",
      "value_template": "{{ value_json.state }}"
    },
    {
      "topic": "zigbee2mqtt/Switch Provisie Licht/availability",
      "value_template": "{{ value_json.state }}"
    }
  ],
  "availability_mode": "all",
  "command_topic": "zigbee2mqtt/Switch Provisie Licht/set/switch_type",
  "device": {
    "configuration_url": "https://framboos.bv.lan:8080/#/device/0x54...ca/info",
    "identifiers": [
      "zigbee2mqtt_0x54...ca"
    ],
    "manufacturer": "Xiaomi",
    "model": "Aqara single switch module T1 (without neutral). Doesn't work as a router and doesn't support power meter (SSM-U02)",
    "name": "Switch Provisie Licht"
  },
  "entity_category": "config",
  "icon": "mdi:tune",
  "name": "Switch Provisie Licht switch type",
  "options": [
    "toggle",
    "momentary"
  ],
  "state_topic": "zigbee2mqtt/Switch Provisie Licht",
  "unique_id": "0x54...ca_switch_type_zigbee2mqtt",
  "value_template": "{{ value_json.switch_type }}"
}
```
However the messages sent on the `state_topic` does not always contain a `switch_type`
```json
{
  "brightness": 254,
  "color_mode": "color_temp",
  "color_temp": 370,
  "state": "OFF"
}
```
This causes error messages in 
```
WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'switch_type' when rendering '{{ value_json.switch_type }}'
(MainThread) [homeassistant.components.mqtt.select] Invalid option for select.switch_provisie_licht_switch_type: '' (valid options: ['toggle', 'momentary'])
```
3. Whenever I restart the Zigbee2MQTT the `switch_type` setting is lost.

### What did you expect to happen?

See above

### How to reproduce it (minimal and precise)

See above

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

12-09-2020

### Adapter

SONOFF ZBDongle-P

paulbusse, 2023-02-15:
Any updates - bis? :-)

paulbusse, 2023-03-19:
Any updates - tris? :-)

mazafakawaka, 2023-04-19:
I am also ave the same issue.

paulbusse, 2023-05-20:
Any progress

mazafakawaka, 2023-06-22:
the same problem

paulbusse, 2023-07-28:
Hi, is there anything we can do about this?

paulbusse, 2023-08-30:
Still waiting on a response

paulbusse, 2023-10-02:
Is there a reason this pronblem is not picked up

paulbusse, 2023-11-03:
I can be very stubborn too :-)

closed 2024-06-01 by github-actions[bot]