Koenkk/zigbee2mqtt #15511: Module LoraTap SC500ZB: interview failed

Leopol123, 2022-12-12:
### What happened?

Hello, 

I recently buy a LoraTap Zigbee module knowned as SC500ZB : [https://fr.loratap.com/products/sc500zb-v2](url) .

I try to add it using Zigbee2MQTT and the module cannot be paired.

Thanks for your help.

### What did you expect to happen?

Module paired with a constructor and recognised.

### How to reproduce it (minimal and precise)

I try to paired it near to zigbee cordinator. Module is plugged using a powercord.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8 (Sonoff  ZigBee 3.0 USB Dongle Plus)

### Debug log

`2022-12-12T16:48:17.643Z zigbee-herdsman:controller:log Device '0xa4c138c9a6da7e14' is already in database with different networkAddress, updating networkAddress
2022-12-12T16:48:17.644Z zigbee-herdsman:controller:log Interview '0xa4c138c9a6da7e14' start
Zigbee2MQTT:info  2022-12-12 17:48:17: Starting interview of '0xa4c138c9a6da7e14'
Zigbee2MQTT:info  2022-12-12 17:48:17: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138c9a6da7e14","ieee_address":"0xa4c138c9a6da7e14","status":"started"},"type":"device_interview"}'
2022-12-12T16:48:17.670Z zigbee-herdsman:controller:device:log Interview - start device '0xa4c138c9a6da7e14'
2022-12-12T16:48:17.671Z zigbee-herdsman:adapter:ezsp:debg Requesting 'Node Descriptor' for '53055'
2022-12-12T16:48:17.671Z zigbee-herdsman:adapter:ezsp:driv ZDO Node_Desc_req params: {"dstaddr":53055}
2022-12-12T16:48:17.671Z zigbee-herdsman:adapter:ezsp:ezsp ==> setExtendedTimeout: {"remoteEui64":{"_value":{"type":"Buffer","data":[164,193,56,201,166,218,126,20]}},"extendedTimeout":true}
2022-12-12T16:48:17.679Z zigbee-herdsman:adapter:ezsp:uart <-- [520ba1a9212aea25937e]
2022-12-12T16:48:17.679Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (5,2,0): 520ba1a9212aea25937e
2022-12-12T16:48:17.679Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (6)
2022-12-12T16:48:17.679Z zigbee-herdsman:adapter:ezsp:uart --> [8610be7e]
2022-12-12T16:48:17.679Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 4980017500ff
2022-12-12T16:48:17.680Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x75: {"_id_":117,"_cls_":"findKeyTableEntry","_isRequest_":false,"index":255}
2022-12-12T16:48:17.680Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (2) success
2022-12-12T16:48:17.680Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"setExtendedTimeout","_id_":126,"_isRequest_":true,"remoteEui64":{"_value":{"type":"Buffer","data":[164,193,56,201,166,218,126,20]}},"extendedTimeout":true}
2022-12-12T16:48:17.680Z zigbee-herdsman:adapter:ezsp:uart --> DATA (2,6,0): 4a00017e00147edaa6c938c1a401
2022-12-12T16:48:17.680Z zigbee-herdsman:adapter:ezsp:uart --> [260821a92a2a01cc8332831d6bf1936f6a7e]
2022-12-12T16:48:17.680Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (3)
2022-12-12T16:48:17.686Z zigbee-herdsman:adapter:ezsp:uart <-- [6308a1a92a2a4e0f7e]
2022-12-12T16:48:17.686Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (6,3,0): 6308a1a92a2a4e0f7e
2022-12-12T16:48:17.686Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (7)
2022-12-12T16:48:17.686Z zigbee-herdsman:adapter:ezsp:uart --> [87009f7e]
2022-12-12T16:48:17.686Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 4a80017e00
2022-12-12T16:48:17.686Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x7e: {"_id_":126,"_cls_":"setExtendedTimeout","_isRequest_":false}
2022-12-12T16:48:17.686Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (3) success
2022-12-12T16:48:17.686Z zigbee-herdsman:adapter:ezsp:ezsp ==> sendUnicast: {"type":0,"indexOrDestination":53055,"apsFrame":{"clusterId":2,"profileId":0,"sequence":10,"sourceEndpoint":0,"destinationEndpoint":0,"groupId":0,"options":320},"messageTag":11,"message":{"type":"Buffer","data":[10,63,207]}}
2022-12-12T16:48:17.686Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"sendUnicast","_id_":52,"_isRequest_":true,"type":0,"indexOrDestination":53055,"apsFrame":{"clusterId":2,"profileId":0,"sequence":10,"sourceEndpoint":0,"destinationEndpoint":0,"groupId":0,"options":320},"messageTag":11,"message":{"type":"Buffer","data":[10,63,207]}}
2022-12-12T16:48:17.687Z zigbee-herdsman:adapter:ezsp:uart --> DATA (3,7,0): 4b00013400003fcf000002000000400100000a0b030a3fcf
2022-12-12T16:48:17.687Z zigbee-herdsman:adapter:ezsp:uart --> [370921a9602a158d96944a27aa5592099d4e27a1e6cd6db432fb837e]
2022-12-12T16:48:17.687Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (4)
2022-12-12T16:48:17.695Z zigbee-herdsman:adapter:ezsp:uart <-- [7409a1a9602a158967f17e]
2022-12-12T16:48:17.695Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (7,4,0): 7409a1a9602a158967f17e
2022-12-12T16:48:17.695Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (0)
2022-12-12T16:48:17.696Z zigbee-herdsman:adapter:ezsp:uart --> [8070787e]
2022-12-12T16:48:17.696Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 4b80013400003b
2022-12-12T16:48:17.696Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x34: {"_id_":52,"_cls_":"sendUnicast","_isRequest_":false,"status":0,"sequence":59}
2022-12-12T16:48:17.696Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (4) success
2022-12-12T16:48:22.482Z zigbee-herdsman:adapter:ezsp:uart <-- [0409b1a96b2a158d96944a27aa5592099c4e2790e6a86792987e]
2022-12-12T16:48:22.482Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,4,0): 0409b1a96b2a158d96944a27aa5592099c4e2790e6a86792987e
2022-12-12T16:48:22.482Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2022-12-12T16:48:22.482Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2022-12-12T16:48:22.482Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 4b90013f00003fcf000002000000400000003b0b6600
2022-12-12T16:48:22.482Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x3f: {"_id_":63,"_cls_":"messageSentHandler","_isRequest_":false,"type":0,"indexOrDestination":53055,"apsFrame":{"profileId":0,"clusterId":2,"sourceEndpoint":0,"destinationEndpoint":0,"options":64,"groupId":0,"sequence":59},"messageTag":11,"status":102,"message":{"type":"Buffer","data":[]}}
2022-12-12T16:48:22.814Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0
2022-12-12T16:48:22.814Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
2022-12-12T16:48:22.814Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
2022-12-12T16:48:22.814Z zigbee-herdsman:adapter:ezsp:uart --> DATA (4,1,0): 4c00010500
2022-12-12T16:48:22.814Z zigbee-herdsman:adapter:ezsp:uart --> [410e21a9512a34597e]
2022-12-12T16:48:22.814Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (5)
2022-12-12T16:48:22.819Z zigbee-herdsman:adapter:ezsp:uart <-- [150ea1a9512a9f547e]
2022-12-12T16:48:22.819Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,5,0): 150ea1a9512a9f547e
2022-12-12T16:48:22.819Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (2)
2022-12-12T16:48:22.819Z zigbee-herdsman:adapter:ezsp:uart --> [82503a7e]
2022-12-12T16:48:22.819Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 4c80010500
2022-12-12T16:48:22.820Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_id_":5,"_cls_":"nop","_isRequest_":false}
2022-12-12T16:48:22.820Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (5) success
2022-12-12T16:48:27.672Z zigbee-herdsman:adapter:ezsp:debg Node descriptor request for '53055' failed (Error: {"address":53055,"clusterId":32770,"sequence":10} after 10000ms), retry
2022-12-12T16:48:27.672Z zigbee-herdsman:controller:device:log Interview - quirks check for 'undefined'-'undefined'-'Unknown'
2022-12-12T16:48:27.672Z zigbee-herdsman:controller:device:log Interview - quirks did not match
2022-12-12T16:48:27.672Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0xa4c138c9a6da7e14', attempt 1`

Leopol123, 2022-12-20:
Hello,

I bought two of these modules, I try the second one, and it's working perfectly (knowed as TF130F).

I try again and again association on first one without succeed. I open both of my modules, and there are not from the same release. 

One PCB is black with a reference 2218-SC500ZB-KP3210 => not working
The second one is green with reference SC-ZB-CONTROL-V4 => working

I bought those two modules on same vendor (loratap official) on the same order. So be careful with SC500ZB ! I contact the vendor, he asked me to pair with tuya gateway that i don't have.

I wil close this ticket with answer from aliexpress vendor in few days.

Thanks for zigbee2mqtt, great solution !

closed 2023-01-27 by github-actions[bot]

uNiqu3MK, 2023-10-30:
same problem here.. do you find way to pair with zb2mqtt?

Leopol123, 2023-11-01:
No, I buy another model: QS-Zigbee-CP03 that is working fine. I don't test  on new zigbee2mqtt release.