Koenkk/zigbee2mqtt #15521: [New device support]: Philips Hue GU10 White Ambiance 9290019533

tobiasfrank87, 2022-12-12:
### Link

https://www.philips-hue.com/en-sg/p/hue-white-ambiance-gu10---smart-spotlight/8719514339941

### Database entry

{"id":16,"type":"Router","ieeeAddr":"0x001788010bcb39ef","nwkAddr":20892,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"929003046901","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":268,"inClusterList":[0,3,4,5,6,8,4096,64514,768],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"929003046901","manufacturerName":"Philips","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":3,"dateCode":"20200327","swBuildId":"1.65.11_hB798F2B"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":3,"dateCode":"20200327","swBuildId":"1.65.11_hB798F2B","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1670887344796,"defaultSendRequestWhen":"immediate"}

### Comments

I ran in a weird problem, that my Philips White Ambiance GU10 spots from a HUE Pillar lamp are beeing marked as unsupported. I found a similiar problem a year ago and now i have the same problem with the same spots. https://github.com/Koenkk/zigbee2mqtt/issues/9428

tobiasfrank87, 2022-12-12:
The Model Code at the spot is: 9290019533 although the database entry show different digits.

danieledwardgeorgehitchcock, 2022-12-13:
Could you just clarify, are these bulbs that have been removed from a Hue fixture I.e. the Pillar Spot that you mention above, or are they standard bulbs like as what you have linked?

tobiasfrank87, 2022-12-13:
[image]
[image]

i attached pictures of the spots i took from the HUE Pillar lamp

danieledwardgeorgehitchcock, 2022-12-13:
Thanks,

I've added it in the PR above

closed 2022-12-13 by Koenkk