Koenkk/zigbee2mqtt #15522: Xiaomi WXKG01LM action stops working after minutes

benixgil, 2022-12-13:
### What happened?

I have used an Hacs integration to connect the Xiaomi wireless switches to home assistant. But since I'm using zigbee2mqtt now, I have decided to migrate them to Zigbee2mqtt.
At first, after pairing all seems fine, single, double, triple and hold clicks all work fine in action.
But after a few minutes they stop working. Everything else, battery, link, voltage, work fine. But action shows N/A and doesn't change to "empty string" like it usually does every time I click.
The only way I have found to fix it's to repair. But again, after a few minutes, the same happens again.
I think repairing every 15 minutes it's not the way.
Sorry to not provide any more information, I'm new to this platform. Don't even know how to debud or how to show you any logs.

### What did you expect to happen?

Action responded like it doesn on other devices.

### How to reproduce it (minimal and precise)

Just pair it and it happen. Single click or any other click doesn't work after a few minutes.

### Zigbee2MQTT version

1.28

### Adapter firmware version

20220813141534-if00

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus v2

kirovilya, 2022-12-13:
Try updating Z2M to version 1.28.4

benixgil, 2022-12-13:
Sorry @kirovilya, my bad, I have wrote it wrong. I'm actually using 1.28.4-1. I have tried reinstalling z2m but the problem remains. Which is really weird, cause none of my other devices, I even have Ikea shortcut buttons, and to none off this happens.

MrHollowPS, 2022-12-15:
I found a way to fix this by creating a SWITCH node in node red which ignores the empty string action. I had the same problem as you did, this was my workaround, it also happens with ikea buttons.

benixgil, 2022-12-17:
Well, I don't know how to use node red, I've just went back to the way they were setup before.
I have tried way to many times in way to many different ways. So I have just connected them back to the Xiaomi app and connected them to HA through the Xiaomi integration. But thanks for your help. When it starts working again, I'll migrate them to HA Z2M.
Fun fact, my Ikea buttons work fine. Nothing like this.

mikeo999, 2023-01-23:
I am having the same problem with the WXKG01LM switch and Sonoff 3.0 (e-version). 
They work for about 1 hours and they stop working.  Nothing in the logfiles and all other Zigbee keeps working.

For now bought a Sonoff 3.0 (p-version) and all is working fine.

Zigbee2MQTT version 1.29.2-1
Home Assistant: 2023.1.7

d-allin, 2023-01-23:
I resolved this issue by flashing some firmware to the receiver.
See full discussion w/ instructions at https://github.com/Koenkk/zigbee2mqtt/discussions/15443#discussioncomment-4409143

mikeo999, 2023-01-24:
> I resolved this issue by flashing some firmware to the receiver. See full discussion w/ instructions at [#15443 (comment)](https://github.com/Koenkk/zigbee2mqtt/discussions/15443#discussioncomment-4409143)

Thanks will try this soon. Just finished connection all devices to the P version. When switching back to the E version i need some time to repair them all again.

closed 2023-03-03 by github-actions[bot]

litecross91, 2023-10-17:
Some of my devices are not responding after 1-2 hours (WXKG01LM, RTCGQ01LM) Server: rpi3, HA, Coordinator: dongle-e, FW6.10.3.0 build 297, ezsp v8, Zigbee2mqtt 1.33.1, mosquito broker

d-allin, 2023-10-17:
> Some of my devices are not responding after 1-2 hours (WXKG01LM, RTCGQ01LM) Server: rpi3, HA, Coordinator: dongle-e, FW6.10.3.0 build 297, ezsp v8, Zigbee2mqtt 1.33.1, mosquito broker

See https://github.com/Koenkk/zigbee2mqtt/discussions/15443#discussioncomment-4627330