Koenkk/zigbee2mqtt #15523: Smarthings Water leak sensor (2018 model) - IM6001-WLP01

cloudbr34k84, 2022-12-13:
### What happened?

For some reason when I pair the advice i cant get all the entities exposed. 
[image]

But if i try and repair i get the opposite entities (see example of another ST Water Sensor that shows Battery etc
[image]

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

Sonoff Coordinator

sjorge, 2022-12-15:
Do the nulls disapear if you manually bridge the contact? (A drop of water should work)

The missing values after you repaired are on the ssIasZone cluster which AFAIK has no steps in the configure function to setup.
Only msTempersture and genPowerCfg do, and those are the only values populated.

cloudbr34k84, 2022-12-15:
Thanks, you I bridged the gap as you call it. Seems to be working now thanks !!

closed 2022-12-15 by cloudbr34k84