Koenkk/zigbee2mqtt #15525: Error: Cannot find module '../lib/exposes'

cloudbr34k84, 2022-12-13:
### What happened?

Go the follow error when adding the Sonoff Converter File
```
[17:39:47] INFO: Preparing to start...
[17:39:47] INFO: Socat not enabled
[17:39:48] INFO: Starting Zigbee2MQTT...
node:internal/modules/cjs/loader:959
  throw err;
  ^
Error: Cannot find module '../lib/exposes'
Require stack:
- /app/dist/util/utils.js
- /app/dist/util/settings.js
- /app/index.js
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:956:15)
    at Function.Module._load (node:internal/modules/cjs/loader:804:27)
    at Module.require (node:internal/modules/cjs/loader:1028:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at /app/dist/util/externally-loaded.js:1:17
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
```

### How to reproduce it (minimal and precise)

I added this file.js
I restarted Z2M and this is the error im now getting. Sorry no idea how to even understand this :(

```
const exposes = require('../lib/exposes');
const fz = {...require('../converters/fromZigbee'), legacy: require('../lib/legacy').fromZigbee};
const tz = require('../converters/toZigbee');
const constants = require('../lib/constants');
const reporting = require('../lib/reporting');
const extend = require('../lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const ota = require('../lib/ota');

const fzLocal = {
    // SNZB-02 reports stranges values sometimes
    // https://github.com/Koenkk/zigbee2mqtt/issues/13640
    SNZB02_temperature: {
        ...fz.temperature,
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.measuredValue > -10000 && msg.data.measuredValue < 10000) {
                return fz.temperature.convert(model, msg, publish, options, meta);
            }
        },
    },
    router_config: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty('currentLevel')) {
                result.light_indicator_level = msg.data['currentLevel'];
            }
        },
    },
};

module.exports = [
    {
        zigbeeModel: ['BASICZBR3'],
        model: 'BASICZBR3',
[... 210 lines cut ...]
        model: 'ZBDongle-E',
        vendor: 'SONOFF',
        description: 'Sonoff Zigbee 3.0 USB Dongle Plus (EFR32MG21) with router firmware',
        fromZigbee: [fz.linkquality_from_basic, fzLocal.router_config],
        toZigbee: [],
        exposes: [exposes.numeric('light_indicator_level').withDescription('Brightness of the indicator light').withAccess(ea.STATE)],
        configure: async (device, coordinatorEndpoint, logger) => {
            device.powerSource = 'Mains (single phase)';
            device.save();
        },
    },
];
```

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

Sonoff Coordinator

cloudbr34k84, 2022-12-13:
> you need to use absolute references for external converters. This should be in the format of `const exposes = require('zigbee-herdsman-converters/lib/exposes');`

i have no idea about this, but im looking at other examples of converter.js files and they all have this at the top. Is that what i need?

If so how come the file doenst have it already like others do??
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');
```

danieledwardgeorgehitchcock, 2022-12-13:
Looking at the above, I am not sure what you are trying to do - External Converters are usually for adding support / modifying single devices and will need to be included as a reference in your `configuration.yaml` file - these are what require an absolute reference. It looks like you are trying to modify / replace the whole `sonoff.js` file away from its standard directory? If you are modifiying it in place, then relative references are fine

closed 2022-12-17 by cloudbr34k84