Koenkk/zigbee2mqtt #15527: Error with alchemy.js  - TypeError: extend.light_onoff_brightness is not a function

cloudbr34k84, 2022-12-13:
### What happened?

[20:29:02] INFO: Preparing to start...
[20:29:03] INFO: Socat not enabled
[20:29:03] INFO: Starting Zigbee2MQTT...
/app/dist/util/externally-loaded.js:488
        extend: extend.light_onoff_brightness(),
                       ^
TypeError: extend.light_onoff_brightness is not a function
    at /app/dist/util/externally-loaded.js:488:24
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)

### How to reproduce it (minimal and precise)

Just trying to add this converter, and i get the follow error when restarting Z2m?? what does this mean?

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

Sonoff Coordinator

danieledwardgeorgehitchcock, 2022-12-13:
Can you post up your external converter?

cloudbr34k84, 2022-12-13:
hey, i deleted it , bit its the alchemy.js code

```
const extend = require('zigbee-herdsman-converters/lib/extend');

module.exports = [
    {
        zigbeeModel: ['AL8TC13W-AP'],
        model: 'AL8TC13W-AP',
        vendor: 'Alchemy',
        description: 'Downlight with tuneable white',
        extend: extend.light_onoff_brightness_colortemp({colorTempRange: [153, 370]}),
    },
];
```

danieledwardgeorgehitchcock, 2022-12-13:
So you're using the base of the alchemy.js code as an external converter? If so, try this:

```javascript
const extend = require('zigbee-herdsman-converters/lib/extend');

const definition = {
    zigbeeModel: ['AL8TC13W-AP'],
    model: 'AL8TC13W-AP',
    vendor: 'Alchemy',
    description: 'Downlight with tuneable white',
    extend: extend.light_onoff_brightness_colortemp({colorTempRange: [153, 370]}),
};

module.exports = definition;
```

cloudbr34k84, 2022-12-13:
OK great that seemed to work and Z2M and successfully restarted, now to add the lights

closed 2022-12-17 by cloudbr34k84